dariuszseweryn/RxAndroidBle #95: Check for characteristics

Orbyt, 2016-11-11:
I see that RxBleConnection has a `.discoverServices()` method. Is there a similar method to check if certain characteristics exits? As of now, the way I think I would have to do this is loop through all possible characteristics and try a read operation and check if it throws an error/is null.

Is there a simpler way to discover characteristics similar to the above `.discoverServices()` method.

uKL, 2016-11-11:
Hey Orbyt!

For questions please use stackoverflow.com with a #rxandroidble tag!

Cheers!

closed 2016-11-11 by uKL

Orbyt, 2016-11-11:
This wouldn't be a suitable question for SO as there is no relevant code or examples to be presented, and it is more of a question of included functionality of this library. Please reconsider and let me know.

uKL, 2016-11-11:
In general, your understanding is correct. You'd need to discover services then use `RxBleDeviceServices#getCharacteristic` and check if `BleCharacteristicNotFoundException` has been emitted.