dariuszseweryn/RxAndroidBle #97: Add more context to `BleDisconnectedException`.

dariuszseweryn, 2016-11-16:
### Summary
<br />Would be nice to have more context in the `BleDisconnectedException`. For instance the device mac address if available. It would be useful in https://github.com/Polidea/react-native-ble-plx/

dariuszseweryn, 2017-02-02:
Version `1.2.0-SNAPSHOT`

closed 2017-02-02 by dariuszseweryn

RobLewis, 2017-02-02:
Sorry, I am new to this. 

I have searched your GitHub site and cannot find any reference to 1.2.0-SNAPSHOT. Where is it, please? 
Can it be added to my project by replacing this line in build.gradle: 
   compile 'com.polidea.rxandroidble:rxandroidble:1.1.0'
If so, what would I use? 

Do you have any idea when 1.2.0 will be officially released? 

Are the fixes you have recently made incorporated in the current 1.1.0 version? Is there any sort of Change Log available? 

I just discovered that there is apparently an open issue regarding API 21 compatibility. This is the API version that I have to write to. Am I going to have a problem? 

Thank you for any help you can give! I really appreciate the great work you are doing. 

--Rob Lewis

dariuszseweryn, 2017-02-03:
@RobLewis 
To use the snapshot version you would need to add the Sonatype Snapshot Repository
`maven { url "https://oss.sonatype.org/content/repositories/snapshots" }` in your `build.gradle`.

`1.2.0` Should be released `soon™` - No date yet though.

`CHANGELOG.md` log is updated right before each official release. You can still check the commits that were merged after the last release to know what is in `1.2.0-SNAPSHOT` and what would be in the official `1.2.0`

Andoid Lollipop has introduced a new API for BLE scanning. This library is still not using it. It uses the old BLE scanning API which is now deprecated but still works. I anticipate no problems until the deprecated API will be removed. The library will probably be in version `2.0.0` till that time and will use the new API.

Best