dariuszseweryn/RxAndroidBle #108: Can't connect in example4_characteristic

BudinMilk, 2016-12-22:
### Summary
@OnClick(R.id.connect)
    public void onConnectToggleClick() {

        if (isConnected()) {
            triggerDisconnect();
        } else {
            connectionObservable.subscribe(rxBleConnection -> {
                Log.d(getClass().getSimpleName(), "Hey, connection has been established!");
                runOnUiThread(this::updateUI);
            }, this::onConnectionFailure);
        }
    }
<br />

#### Preconditions
<br />

#### Steps to reproduce actual result
1. Click "connect" button for connect, but connection always can't establish.
<br />

#### Actual result
connection can't establish.
<br />
#### Expected result
Click "connect" button can connection establish.
<br />

dariuszseweryn, 2016-12-22:
Hello!
Thank you for the report. Could you describe what is happening and / or paste the logs from the application? It would greatly help.
Best Regards

BudinMilk, 2016-12-23:
Hello!

Here is my screen, the other buttons always disabled
[image]

And here is my log, thank you!
I/Timeline: Timeline: Activity_launch_request time:1395848
V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@f6e496e
V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@568f80f
D/ActivityThreadInjector: clearCachedDrawables.
D/OpenGLRenderer: endAllStagingAnimators on 0x55832f2350 (RippleDrawable) with handle 0x55832eee50
D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(222846792)
D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(222846792)
D/BluetoothGatt: connect() - device: 18:7A:00:00:00:00, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=7059e136-3d1d-46ec-87b5-ff89b98333a3
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=18:7A:00:00:00:00
D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=133
D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(30052315)
D/RxBle#RadioOperationConnect$$Lambda: No subscribers, finishing operation
D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(222846792)
D/RxBle#Radio:  STARTED RxBleRadioOperationDisconnect(30052315)
D/BluetoothManager: getConnectionState()
D/BluetoothManager: getConnectedDevices
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=5
D/RxBle#Radio: FINISHED RxBleRadioOperationDisconnect(30052315)
D/AndroidRuntime: Shutting down VM
E/AndroidRuntime: FATAL EXCEPTION: main
                  Process: com.polidea.rxandroidble.sample, PID: 12309
                  java.lang.NullPointerException: Attempt to invoke virtual method 'rx.Subscription rx.Observable.subscribe(rx.functions.Action1, rx.functions.Action1)' on a null object reference
                      at com.polidea.rxandroidble.sample.example4_characteristic.CharacteristicOperationExampleActivity.onConnectToggleClick(CharacteristicOperationExampleActivity.java:116)
                      at com.polidea.rxandroidble.sample.example4_characteristic.CharacteristicOperationExampleActivity_ViewBinding$1.doClick(CharacteristicOperationExampleActivity_ViewBinding.java:38)
                      at butterknife.internal.DebouncingOnClickListener.onClick(DebouncingOnClickListener.java:22)
                      at android.view.View.performClick(View.java:5207)
                      at android.view.View$PerformClick.run(View.java:21177)
                      at android.os.Handler.handleCallback(Handler.java:739)
                      at android.os.Handler.dispatchMessage(Handler.java:95)
                      at android.os.Looper.loop(Looper.java:148)
                      at android.app.ActivityThread.main(ActivityThread.java:5441)
                      at java.lang.reflect.Method.invoke(Native Method)
                      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:738)
                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:628)

uKL, 2016-12-28:
Thanks, this has been fixed in master! Please let us know if you'll find anything else.

closed 2016-12-28 by uKL