dariuszseweryn/RxAndroidBle #109: Enhance sample application to provide a menu of activities

RobLewis, 2016-12-30:
### Summary
I was able to load the RxAndroidBle-master project into Android Studio, and it compiles and runs. Thank you! But when it runs, I only see Example 1, ScanActivity, and I don't see any way to access the other examples. Is there some way to have a main menu that would allow choosing an activity? 
I see there is a SampleApplication class, but it doesn't seem to do much. 
And can you tell me what this comment found in it means?
/**
* In practise you will use some kind of dependency injection pattern.
*/
<br />
#### Preconditions
Android Studio
<br />
#### Steps to reproduce actual result
1. Install RxAndroidBle-master project on Android tablet
2. Run it
3. Try to exercise all functions
<br />
#### Actual result
Only Example 1, ScanActivity, is accessible
<br />
#### Expected result
A menu for selecting all the available Example activities should appear
<br />

uKL, 2016-12-30:
Hey Rob!

Thanks for the report. You are more than welcome to contribute with a menu in the example.

The comment means that there are clever ways to maintain single instance of the client. For example you can use the Dagger with a application scope component. For more information about dependency injection pattern please go to https://google.github.io/dagger/

dariuszseweryn, 2016-12-31:
Hello Rob,

All of the other Activities in the application need a bluetooth device to run. This device needs to be scanned in the Example 1 first and to access other screens you need to tap the device you want to use on the list.

Best Regards

RobLewis, 2017-01-03:
I don't know what changed, but the app seems to be working now.

closed 2017-01-03 by RobLewis

dariuszseweryn, 2017-01-03:
Maybe you have turned you BT adapter off and on again. It sometimes help.