dariuszseweryn/RxAndroidBle #110: Queuing for 20 seconds when one device is powered off

lovedise, 2017-01-03:
### Summary
<br />
When you receive data from two devices and turn off the power of one device, the other should continuously receive the data, but the connection can not be disconnected and readCharacteristic event is queued.

```
distanceDevice?.let { distance ->
                distance.connectObservable.concatMap { connect ->
                    Observable.interval(1, TimeUnit.SECONDS).flatMap {
                        connect.readCharacteristic(distance.READ_DISTANCE_UUID)
                    }
                }
                .bindUntilEvent(this, ActivityEvent.PAUSE)
                .subscribeOn(Schedulers.io())
                .observeOn(AndroidSchedulers.mainThread())
                .subscribe({
                    Timber.d("distance : ${HexString.getHexToDec(HexString.bytesToHex(it))}")
                }, {
                    it.printStackTrace()
                })

lightDevice?.let { light ->
                light.connectObservable.concatMap { connect ->
                    Observable.interval(1, TimeUnit.SECONDS).flatMap {
                        connect.readCharacteristic(distance.READ_LIGHT_UUID)
                    }
                }
                .bindUntilEvent(this, ActivityEvent.PAUSE)
                .subscribeOn(Schedulers.io())
                .observeOn(AndroidSchedulers.mainThread())
                .subscribe({
                    Timber.d("distance : ${HexString.getHexToDec(HexString.bytesToHex(it))}")
                }, {
                    it.printStackTrace()
                })
```

<br/><br/>
D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(111281633)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(60626133)
D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(60626133)
W/BluetoothGatt: onCharacteristicRead() - Device=00:05:5B:01:15:2D handle=24 Status=0
D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=499ee8bf-b20a-4dfb-8e47-a6ac0ed5baa1 status=0
D/StageActivity$prepareView: distance : 0
D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(60626133)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(219459977)
D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(219459977)
W/BluetoothGatt: onCharacteristicRead() - Device=00:06:5B:03:15:2A handle=24 Status=0
D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=5a1aad12-5260-49c2-80ed-4bb80a63eaa1 status=0
D/StageActivity$prepareView: distance : 5125
D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(219459977)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(37187581)
D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(37187581)
W/BluetoothGatt: onCharacteristicRead() - Device=00:05:5B:01:15:2D handle=24 Status=0
D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=499ee8bf-b20a-4dfb-8e47-a6ac0ed5baa1 status=0
D/StageActivity$prepareView: distance : 0
D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(37187581)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(45065265)
D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(45065265)
W/BluetoothGatt: onCharacteristicRead() - Device=00:06:5B:03:15:2A handle=24 Status=0
D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=5a1aad12-5260-49c2-80ed-4bb80a63eaa1 status=0
D/StageActivity$prepareView: distance : 5125
#### Preconditions

<br /><br/>
#### Steps to reproduce actual result<br/>
1. Connect to two ble devices <br/>
2. Data is received continuously once every second<br/>
3. Turn off the power of one device<br/><br/>
#### Actual result
<br /><br/>
D/RxBle#Radio:  QUEUED RxBleRadioOperationCharacteristicRead(198164112)
D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(198164112)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(229755585)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(7286514)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(160355999)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(1012440)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(192787565)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(216677614)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(239864747)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(144471732)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(260669401)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(131863978)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(18714743)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(170818384)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(180123909)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(29480230)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(196284675)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(134315180)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(222407665)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(205346658)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(58587471)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(8140488)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(108258461)
<br /><br/>
#### Expected result
<br /><br/>
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(179872761)
D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(179872761)
W/BluetoothGatt: onCharacteristicRead() - Device=00:05:5B:01:15:2D handle=24 Status=0
D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=499ee8bf-b20a-4dfb-8e47-a6ac0ed5baa1 status=0
D/StageActivity$prepareView: distance : 0
D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(179872761)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(140367725)
D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(140367725)
W/BluetoothGatt: onCharacteristicRead() - Device=00:05:5B:01:15:2D handle=24 Status=0
D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=499ee8bf-b20a-4dfb-8e47-a6ac0ed5baa1 status=0
D/StageActivity$prepareView: distance : 0
D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(140367725)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(247361697)
D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(247361697)
W/BluetoothGatt: onCharacteristicRead() - Device=00:05:5B:01:15:2D handle=24 Status=0
D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=499ee8bf-b20a-4dfb-8e47-a6ac0ed5baa1 status=0
D/StageActivity$prepareView: distance : 0
D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(247361697)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(156337045)
D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(156337045)
W/BluetoothGatt: onCharacteristicRead() - Device=00:05:5B:01:15:2D handle=24 Status=0
D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=499ee8bf-b20a-4dfb-8e47-a6ac0ed5baa1 status=0
D/StageActivity$prepareView: distance : 0

dariuszseweryn, 2017-01-03:
Hello @lovedise
If I understood you correctly: When you connect to two devices and turn off one of them - both connections are stalled until the Android OS decides that the now-offline device is no longer connected and then queued operations are being processed normally, right?
Best Regards

lovedise, 2017-01-03:
@dariuszseweryn yes, right

dariuszseweryn, 2017-01-03:
Unfortunately due to Bluetooth Stack being synchronous in meaning that it can only process one asynchronous radio operation at a time (read / write characteristic / descriptor, etc.) the library cannot continue with reading characteristic from the second device before the read from the first one will either succeed or fail. From my knowledge it is impossible to overcome this limitation.

In case you know something I do not - do not hesitate to comment / reopen this issue.

As for now - closing.

Best Regards

closed 2017-01-03 by dariuszseweryn