dariuszseweryn/RxAndroidBle #113: More intuitive use for requestMtu?

cpboyd, 2017-01-03:
### Summary
<br />
I can't imagine many use cases where a user would ONLY want to set the MTU.

As it is currently, once you've subscribed for notifications, you can't subsequently request a new MTU (as the connection is already established... performing a `flatMap()` on the connection observable results in a `BleAlreadyConnectedException`)

Below is the best way that I seem to be able to get the MTU to take effect.
Which requires a **connect-> requestMtu -> disconnect -> connect**
```java
            Observable<RxBleConnection> connectionObservable = bleManager.establishConnection(mDeviceAddress, true);
            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.LOLLIPOP) {
                connectionObservable.flatMap(rxBleConnection -> rxBleConnection.requestMtu(BtHelpers.BLE_TI_MTU))
                        .first()
                        .doAfterTerminate(this::connect)
                        .subscribe(this::handleMtu, this::handleRxErrors);
            } else {
                connect();
            }
```

Is there a way we could possibly include a MTU negotiation with a `setupNotification` so that there's a single connection?

#### Preconditions
<br />
#### Steps to reproduce actual result
1. Establish a BLE connection for a specific characteristic.
2. Cannot negotiate MTU after connection is established.
#### Actual result
<br />
Have to set MTU separately from everything else.
#### Expected result
<br />
Would be nice to negotiate MTU on an existing connection.

dariuszseweryn, 2017-01-03:
Hello @cpboyd 
I do not exactly follow your use case. You could just create a flow of data that would give you a connection with MTU being set already. You could use code similar to one below:
    Observable<RxBleConnection> establishConnectionWithMtu = bleDevice.establishConnection(context, false) // establish connection
        .flatMap(connection -> { // ... when it is established...
            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.LOLLIPOP) { // ... if it is Lollipop or later...
                return connection.requestMtu(BtHelpers.BLE_TI_MTU) // ... request MTU...
                    .map(mtu -> connection) // ... and when the MTU is set return the RxBleConnection
            } else { // ... if it is OS older than Lollipop...
                return Observable.just(connection) // ... return just the RxBleConnection
            }
        });

If I didn't get your use case please elaborate more
Best Regards

cpboyd, 2017-01-03:
@dariuszseweryn 
I didn't think to just map back to the original connection.

With this method, is the best way to report/handle the MTU via `doOnNext`?

e.g.:
```java
return connection.requestMtu(BtHelpers.BLE_TI_MTU)
    .doOnNext(i -> Log.d(TAG, "Negotiated MTU: " + i)).map(mtu -> connection);
```

closed 2017-01-03 by cpboyd

dariuszseweryn, 2017-01-04:
Everything depends on your use case. `Observable` pattern gives you a great flexibility. Your code is suitable if you want to just log the value negotiated.

Best Regards