dariuszseweryn/RxAndroidBle #114: Crash occurs while monitoring the Bluetooth cadence

kexuebiao, 2017-01-03:
### Summary
Crash occurs while monitoring the Bluetooth cadence
<br />
#### Preconditions
Android 6.0 , htc A9w mobile
<br />
#### Steps to reproduce actual result
1.Connect to a Cycling Speed and Cadence ble device
2.Notify Cycling Speed and Cadence Service
3.Turn on the power of the Bluetooth device, wait for the phone to connect the device successfully and receive the cadence data, and then turn off the device power, repeat the above operation several times,
4.And then will be observed after a connection is successful, and Crash occurs when monitor the Bluetooth cadence data
<br />
#### Actual result
<br />
D/RxBle#BluetoothGatt: onCharacteristicChanged characteristic=00002a5b-0000-1000-8000-00805f9b34fb
W/art: Suspending all threads took: 20.080ms
D/RxBle#BluetoothGatt: onCharacteristicChanged characteristic=00002a5b-0000-1000-8000-00805f9b34fb
W/BluetoothGatt: Unhandled exception in callback
rx.exceptions.OnErrorNotImplementedException: Task rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker@92b940d rejected from java.util.concurrent.ThreadPoolExecutor@db82fc2[Terminated, pool size = 0, active threads = 0, queued tasks = 0, completed tasks = 855]
                                                                        at rx.internal.util.InternalObservableUtils$ErrorNotImplementedAction.call(InternalObservableUtils.java:386)
                                                                        at rx.internal.util.InternalObservableUtils$ErrorNotImplementedAction.call(InternalObservableUtils.java:383)
                                                                        at rx.internal.util.ActionSubscriber.onError(ActionSubscriber.java:44)
                                                                        at rx.observers.SafeSubscriber._onError(SafeSubscriber.java:153)
                                                                        at rx.observers.SafeSubscriber.onError(SafeSubscriber.java:115)
                                                                        at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:60)
                                                                        at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:30)
                                                                        at rx.Observable.subscribe(Observable.java:10236)
                                                                        at rx.Observable.subscribe(Observable.java:10203)
                                                                        at rx.Observable.subscribe(Observable.java:10008)
                                                                        at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onCharacteristicChanged(RxBleGattCallback.java:132)
                                                                        at android.bluetooth.BluetoothGatt$1.onNotify(BluetoothGatt.java:438)
                                                                        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:399)
                                                                        at android.os.Binder.execTransact(Binder.java:461)
                                                                     Caused by: java.util.concurrent.RejectedExecutionException: Task rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker@92b940d rejected from java.util.concurrent.ThreadPoolExecutor@db82fc2[Terminated, pool size = 0, active threads = 0, queued tasks = 0, completed tasks = 855]
                                                                        at java.util.concurrent.ThreadPoolExecutor$AbortPolicy.rejectedExecution(ThreadPoolExecutor.java:2014)
                                                                        at java.util.concurrent.ThreadPoolExecutor.reject(ThreadPoolExecutor.java:794)
                                                                        at java.util.concurrent.ThreadPoolExecutor.execute(ThreadPoolExecutor.java:1340)
                                                                        at java.util.concurrent.Executors$DelegatedExecutorService.execute(Executors.java:584)
                                                                        at rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker.schedule(ExecutorScheduler.java:79)
                                                                        at rx.internal.operators.OperatorObserveOn$ObserveOnSubscriber.schedule(OperatorObserveOn.java:188)
                                                                        at rx.internal.operators.OperatorObserveOn$ObserveOnSubscriber$1.request(OperatorObserveOn.java:145)
                                                                        at rx.Subscriber.setProducer(Subscriber.java:209)
                                                                        at rx.Subscriber.setProducer(Subscriber.java:205)
                                                                        at rx.internal.operators.OperatorObserveOn$ObserveOnSubscriber.init(OperatorObserveOn.java:139)
                                                                        at rx.internal.operators.OperatorObserveOn.call(OperatorObserveOn.java:75)
                                                                        at rx.internal.operators.OperatorObserveOn.call(OperatorObserveOn.java:40)
                                                                        at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:44)
                                                                        at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:30) 
                                                                        at rx.Observable.subscribe(Observable.java:10236) 
                                                                        at rx.Observable.subscribe(Observable.java:10203) 
                                                                        at rx.Observable.subscribe(Observable.java:10008) 
                                                                        at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onCharacteristicChanged(RxBleGattCallback.java:132) 
                                                                        at android.bluetooth.BluetoothGatt$1.onNotify(BluetoothGatt.java:438) 
                                                                        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:399) 
                                                                        at android.os.Binder.execTransact(Binder.java:461) 

<br />
#### Expected result
<br />

dariuszseweryn, 2017-01-04:
Hello @kexuebiao 

A quick question - do you hold reference to the `RxBleClient`? It seems that the Executor powering `RxBleGattCallback` `Scheduler` is stopped and it happens when the `RxBleClientImpl` is being finalized ( https://github.com/Polidea/RxAndroidBle/blob/master/rxandroidble/src/main/java/com/polidea/rxandroidble/RxBleClientImpl.java#L97 )

kexuebiao, 2017-01-05:
Not all mobile phones will appear this problem,
In the HTC One A9 and XiaoMi Note find this problem

this is the main code i write,
https://github.com/kexuebiao/bleDemo/blob/master/app/src/main/java/cc/ble/bledemo/module/BleDevice.java

dariuszseweryn, 2017-01-05:
You have to persist `RxBleClient`. Use only one instance in the whole application. 

In your code you're creating a new instance for every `BleDevice` and the `RxBleClient` is not referenced - so it will be garbage collected and the Executor will be stopped.

kexuebiao, 2017-01-05:
Thank you reminded that I fix the code to see if i can solve this problem

closed 2017-01-13 by kexuebiao