dariuszseweryn/RxAndroidBle #119: Calling scanBleDevices() on RxBleClient uses Main thread?

kenwdelong, 2017-01-19:
### Summary
I'm finding that even if I subscribe on a different thread, RxBleClient.scanBleDevices is emitting devices on the main thread.<br />
#### Steps to reproduce actual result
Call on the main thread:
rxBleClient.scanBleDevices()
  .subscribeOn(Schedulers.io())
  .doOnNext(d -> // log the device and thread)
  .subscribe();

#### Actual result
I see the log messages coming on the main thread, although I've also seen identical log messages coming on BOTH the main thread and an io() thread.
<br />
#### Expected result
I would expect the rxBleDevice objects to be emitted on the thread I subscribed on.<br />

dariuszseweryn, 2017-01-19:
Hello @kenwdelong 
Thank you for your report. This is expected behaviour - although not in-line with what you would expect from a library using `Observable` pattern. Generally speaking Android BLE stack is behaving better (is more stable) when called from the main thread. The library is calling every BLE call from the main thread to achieve maximum stability.
Best Regards

closed 2017-01-19 by dariuszseweryn

kenwdelong, 2017-01-19:
OK, thanks for the explanation.  I was afraid I was seeing things! :-)

kenwdelong, 2017-01-20:
Does this mean that we should take care to always call RxAndroidBle on the MainThread, or is that something that you take care of, switching to MainThread inside the library if we use the wrong thread?

dariuszseweryn, 2017-01-20:
Library does it for you inside.