dariuszseweryn/RxAndroidBle #121: BleCharacteristicNotFoundException after disabling/re-enabling Bluetooth

cpboyd, 2017-01-20:
### Summary
I've noticed the BleCharacteristicNotFoundException occurring occasionally after the app is backgrounded or seemingly random times.

I've been able to reproduce this consistently by disabling Bluetooth (while connected) and then re-enabling it and connecting again.

This may be due to my limited understanding of Rx, but I can't seem to determine why the error is getting thrown.  All old subscriptions should be unsubscribed.

#### Preconditions
My setup is as follows:
(subList is a SubscriptionList that gets unsubscribed before this connect/reconnect)
```java
                connectionObservable.flatMap(
                rxBleConnection -> getNotif(rxBleConnection, notificationUuid, withAck), // subscribing for notifications
                (rxBleConnection, notificationObservable) -> {
                    Log.d(TAG, "Test...");
                    rxBleConnection.writeCharacteristic(writeUuid, "test".getBytes()).subscribe((bytes) -> Log.d(TAG, "Wrote test"), (tr) -> Log.d(TAG, "RxTest", tr));
                    // connection is established and notification prepared
                    subList.add(inputSubject.onBackpressureBuffer().flatMap(bytes -> {
                        Log.d(TAG, "Writing " + bytes.length + " bytes");
                        return rxBleConnection.writeCharacteristic(writeUuid, bytes);
                    }).subscribe());
                    return notificationObservable.doOnNext(responseBytes -> outputSubject.onNext(responseBytes)); // pass the bytes to the receiver
                }
```

Note: The "test" write does not throw an error.

#### Steps to reproduce actual result
1.  Connect to device.
2.  Send/receive data.
3.  Disable Bluetooth from Android system menu.
4.  Re-enable and re-connect.
5. Attempting to write new data will result in the BleCharacteristicNotFoundException, but the data actually gets written (and I see the proper response come back from the device.)

#### Actual result
Error is thrown, but characteristic seems to be written.
#### Expected result
Error should not be thrown.

dariuszseweryn, 2017-01-20:
Hello,
I am trying to reason about your code but it is hard as I do not see what exactly is the `connectionObservable` in your case. There is also no context about when your setup is run. When does your `inputSubject` emit? Can you confirm that the service discovery on a particular connection worked properly? (I have seen situations that the Android indicated that discovery ended but an empty list of services was returned) Your device, OS version?
Btw. it would be better to call `subList.unsubscribe();` in `.onTerminate()` of your `raw` `Subscription`

cpboyd, 2017-01-20:
@dariuszseweryn 

I've updated the code in the original post.

`connectionObservable` is an `Observable<RxBleConnection>` created by calling `bleClient.getBleDevice(macAddress).establishConnection(context, true)`
I use `inputSubject` is simply a `PublishSubject<byte[]>` and I use it with the following function:

```java
public Observable<String> sendCmd(String data) {
    Log.d(TAG, "Writing data: " + data);
    inputSubject.onNext(data.getBytes());
    return outputSubject.filter(s -> {
        // Allow for potentially corrupted JSON:
        return s.startsWith("{\"CMD\":\"" + data + '"');
    }).first();
}
```

`outputSubject` is a custom `Subject` that reassembles the individual packets into a full response.

dariuszseweryn, 2017-01-21:
But when your code from the first post is executed? It still lacks the context (state).
Point is - it looks like you have some state in your code that is wrong after turning the bluetooth off / on. I am judging this because when you say that you reinitialize everything from scratch the problem is gone and even with the exception being emitted you say that the peripheral receives the data.
I think I was not experiencing this issue when turning off / on the adapter. Will double check on Monday.

cpboyd, 2017-01-22:
This was in a singleton that's passed the context but only initializes the RxBleClient once.

Ultimately, I found that I can re-initialize everything EXCEPT the `RxBleClient` and `RxBleDevice` and it seems to get rid of the error, so I don't think there's an issue with the library.

closed 2017-01-22 by cpboyd

dariuszseweryn, 2017-01-22:
Glad to hear that you have solved your problem. Thing is that everything that is created in a context of a specific connection should be only used within the context. I will think of how to inform about it clearly.