dariuszseweryn/RxAndroidBle #122: Using the method RxBleConnection#setupNotification to open the notification is invalid

RiverZhong, 2017-01-22:
I have a device that I try to use #setupNotification to open the notification,but the Throwable in method onError is null.I use Google API to open this Characteristic notification is successful.How can I get the BluetoothGatt，I would like to use google api while using this open source library。

dariuszseweryn, 2017-01-22:
Hello @RiverZhong 
Could you paste your code, logs, anything that can be useful to debug the issue?
Currently you cannot obtain the `BluetoothGatt` from the `RxBleConnection` in other way than force casting to `RxBleConnectionImpl` and getting the `BluetoothGatt` by reflection.

dariuszseweryn, 2017-01-31:
@RiverZhong Any update?

RiverZhong, 2017-01-31:
Thank you for your help, but now is China's Spring Festival, I'm on vacation, and had no device.

RiverZhong, 2017-01-31:
Google example requires UUID corresponding descriptors, in my test equipment is empty，but using setNotification() is effective

dariuszseweryn, 2017-01-31:
My colleague @uKL is working currently on this issue. Judging on the Bluetooth Specification the Client Characteristic Config descriptor is optional but it shall be set to appropriate value to make the notifications / indications work. It is not clearly described what should be the state in both scenarios in the Specification.

Thing is that in my understanding - if a characteristic supports notifications / indications the CCC descriptor must be present. Unfortunately some device vendors think different.

We will somehow allow to setup a notification without the CCC descriptor but the user will have to specifically set it this way.

dariuszseweryn, 2017-02-02:
Version `1.2.0-SNAPSHOT` has an alternative way of setting the notifications / indications that do not need the descriptor. Check them out.

closed 2017-02-02 by dariuszseweryn

RiverZhong, 2017-02-02:
Thanks very much