dariuszseweryn/RxAndroidBle #125: Error on read.readCharacteristic

Dionysios, 2017-01-26:
### Summary
<br /> Hi! I am trying to read the value from a certain characteristic by it always return this error instead
 Read failurejava.lang.ClassCastException: rx.Observable cannot be cast to byte[].
Notifications and write operations work fine in the same code. 
#### Preconditions
<br />
#### Steps to reproduce actual result
1.
2.
3.
#### Actual result
<br />  Read failurejava.lang.ClassCastException: rx.Observable cannot be cast to byte[]
#### Expected result
<br /> Value from characteristic

dariuszseweryn, 2017-01-26:
Hello @Dionysios 
Could you share your code and provide logs? Device, OS version, version of the library used. This could be helpful.

Dionysios, 2017-01-26:
Hi! Here is the class.
`
public class ConnectionActivity  extends RxAppCompatActivity  {

    public static final String EXTRA_CHARACTERISTIC_UUID = "extra_uuid";
    private RxBleDevice mBleDevice;
    private Button mConnectToggleButton;
    private Button mReadButton;
    private Button mNotifyButton;
    private Button mWriteButton;
    private UUID characteristicUuid;
    private TextView mReadOutputView;
    private TextView mReadHexOutputView;
    private Observable<RxBleConnection> mConnectionObservable;
    private PublishSubject<Void> disconnectTriggerSubject = PublishSubject.create();
    private TextView mWriteInput;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_example4);
        characteristicUuid = (UUID) getIntent().getSerializableExtra(EXTRA_CHARACTERISTIC_UUID);
        String macAddress = getIntent().getStringExtra(DeviceActivity.EXTRA_MAC_ADDRESS);
        setTitle(getString(R.string.mac_address, macAddress));
        mBleDevice = Application.getRxBleClient(this).getBleDevice(macAddress);

        mConnectToggleButton = (Button) findViewById(R.id.connect);
        mReadButton = (Button) findViewById(R.id.read);
        mNotifyButton = (Button) findViewById(R.id.notify);
        mReadOutputView = (TextView) findViewById(R.id.read_output);
        mWriteButton = (Button) findViewById(R.id.write);
        mReadHexOutputView = (TextView) findViewById(R.id.read_hex_output);
        mWriteInput = (TextView) findViewById(R.id.write_input);

        mConnectToggleButton.setOnClickListener(this::onConnectToggleClick);
        mReadButton.setOnClickListener(this::onReadClick);
        mWriteButton.setOnClickListener(this::onWriteClick);
        mNotifyButton.setOnClickListener(this::onNotifyClick);
        mConnectionObservable = prepareConnectionObservable();
    }

    private Observable<RxBleConnection> prepareConnectionObservable() {
        return mBleDevice
                .establishConnection(this, false)
                .takeUntil(disconnectTriggerSubject)
               // .compose(bindUntilEvent(PAUSE))
                .doOnUnsubscribe(this::clearSubscription)
                .compose(new ConnectionSharingAdapter());
    }

    public void onConnectToggleClick(View view) {
        if (isConnected()) {
            Log.i("test", "we are here is " + isConnected());
            triggerDisconnect();
        } else {
            Log.i("test", "we are here new connection");
                mConnectionObservable.subscribe(rxBleConnection -> {
                    Log.d(getClass().getSimpleName(), "Hey, connection has been established!");
                    Log.d("test", "Hey, connection has been established!");
                    //updateUI();
                    runOnUiThread(this::updateUI);
                }, this::onConnectionFailure);

        }
    }

    public void onReadClick(View view) {
        Log.d("test", "Calling read function");
        if (isConnected()) {
            mConnectionObservable
[... 91 lines cut ...]
        Snackbar.make(findViewById(R.id.main), "Notifications error: " + throwable, Snackbar.LENGTH_SHORT).show();
    }

    private void onWriteSuccess() {
        //noinspection ConstantConditions
        Snackbar.make(findViewById(R.id.main), "Write success", Snackbar.LENGTH_SHORT).show();
    }

    private void onWriteFailure(Throwable throwable) {
        //noinspection ConstantConditions
        Snackbar.make(findViewById(R.id.main), "Write error: " + throwable, Snackbar.LENGTH_SHORT).show();
    }

    private byte[] getInputBytes() {
        return HexString.hexToBytes(mWriteInput.getText().toString());
    }
}
`
and the log also filtered by tag test
01-26 18:03:07.506 6061-6061/com.semcon.ble.myapplicationble I/test: we are here new connection
01-26 18:03:07.907 6061-6197/com.semcon.ble.myapplicationble D/test: Hey, connection has been established!
01-26 18:03:09.217 6061-6061/com.semcon.ble.myapplicationble D/test: Calling read function
01-26 18:03:09.355 6061-6061/com.semcon.ble.myapplicationble D/test: Read failurejava.lang.ClassCastException: rx.Observable cannot be cast to byte[]
01-26 18:03:12.143 6061-6061/com.semcon.ble.myapplicationble D/test: Calling read function
01-26 18:03:12.229 6061-6061/com.semcon.ble.myapplicationble D/test: Read failurejava.lang.ClassCastException: rx.Observable cannot be cast to byte[]

dariuszseweryn, 2017-01-26:
Which version of the library do you use? Could you provide a stacktrace for the `ClassCastException`?

dariuszseweryn, 2017-01-26:
I think that you may facing an issue with Retrolambda which is mixing some references -> you get a `ClassCastException`. Otherwise the compiler would whine.

Dionysios, 2017-01-26:
Library version should be 
`<manifest xmlns:android="http://schemas.android.com/apk/res/android"

    package="com.polidea.rxandroidble"

    android:versionCode="1"

    android:versionName="1.1.0-SNAPSHOT" >`

Get stacktrace does not return a lot

`01-26 18:33:34.176 20278-20278/com.semcon.ble.myapplicationble D/test: Calling read function
01-26 18:33:39.422 20278-20278/com.semcon.ble.myapplicationble D/test: Read failure[Ljava.lang.StackTraceElement;@3a01228
01-26 18:33:39.423 20278-20278/com.semcon.ble.myapplicationble D/test: Read failurejava.lang.ClassCastException: rx.Observable cannot be cast to byte[]`

Should I try with another version of the library?

dariuszseweryn, 2017-01-26:
you could try with `1.1.0` and expanding all lambdas / method references to anonymous classes.

dariuszseweryn, 2017-01-27:
@Dionysios Any news?

Dionysios, 2017-01-30:
Hi! I managed to solve it by a number of things.
Update android build tools from 2.1.3 -> 2.2.3
import the library straigh from gradle -> compile "com.polidea.rxandroidble:rxandroidble:1.1.0"
add the retrolambda plugin as you have it on the installation instructions
and finally and most important remove  jackOptions I had before

dariuszseweryn, 2017-01-30:
Yeah. Unfortunately the library is not compatible with Jack at this moment. We plan to migrate to Kotlin or just get rid of lambdas as it makes a great compatibility problem.

The problem is you have to use either our library or Jack. :/ Anyways - I am glad that you have it running.

RobLewis, 2017-01-30:
Wait, am I reading this correctly? 
Though I have Java experience, I'm new to Android, Bluetooth, and Reactive programming. I went to a good deal of trouble figuring out how to set up Jack and use lambdas in Android Studio. Now you are telling me that I can't do this with your library? 
If this is the case, wouldn't it have been a good idea to mention this in your documentation? 

If retrolambda is an option, I didn't see any instructions on using it. Could you point me at them? What is the downside? And am I supposed to remove this from my app's build.gradle: 
       ` jackOptions { enabled true }`
How about this? 
`compileOptions { sourceCompatibility JavaVersion.VERSION_1_8
            targetCompatibility JavaVersion.VERSION_1_8 }`

Anything you can give me in the way of detailed and specific instructions on how to recover from what looks like a small disaster would be VERY MUCH appreciated. I've been compiling 1.8 code using RxAndroidBle and haven't noticed any special problems, though I have yet to actually try running anything except some simple JUnit tests with RxBleClientMock.

dariuszseweryn, 2017-01-30:
I am sorry. This was an unintended consequence of using Retrolambda with the library. Unfortunately Jack does not support lambdas in library projects. Will definitely add that at the top of the `Readme.md`

If you only use Jack for the lambda expressions then changing to Retrolambda is quite easy - https://hackernoon.com/living-android-without-kotlin-db7391a2b170#.9s6aieeki - here you have a paragraph about lambdas and the difference between Jack and Retrolambda setup. I hope this will help you.

The problems with the compatibility are surfacing in the runtime when some generated lambdas are not matching what is expected.

dariuszseweryn, 2017-02-02:
Version `1.2.0-SNAPSHOT` should be now compatible with Jack.

closed 2017-02-02 by dariuszseweryn

RobLewis, 2017-03-31:
Did you see Google's recent [announcement](https://android-developers.googleblog.com/2017/03/future-of-java-8-language-feature.html) that Jack is being deprecated and Java 8 features will soon be added to the base system? Will this have any effect on this issue?

dariuszseweryn, 2017-04-01:
Yes - we have seen it. The library does not use lambdas anymore so we are unaffected by that change unless we decide to go back using lambdas. Thanks for info though. Best Regards.