dariuszseweryn/RxAndroidBle #127: Filter devices with ManufacturerData

Dionysios, 2017-02-01:
### Summary
<br /> Hi, I am not sure if I missed somehow by the ScanBleDevices can only accept UUID as filter.
Is it possible to scan with other filters eg manufacturer data?
Thanks!
#### Preconditions
<br />   public abstract Observable<RxBleScanResult> scanBleDevices(@Nullable UUID... filterServiceUUIDs);
#### Steps to reproduce actual result
1.
2.
3.
#### Actual result
<br />
#### Expected result
<br />

uKL, 2017-02-01:
Currently there is no such feature. Feel free to contribute :)

dariuszseweryn, 2017-02-01:
Of course it is possible to filter the devices with the manufacturer data. The thing is you have to do it manually as the library has no built-in feature. ;) This state will probably change once the Lollipop scanning API will be supported. You can still contribute to it! #74 Best Regards.

dariuszseweryn, 2017-05-22:
Just added the new scanning API to `1.3.0-SNAPSHOT`. Feel free to check it out.

closed 2017-05-22 by dariuszseweryn