dariuszseweryn/RxAndroidBle #128: Device not found in scan if Service UUID is specified

RobLewis, 2017-02-01:
### Summary
`.scanBleDevices( )` finds mocked device but `.scanBleDevices( asServiceIdArray )` doesn't<br />
#### Preconditions
Set up RxBleDeviceMock with a Service UUID `asServiceId`. <br />
#### Steps to reproduce actual result
1. call `.scanBleDevices( asServiceIdArray )` (where `asServiceIdArray` is a 1-element array of the defined Service UUID `asServiceId`) 
2. subscribe to `Observable<RxBleScanResult>` and view results
3.
#### Actual result
No items are emitted<br />
#### Expected result
A single `RxBleScanResult` for the mocked device. Note that if `.scanBleDevices( )` is called with no arguments, the device is found. The same null result is obtained if the `asServiceIdArray` is replaced with `asServiceId` (i.e., not an array)<br />

dariuszseweryn, 2017-02-01:
Hello @RobLewis 
Thanks for the report - could you share a minimal code that is showing the issue? Thanks!

RobLewis, 2017-02-02:
These excerpts demonstrate what I'm seeing: 

```
static final String AS_UUID_HEAD  =  "F954"; 
static final String AS_UUID_TAIL  =  "-91B3-BD9A-F077-80F2A6E57D00";
static final String asServiceIdString    =  AS_UUID_HEAD + "1234" + AS_UUID_TAIL; // Service UUID
static final UUID asServiceId = UUID.fromString( asServiceIdString ); 
static final UUID[] asServiceIdArray = { asServiceId }; // varargs argument to scanBleDevices

// In the Mock Client Builder:
rxBleClientMock = new RxBleClientMock.Builder( )
                .addDevice( new RxBleClientMock.DeviceBuilder( )
                        .deviceMacAddress( macAddress )
                        .deviceName( deviceName )
                        .scanRecord( scanRecordBytes )
                        .rssi( rssiValue )
                        .addService(
                                asServiceId,
                                new RxBleClientMock.CharacteristicsBuilder( ) 
                                        .addCharacteristic( 
                                                asDataCharId, asDataCharBytes,
                                                new RxBleClientMock.DescriptorsBuilder() 
                                                        .addDescriptor(clientCharConfigID, asDisableNotifications) 
                                                        .build() // the Descriptor  ...  etc.

// in the code
  Observable<RxBleDevice> scanForAeroscope() { 
        System.out.println( "Entering scanForAS" );
        return rxBleClientMock
                .scanBleDevices( asServiceIdArray ) // returns Observable<RxBleScanResult> 
                //.scanBleDevices(  ) // returns Observable<RxBleScanResult>
```

Note that the last, commented-out line works to find and return the scan result, while I get nothing back from the call with the asServiceIdArray parameter. And I get the same result if I use just the UUID alone, instead of putting it into a 1-element array.

mzgreen, 2017-02-03:
@RobLewis Are you subscribing to the `Observable<RxBleDevice>` returned from `scanForAS()` method? I wasn't able to reproduce the issue and I don't see a call to `subscribe()` in your sample code. Calling `scanBleDevices` with or without parameters doesn't do anything except creating an `Observable`. You have to subscribe to it in order to perform a scan operation.

RobLewis, 2017-02-03:
Sorry, yes, I am subscribing. Here's the code that calls `scanForAS()`: 
```
        scanSubscription = scanForAS() 
                .doOnSubscribe( () -> System.out.println( "doOnSubscribe in connect_OK") ) // Prints, so must be subscribing OK
                .doOnNext( item -> System.out.println("doOnNext in Test connect_OK item: " + item.toString() ) ) // Doesn't print!
                .doOnUnsubscribe( () -> System.out.println( "doOnUnsubscribe in connect_OK") ) 
                .subscribe( 
                        this::connectToAS, // argument is RxBleDevice
                        this::handleScanError );
```

As I mentioned, it does work to return the Mocked device if I leave out the argument in the call to `scanBleDevices()`

mzgreen, 2017-02-03:
@RobLewis please make sure that all of the services that you're searching for are added to `RxBleClientMock`. If your `RxBleClientMock` doesn't have all of them then it won't return anything. Android BLE API works the same way.

RobLewis, 2017-02-03:
Here's my complete call to the Mock Builder: 

```
        rxBleClientMock = new RxBleClientMock.Builder( )
                .addDevice( new RxBleClientMock.DeviceBuilder( )
                        .deviceMacAddress( macAddress )
                        .deviceName( deviceName )
                        .scanRecord( scanRecordBytes )
                        .rssi( rssiValue )
                        .addService( 
                                asServiceId, 
                                new RxBleClientMock.CharacteristicsBuilder( ) 
                                        .addCharacteristic( 
                                                asDataCharId, asDataCharBytes,
                                                new RxBleClientMock.DescriptorsBuilder() 
                                                        .addDescriptor(clientCharConfigID, asDisableNotifications)
                                                        .build() 
                                        ) 
                                        .addCharacteristic(
                                                asInputCharId, asInputCharBytes,
                                                new ArrayList<BluetoothGattDescriptor>( 0 ) ) // (empty List)
                                        .addCharacteristic(
                                                asStateCharId, asStateCharBytes,
                                                new ArrayList<BluetoothGattDescriptor>( 0 ) ) 
                                        .addCharacteristic(
                                                asOutputCharId, asOutputCharBytes,
                                                new RxBleClientMock.DescriptorsBuilder() // add Descriptor
                                                        .addDescriptor(clientCharConfigID, asDisableNotifications) 
                                                        .build() // the Descriptor
                                        ) 
                                        .build( ) // Characteristics
                        ) // addService (the only one)
                        .build( ) // Device
                ) // addDevice
                .build(); // Client
```

One thing I haven't been able to find: what is the content and format of a ScanRecord? Right now I'm just using a dummy byte[3] for `scanRecordBytes` in DeviceBuilder. 

In any case, `asServiceId` is the only Service UUID I have defined in the Mock Client, and the only one I'm looking for. This same constant is used to build the Mock Client and to filter the `scanBleDevices()` call. I've tried using it directly and putting it into a single-item array. Same result: not found.

RobLewis, 2017-02-13:
Update: when I switch from the mock client to a real one, the Service UUID filter in `scanBleDevices( serviceUUID )` works correctly.

dariuszseweryn, 2017-08-30:
@RobLewis Have you encountered the problem when writing unit tests or while using the code on the actual device?

RobLewis, 2017-08-30:
It has always worked fine with a real device. I can't really answer about unit tests because I haven't looked at them in quite a while and this project is pretty much done. The code excerpts I posted earlier should tell the story.

dariuszseweryn, 2017-09-19:
I think that it is related to #262 and in tests should work if used with `Robolectric`. The new API #239 that we work on should make it possible to use the mock library in the tests without `Robolectric`. Closing.

closed 2017-09-19 by dariuszseweryn