dariuszseweryn/RxAndroidBle #129: BleGattException status 133 & 129

prasanthmurugan, 2017-02-04:
Case :
=>When the the beacon or other device is active before i launch my application the connection is successfully i can do my process but when i disconnect and reconnect the beacon or other device following exception is thrown and then the i can't connect again. After relaunching the application it works perfectly as before.

Following is the exception i face while connect and discover services using RxAndroidBle project

**ERROR:**
BleGattException{macAddress=E8:8D:28:97:E2:98, status=133 (85),bleGattOperationType=BleGattOperation{description='CONNECTION_STATE'}}

Kovartthan, 2017-02-04:
Yes, I also having this issue .....if anyone know how to fix it!!!!

Prethiviraj, 2017-02-04:
i  am having issue on 129  Discovery service  ,it does't constant service discovery sometimes read write faster but sometime  fires the error ...

dariuszseweryn, 2017-02-06:
Hello all. 
Could you share the code that you are using? How many devices at once you try to connect? What phone / OS? etc.
On some devices / oses the BLE stack is not very stable and such errors (133 / 129) are happening more or less randomly. These errors come from the Android `BluetoothGattCallback` and have little to do with the library.
Best Regards

Kovartthan, 2017-02-07:
I was trying to connect the beacon with **Samsung S3 (Android 4.3 )**  . I am getting success for **read/ write operation** but sometimes it throws the **BleGatt Exception Status 133** at some point but i need result constantly . I need to know What is that exception ? Why it is occurring . I am agree with your above point and it s absolutely true **Mr . dariuszseweryn** .

dariuszseweryn, 2017-02-07:
If all of you have the same problem on the same device / os - I unfortunately cannot help you. This is already an outdated phone / system. And Galaxy S3 with Android 4.3 is particularly buggy in terms of BLE. On newer devices random BLE errors happen but S3 + 4.3 they happen a lot more. I have incorporated in the library all my knowledge of how to stabilise the stack. :(

Kovartthan, 2017-02-07:
Above 5.0 android devices also sometimes i got that error .. :(  @dariuszseweryn

dariuszseweryn, 2017-02-07:
Yup these sometimes happen. I have seen these errors even on 7.0. As you can see here: https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h status 133 is `GATT_ERROR` and 129 is `GATT_INTERNAL_ERROR` which are quite vague and do not tell a lot. 
Keep in mind that BLE operates on the same frequency as the WiFi. It is a wireless communication and it may suffer from interference.
I have no more ideas of how to stabilise the system more than it is now. I am closing. If anyone here will get some more information on how to get Android 4.3 (especially on SGS3) to work better - feel free to comment.

closed 2017-02-07 by dariuszseweryn