dariuszseweryn/RxAndroidBle #130: Indication subscription does not release connection when unsubscribed

fabiorodella, 2017-02-06:
### Summary
I have a Bluetooth scale which transmits values through indication, and only stays on while it's taking/sending measurements. To make sure I get any measurements, whenever the scale disconnects I start a "pending" notification through the following code:
```java
Subscription subscription = prepareDeviceConnection(device)
                .observeOn(AndroidSchedulers.mainThread())
                .flatMap(rxBleConnection -> {
                    return rxBleConnection.setupIndication(characteristic.getUuid());
                })
                .doOnNext(notificationObservable -> {
                    Log.d(TAG, "Observation started for characteristic " + characteristic.getUuid());
                })
                .doOnUnsubscribe(() -> {
                    Log.d(TAG, "Unsubscribing observation with ID " + subId);
                    connectedDevice.removeSubscription(subId);
                })
                .flatMap(notificationObservable -> notificationObservable)
                .subscribe(
                        ...
                );

Log.d(TAG, "Subscribing observation with ID " + subId);
connectedDevice.addSubscription(subId, subscription);
```

where the `prepareDeviceConnection` method establishes a connection like this:
```java
Observable<RxBleConnection> connectionObservable = connectedDevice.getConnectionObservable();
if (connectionObservable == null) {
    connectionObservable = connectedDevice.getBleDevice().establishConnection(this, true)
            .flatMap(rxBleConnection -> {
                // If the service implementation requires that device is bonded first, do it
                if (requiresBonding) {
                    return establishBond(rxBleConnection, connectedDevice.getBleDevice());
                } else {
                    return Observable.just(rxBleConnection);
                }
            })
            .doOnUnsubscribe(() -> {
                Log.d(TAG, "Unsubscribing connection observable for " + device.getDisplayName());
                connectedDevice.setConnectionObservable(null);
            })
            .compose(new ConnectionSharingAdapter());

    Log.d(TAG, "Creating connection observable for " + device.getDisplayName());
    connectedDevice.setConnectionObservable(connectionObservable);
}
```

This works perfectly so that when the scale becomes visible, the connection is established and the values come through via the indication subscription.

The problem though is that the user can opt to forget the scale after this has been setup. In this case, I just call `unsubscribe` in the indication subscription created earlier, and even though I can see that the the `doOnUnsubscribe` from the first code block is invoked correctly, the `doOnUnsubscribe` for the connection observable is *not* called. If I step on the scale again I can see the `Observation started for characteristic...` message in logcat.

I expected the connection to be released after the call to `unsubscribe`, and in fact this is exactly what happens if I use a write operation instead of indication. Were my assumption and/or code wrong?

#### Preconditions
As described above

#### Steps to reproduce actual result
1. Start a connection and indication subscription while a device is not available.
2. Unsubscribe from the subscription created in 1 while the device is not available.
#### Actual result
Connection request stays pending
#### Expected result
Connection request is cancelled

dariuszseweryn, 2017-02-06:
Hello,
Could you add the logs from your application?
Could you add a log for `doOnSubscribe` for both `prepareDeviceConnection()` and the first block of code? It would be easier to debug what is happening.

fabiorodella, 2017-02-06:
Sure thing, here you go (the new `doOnSubscribe` logs start with `Actually subscribing...`):

This next block is the "pairing process" where I write to a characteristic in the scale and it turns off immediately after:
```
D/BluetoothService: Subscribing connection state with ID bfaf7229-b764-4351-b861-df2be160d2c7
D/BluetoothService: Creating connection observable for A&D_UC-352BLE_459997
D/BluetoothService: Actually subscribing connection observable for A&D_UC-352BLE_459997
D/BluetoothService: Connection state changed for A&D_UC-352BLE_459997: RxBleConnectionState{CONNECTING}
D/BluetoothService: Subscribing write with ID d2407256-a1a5-4036-bf67-3ee43822e7c9
D/BluetoothGatt: connect() - device: 6C:EC:EB:45:99:97, auto: true
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=25f1b2fb-e8b8-4ec0-b033-3f3eaa6051d2
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=6C:EC:EB:45:99:97
D/BluetoothService: Connection state changed for A&D_UC-352BLE_459997: RxBleConnectionState{CONNECTED}
D/BluetoothService: Device is alread bonded
D/BluetoothGatt: discoverServices() - device: 6C:EC:EB:45:99:97
D/BluetoothGatt: onSearchComplete() = Device=6C:EC:EB:45:99:97 Status=0
D/BluetoothService: Unsubscribing connection observable for A&D_UC-352BLE_459997
D/BluetoothService: Connection state changed for A&D_UC-352BLE_459997: RxBleConnectionState{DISCONNECTED}
D/BluetoothService: Unsubscribing write with ID d2407256-a1a5-4036-bf67-3ee43822e7c9
D/BluetoothManager: getConnectionState()
D/BluetoothManager: getConnectedDevices
D/BluetoothGatt: cancelOpen() - device: 6C:EC:EB:45:99:97
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=6C:EC:EB:45:99:97
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=6
```

Once I detect the device disconnected, I start the pending indication:
```
D/BluetoothService: Starting pending connection for A&D_UC-352BLE_459997
D/BluetoothService: Creating connection observable for A&D_UC-352BLE_459997
D/BluetoothService: Actually subscribing connection observable for A&D_UC-352BLE_459997
D/BluetoothService: Connection state changed for A&D_UC-352BLE_459997: RxBleConnectionState{CONNECTING}
D/BluetoothService: Actually subscribing observation with ID 6C:EC:EB:45:99:97/23434100-1FE4-1EFF-80CB-00FF78297D8B/23434101-1fe4-1eff-80cb-00ff78297d8b
D/BluetoothService: Subscribing observation with ID 6C:EC:EB:45:99:97/23434100-1FE4-1EFF-80CB-00FF78297D8B/23434101-1fe4-1eff-80cb-00ff78297d8b
D/BluetoothGatt: connect() - device: 6C:EC:EB:45:99:97, auto: true
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=0f490655-fd02-4152-bb4f-deb0eabfc12f
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
```

Here is when the user manually forgets the scale
```
D/BluetoothService: Unsubscribing observation with ID 6C:EC:EB:45:99:97/23434100-1FE4-1EFF-80CB-00FF78297D8B/23434101-1fe4-1eff-80cb-00ff78297d8b
D/BluetoothService: Unsubscribing connection state with ID bfaf7229-b764-4351-b861-df2be160d2c7
```

And here is what happens when the scale becomes visible again
```
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=6C:EC:EB:45:99:97
D/BluetoothService: Device is alread bonded
D/BluetoothGatt: discoverServices() - device: 6C:EC:EB:45:99:97
D/BluetoothGatt: onSearchComplete() = Device=6C:EC:EB:45:99:97 Status=0
D/BluetoothGatt: setCharacteristicNotification() - uuid: 23434101-1fe4-1eff-80cb-00ff78297d8b enable: true
D/BluetoothService: Observation started for characteristic 23434101-1fe4-1eff-80cb-00ff78297d8b
```

Let me know if I can help with anything else. Thanks!

dariuszseweryn, 2017-02-06:
You have only shown the cut parts of logs.
Are you sure that once the user forgets the scale you do not resubscribe to the indications observable? I see that the `doOnUnsubscribe` is called properly so the underlying connections should be disposed. Never had a problem with that.
Unfortunately there seems to be something more about managing the state in your application that is missing. With the provided code and logs I cannot pinpoint the issue.
I would need to ask you for a minimal code sample that is reproducing the issue.

fabiorodella, 2017-02-06:
This isn't clear because I left out the timestamps, but that is actually the whole log, I just split it into parts to be able to explain what each section is related to. And I'm pretty sure I'm not resubscribing anywhere, since I added log statements to every `doOnSubscribe` in my code.

One thing that looks odd to me is that the log from the `doOnSubscribe` block (`Actually subscribing observation with ID...`) is showing up before the log for when the whole chain is created (`Subscribing observation with ID...`), although RxJava is not really something I'm very familiar so this could be normal.

I'll see if I can isolate a minimal code sample.

fabiorodella, 2017-02-06:
@dariuszseweryn after some more careful debugging I found that there was indeed a indication subscription still active. Sorry for the false alarm.

closed 2017-02-06 by fabiorodella

Premier-Solutions, 2021-04-12:
We are developing a mobile app to get ble data from different personal medical devices and am currently working on A&D 352 Weight Scale.

There is the same custom service with uuid "23434100-1fe4-1eff-80cb-00ff78297d8b" that is mentioned in this post.
And contains a notifiable characteristic with uuid "23434101-1fe4-1eff-80cb-00ff78297d8b".

Can any one guide me how to parse the returned data.  That is some thing like (0x) 01-B0-02.

Thanks.

fabiorodella, 2021-04-12:
@Premier-Solutions take a look at https://github.com/andengineering/AndroidSampleCode. It's not the easiest code in the world to follow, but it did help me when I was integrating with this scale.

Premier-Solutions, 2021-04-12:
> @Premier-Solutions take a look at https://github.com/andengineering/AndroidSampleCode. It's not the easiest code in the world to follow, but it did help me when I was integrating with this scale.

Thanks a lot fabiorodella.
That was so helpful. I was almost giving up on that.

For any one else stuck in this:

I was getting "01-B0-02" for 68.8 pounds.

First byte 01 is a multiplication factor.
Data is in second and third bytes. B0-02

Converting Hex to Binary:

B0 = 10110000
02 = 10

Combining in little endian format
1010110000

Converting to decimal
1010110000 = 688

Multiplying with multiplication factor 0.1

688 * 0.1  = 68.8