dariuszseweryn/RxAndroidBle #131: Simple connect and Read problem

larbirahmani, 2017-02-07:
### Summary
<br />
I'm trying to Read and connect to a BLE device (intel curie arduino) that has 2 characteristics, I followed the guide and the exemples, but I always get an error of java.lang.IllegalStateException: Fatal Exception thrown on Scheduler.Worker thread.
I think I'm missing something...
#### Preconditions
<br />
#### Steps to reproduce actual result

public class Kara extends AppCompatActivity {

    private static final int REQUEST_ENABLE_BT = 1;

    private Subscription subscription;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_kmov__historique);

        TextView caractest,constate;

        constate = (TextView) findViewById(R.id.connstate);

        caractest = (TextView) findViewById(R.id.caratctest);

        RxBleClient rxBleClient = RxBleClient.create(this);

        String macAddress = "98:4F:EE:0D:17:0D";

        String cara1 ="19B10011-E8F2-537E-4F6C-D104768A1214";
        UUID mycara1 = UUID.fromString(cara1);
        String cara2 ="19B10012-E8F2-537E-4F6C-D104768A1214";
        UUID mycara2 = UUID.fromString(cara2);

        final RxBleDevice device = rxBleClient.getBleDevice(macAddress);

        // permission and activation of blutooth and location
        BluetoothAdapter mBluetoothAdapter;
        // Use this check to determine whether BLE is supported on the device.  Then you can
        // selectively disable BLE-related features.
        if (!getPackageManager().hasSystemFeature(PackageManager.FEATURE_BLUETOOTH_LE)) {
            Toast.makeText(this, R.string.ble_not_supported, Toast.LENGTH_SHORT).show();
            finish();
        }
        // Initializes a Bluetooth adapter.  For API level 18 and above, get a reference to
        // BluetoothAdapter through BluetoothManager.
        final BluetoothManager bluetoothManager =
                (BluetoothManager) getSystemService(Context.BLUETOOTH_SERVICE);
        mBluetoothAdapter = bluetoothManager.getAdapter();
        // Checks if Bluetooth is supported on the device.
        if (mBluetoothAdapter == null) {
            Toast.makeText(this, R.string.error_bluetooth_not_supported, Toast.LENGTH_SHORT).show();
            finish();
            return;
        }

        // Ensures Bluetooth is enabled on the device.  If Bluetooth is not currently enabled,
        // fire an intent to display a dialog asking the user to grant permission to enable it.
        if (!mBluetoothAdapter.isEnabled()) {
            if (!mBluetoothAdapter.isEnabled()) {
                Intent enableBtIntent = new Intent(BluetoothAdapter.ACTION_REQUEST_ENABLE);
                startActivityForResult(enableBtIntent, REQUEST_ENABLE_BT);
            }
        }

        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M) {
            if (checkSelfPermission(android.Manifest.permission.ACCESS_COARSE_LOCATION) != PackageManager.PERMISSION_GRANTED
                    && checkSelfPermission(android.Manifest.permission.ACCESS_FINE_LOCATION) != PackageManager.PERMISSION_GRANTED) {
                String[] permissions = new String[]{android.Manifest.permission.ACCESS_COARSE_LOCATION, android.Manifest.permission.ACCESS_FINE_LOCATION};
                ActivityCompat.requestPermissions(this, permissions, 10);
[... 96 lines cut ...]
                      at rx.internal.operators.OnSubscribeMap$MapSubscriber.onError(OnSubscribeMap.java:88)
                      at rx.observers.Subscribers$5.onError(Subscribers.java:230)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.reportError(OperatorMerge.java:266)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.checkTerminate(OperatorMerge.java:818)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.emitLoop(OperatorMerge.java:579)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.emit(OperatorMerge.java:568)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.onError(OperatorMerge.java:276)
                      at rx.internal.operators.OnSubscribeMap$MapSubscriber.onError(OnSubscribeMap.java:88)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.reportError(OperatorMerge.java:266)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.checkTerminate(OperatorMerge.java:818)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.emitLoop(OperatorMerge.java:579)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.emit(OperatorMerge.java:568)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.onError(OperatorMerge.java:276)
                      at rx.internal.operators.OnSubscribeMap$MapSubscriber.onError(OnSubscribeMap.java:88)
                      at rx.exceptions.Exceptions.throwOrReport(Exceptions.java:216)
                      at rx.internal.operators.OnSubscribeFromCallable.call(OnSubscribeFromCallable.java:50)
                      at rx.internal.operators.OnSubscribeFromCallable.call(OnSubscribeFromCallable.java:33)
                      at rx.Observable.unsafeSubscribe(Observable.java:10140)
                      at rx.internal.operators.OnSubscribeMap.call(OnSubscribeMap.java:48)
                      at rx.internal.operators.OnSubscribeMap.call(OnSubscribeMap.java:33)
                      at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:48)
                  	at rx.internal.operators.OnSubscribeLift.call(OnSubscribe
#### Expected result
<br />
Read targeted characteristics and view them in a text view.

dariuszseweryn, 2017-02-07:
The error is truncated. I would need the full stacktrace to help you.

larbirahmani, 2017-02-07:
Here it is, thank you 

02/07 18:27:35: Launching app
$ adb push C:\Users\Slash\Desktop\KMOV\app\build\outputs\apk\app-debug.apk /data/local/tmp/com.example.kmov.kmov
$ adb shell pm install -r "/data/local/tmp/com.example.kmov.kmov"
	pkg: /data/local/tmp/com.example.kmov.kmov
Success

$ adb shell am start -n "com.example.kmov.kmov/com.example.kmov.kmov.LoginActivity" -a android.intent.action.MAIN -c android.intent.category.LAUNCHER
Connected to process 1950 on device htc-0pja2-HT59CSV00879
W/System: ClassLoader referenced unknown path: /data/app/com.example.kmov.kmov-1/lib/arm64
I/InstantRun: Instant Run Runtime started. Android package is com.example.kmov.kmov, real application class is com.example.kmov.kmov.app.AppController.
I/art: override thread count:-j3
W/System: ClassLoader referenced unknown path: /data/app/com.example.kmov.kmov-1/lib/arm64
I/FirebaseInitProvider: FirebaseApp initialization unsuccessful
W/art: Before Android 4.1, method android.graphics.PorterDuffColorFilter android.support.graphics.drawable.VectorDrawableCompat.updateTintFilter(android.graphics.PorterDuffColorFilter, android.content.res.ColorStateList, android.graphics.PorterDuff$Mode) would have incorrectly overridden the package-private method in android.graphics.drawable.Drawable
D/SQLiteHandler: Fetching user from Sqlite: {email=a, nom=a, created_at=2017-02-06 16:56:15, uid=58989c9faedc66.17771738}
I/Adreno: QUALCOMM build                   : e8f6654, I8f3b63409d
          Build Date                       : 12/16/15
          OpenGL ES Shader Compiler Version: XE031.06.00.02
          Local Branch                     : 
          Remote Branch                    : refs/tags/AU_LINUX_ANDROID_LA.BF64.1.2.2_RB4.06.00.00.180.011
          Remote Branch                    : NONE
          Reconstruct Branch               : NOTHING
W/ample.kmov.kmov: type=1400 audit(0.0:46422780): avc: denied { getopt } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:zygote:s0 tclass=unix_dgram_socket permissive=0
D/SQLiteHandler: Fetching clé from Sqlite: {updated_at=2017-02-07 12:24:12, batterie=50, email=98:4F:EE:0D:17:0D, created_at=2017-02-07 12:24:12, etat=f, nom=98:4F:EE:0D:17:0D}
W/OpenGLRenderer: Fail to change FontRenderer cache size, it already initialized
W/ample.kmov.kmov: type=1400 audit(0.0:46423251): avc: denied { getopt } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:zygote:s0 tclass=unix_dgram_socket permissive=0
W/OpenGLRenderer: Fail to change FontRenderer cache size, it already initialized
W/ample.kmov.kmov: type=1400 audit(0.0:46423253): avc: denied { getopt } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:zygote:s0 tclass=unix_dgram_socket permissive=0
W/OpenGLRenderer: Fail to change FontRenderer cache size, it already initialized
W/OpenGLRenderer: Fail to change FontRenderer cache size, it already initialized
D/BluetoothGatt: connect() - device: 98:4F:EE:0D:17:0D, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=1138438a-41f4-49c6-9a1c-cfadf21a5087
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=98:4F:EE:0D:17:0D
D/BluetoothGatt: discoverServices() - device: 98:4F:EE:0D:17:0D
D/BluetoothGatt: onSearchComplete() = Device=98:4F:EE:0D:17:0D Status=0
E/AndroidRuntime: FATAL EXCEPTION: pool-2-thread-1
                  Process: com.example.kmov.kmov, PID: 1950
                  java.lang.IllegalStateException: Fatal Exception thrown on Scheduler.Worker thread.
                      at rx.internal.schedulers.ScheduledAction.run(ScheduledAction.java:59)
                      at rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker.run(ExecutorScheduler.java:107)
                      at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                      at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                      at java.lang.Thread.run(Thread.java:818)
                   Caused by: rx.exceptions.OnErrorFailedException: Error occurred when trying to propagate error to Observer.onError
                      at rx.observers.SafeSubscriber._onError(SafeSubscriber.java:187)
                      at rx.observers.SafeSubscriber.onError(SafeSubscriber.java:115)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.reportError(OperatorMerge.java:266)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.checkTerminate(OperatorMerge.java:818)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.emitLoop(OperatorMerge.java:579)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.emit(OperatorMerge.java:568)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.onError(OperatorMerge.java:276)
                      at rx.internal.operators.OnSubscribeMap$MapSubscriber.onError(OnSubscribeMap.java:88)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.reportError(OperatorMerge.java:266)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.checkTerminate(OperatorMerge.java:818)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.emitLoop(OperatorMerge.java:579)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.emit(OperatorMerge.java:568)
[... cut ...]
             at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:48)
             at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:30)
             at rx.Observable.unsafeSubscribe(Observable.java:10140)
         	a
D/Error: ERR: TOTAL BYTES WRITTEN: 49432
D/Process: killProcess, pid=1950
D/Process: com.android.internal.os.RuntimeInit$UncaughtHandler.uncaughtException:113 java.lang.ThreadGroup.uncaughtException:693 java.lang.ThreadGroup.uncaughtException:690 
Application terminated.

dariuszseweryn, 2017-02-07:
Still does not look full. Could you keep the `RxBleClient` as a property in your `Activity`?

larbirahmani, 2017-02-07:
02/07 20:15:36: Launching app
$ adb push C:\Users\Slash\Desktop\KMOV\app\build\outputs\apk\app-debug.apk /data/local/tmp/com.example.kmov.kmov
$ adb shell pm install -r "/data/local/tmp/com.example.kmov.kmov"
	pkg: /data/local/tmp/com.example.kmov.kmov
Success

$ adb shell am start -n "com.example.kmov.kmov/com.example.kmov.kmov.LoginActivity" -a android.intent.action.MAIN -c android.intent.category.LAUNCHER
Connected to process 16763 on device htc-0pja2-HT59CSV00879
W/System: ClassLoader referenced unknown path: /data/app/com.example.kmov.kmov-2/lib/arm64
I/InstantRun: Instant Run Runtime started. Android package is com.example.kmov.kmov, real application class is com.example.kmov.kmov.app.AppController.
I/art: override thread count:-j3
W/System: ClassLoader referenced unknown path: /data/app/com.example.kmov.kmov-2/lib/arm64
I/FirebaseInitProvider: FirebaseApp initialization unsuccessful
W/art: Before Android 4.1, method android.graphics.PorterDuffColorFilter android.support.graphics.drawable.VectorDrawableCompat.updateTintFilter(android.graphics.PorterDuffColorFilter, android.content.res.ColorStateList, android.graphics.PorterDuff$Mode) would have incorrectly overridden the package-private method in android.graphics.drawable.Drawable
D/SQLiteHandler: Fetching user from Sqlite: {email=a, nom=a, created_at=2017-02-06 16:56:15, uid=58989c9faedc66.17771738}
I/Adreno: QUALCOMM build                   : e8f6654, I8f3b63409d
          Build Date                       : 12/16/15
          OpenGL ES Shader Compiler Version: XE031.06.00.02
          Local Branch                     : 
          Remote Branch                    : refs/tags/AU_LINUX_ANDROID_LA.BF64.1.2.2_RB4.06.00.00.180.011
          Remote Branch                    : NONE
          Reconstruct Branch               : NOTHING
W/ample.kmov.kmov: type=1400 audit(0.0:46731200): avc: denied { getopt } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:zygote:s0 tclass=unix_dgram_socket permissive=0
D/SQLiteHandler: Fetching clé from Sqlite: {updated_at=2017-02-07 12:24:12, batterie=50, email=98:4F:EE:0D:17:0D, created_at=2017-02-07 12:24:12, etat=f, nom=98:4F:EE:0D:17:0D}
W/OpenGLRenderer: Fail to change FontRenderer cache size, it already initialized
W/ample.kmov.kmov: type=1400 audit(0.0:46733020): avc: denied { getopt } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:zygote:s0 tclass=unix_dgram_socket permissive=0
W/OpenGLRenderer: Fail to change FontRenderer cache size, it already initialized
W/OpenGLRenderer: Fail to change FontRenderer cache size, it already initialized
W/ample.kmov.kmov: type=1400 audit(0.0:46740570): avc: denied { getopt } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:zygote:s0 tclass=unix_dgram_socket permissive=0
W/ample.kmov.kmov: type=1400 audit(0.0:46740941): avc: denied { getopt } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:zygote:s0 tclass=unix_dgram_socket permissive=0
I/art: Background partial concurrent mark sweep GC freed 19600(4MB) AllocSpace objects, 63(1788KB) LOS objects, 40% free, 17MB/28MB, paused 5.349ms total 57.069ms
W/OpenGLRenderer: Fail to change FontRenderer cache size, it already initialized
D/BluetoothGatt: connect() - device: 98:4F:EE:0D:17:0D, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=aed15e11-f471-4c13-b0fb-3e5a8c4e9832
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=98:4F:EE:0D:17:0D
D/BluetoothGatt: discoverServices() - device: 98:4F:EE:0D:17:0D
D/BluetoothGatt: onSearchComplete() = Device=98:4F:EE:0D:17:0D Status=0
E/AndroidRuntime: FATAL EXCEPTION: pool-2-thread-1
                  Process: com.example.kmov.kmov, PID: 16763
                  java.lang.IllegalStateException: Fatal Exception thrown on Scheduler.Worker thread.
                      at rx.internal.schedulers.ScheduledAction.run(ScheduledAction.java:59)
                      at rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker.run(ExecutorScheduler.java:107)
                      at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                      at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                      at java.lang.Thread.run(Thread.java:818)
                   Caused by: rx.exceptions.OnErrorFailedException: Error occurred when trying to propagate error to Observer.onError
                      at rx.observers.SafeSubscriber._onError(SafeSubscriber.java:187)
                      at rx.observers.SafeSubscriber.onError(SafeSubscriber.java:115)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.reportError(OperatorMerge.java:266)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.checkTerminate(OperatorMerge.java:818)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.emitLoop(OperatorMerge.java:579)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.emit(OperatorMerge.java:568)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.onError(OperatorMerge.java:276)
                      at rx.internal.operators.OnSubscribeMap$MapSubscriber.onError(OnSubscribeMap.java:88)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.reportError(OperatorMerge.java:266)
[... cut ...]
             at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:48)
             at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:30)
             at rx.Observable.unsafeSubscribe(Observable.java:10140)
         	a
D/Error: ERR: TOTAL BYTES WRITTEN: 49432
D/Process: killProcess, pid=16763
D/Process: com.android.internal.os.RuntimeInit$UncaughtHandler.uncaughtException:113 java.lang.ThreadGroup.uncaughtException:693 java.lang.ThreadGroup.uncaughtException:690 
Application terminated.

dariuszseweryn, 2017-02-07:
Yeah, I forgot that you have not caught the error to print the log by yourself.
Could you keep the reference to the result of `RxBleClient.create(Context)` in your `Activity` the same way you keep the `private Subscription subscription;`? Because now the `RxBleClient` is only referenced in the `onCreate()` method / scope and it needs to be persisted.

larbirahmani, 2017-02-07:
Sorry I don't understand how to do this.
I have this activity that connects to the device (I have the MAC) and reads 2 characteristics.
no matter where I declare the Rxbleclient I get the same error.

dariuszseweryn, 2017-02-07:
```
private static final int REQUEST_ENABLE_BT = 1;

private RxBleClient rxBleClient;

private Subscription subscription;

@Override
protected void onCreate(Bundle savedInstanceState) {
    super.onCreate(savedInstanceState);
    setContentView(R.layout.activity_kmov__historique);
    
    rxBleClient = RxBleClient.create(this);
    ...
}
```
To get the place where the error is thrown you would need to put a brakepoint in Android Studio on throwing exceptions and check where and exactly what is the cause of the exception.

larbirahmani, 2017-02-07:
Ok, I've done that, and I get a similar error:

02/07 20:56:48: Launching app
Cold swapped changes.
$ adb shell am start -n "com.example.kmov.kmov/com.example.kmov.kmov.LoginActivity" -a android.intent.action.MAIN -c android.intent.category.LAUNCHER -D
Waiting for application to come online: com.example.kmov.kmov.test | com.example.kmov.kmov
Connecting to com.example.kmov.kmov
W/ActivityThread: Application com.example.kmov.kmov is waiting for the debugger on port 8100...
I/System.out: Sending WAIT chunk
I/art: Debugger is active
Connected to the target VM, address: 'localhost:8602', transport: 'socket'
I/System.out: Debugger has connected
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
W/ample.kmov.kmov: type=1400 audit(0.0:47584337): avc: denied { getopt } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:zygote:s0 tclass=unix_dgram_socket permissive=0
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: debugger has settled (1423)
W/System: ClassLoader referenced unknown path: /data/app/com.example.kmov.kmov-2/lib/arm64
I/InstantRun: Instant Run Runtime started. Android package is com.example.kmov.kmov, real application class is com.example.kmov.kmov.app.AppController.
I/art: override thread count:-j2
W/System: ClassLoader referenced unknown path: /data/app/com.example.kmov.kmov-2/lib/arm64
I/FirebaseInitProvider: FirebaseApp initialization unsuccessful
W/ample.kmov.kmov: type=1400 audit(0.0:47585082): avc: denied { getopt } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:zygote:s0 tclass=unix_dgram_socket permissive=0
W/art: Before Android 4.1, method android.graphics.PorterDuffColorFilter android.support.graphics.drawable.VectorDrawableCompat.updateTintFilter(android.graphics.PorterDuffColorFilter, android.content.res.ColorStateList, android.graphics.PorterDuff$Mode) would have incorrectly overridden the package-private method in android.graphics.drawable.Drawable
D/SQLiteHandler: Fetching user from Sqlite: {email=a, nom=a, created_at=2017-02-06 16:56:15, uid=58989c9faedc66.17771738}
I/Adreno: QUALCOMM build                   : e8f6654, I8f3b63409d
          Build Date                       : 12/16/15
          OpenGL ES Shader Compiler Version: XE031.06.00.02
          Local Branch                     : 
          Remote Branch                    : refs/tags/AU_LINUX_ANDROID_LA.BF64.1.2.2_RB4.06.00.00.180.011
          Remote Branch                    : NONE
          Reconstruct Branch               : NOTHING
D/SQLiteHandler: Fetching clé from Sqlite: {updated_at=2017-02-07 12:24:12, batterie=50, email=98:4F:EE:0D:17:0D, created_at=2017-02-07 12:24:12, etat=f, nom=98:4F:EE:0D:17:0D}
W/OpenGLRenderer: Fail to change FontRenderer cache size, it already initialized
W/OpenGLRenderer: Fail to change FontRenderer cache size, it already initialized
D/BluetoothGatt: connect() - device: 98:4F:EE:0D:17:0D, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=8d03516d-7c57-4c07-9c91-e2fe55f0d8fb
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=98:4F:EE:0D:17:0D
D/BluetoothGatt: discoverServices() - device: 98:4F:EE:0D:17:0D
W/ample.kmov.kmov: type=1400 audit(0.0:47592131): avc: denied { getopt } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:zygote:s0 tclass=unix_dgram_socket permissive=0
D/BluetoothGatt: onSearchComplete() = Device=98:4F:EE:0D:17:0D Status=0
E/AndroidRuntime: FATAL EXCEPTION: pool-1-thread-1
                  Process: com.example.kmov.kmov, PID: 10328
                  java.lang.IllegalStateException: Fatal Exception thrown on Scheduler.Worker thread.
                      at rx.internal.schedulers.ScheduledAction.run(ScheduledAction.java:59)
                      at rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker.run(ExecutorScheduler.java:107)
                      at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                      at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                      at java.lang.Thread.run(Thread.java:818)
                   Caused by: rx.exceptions.OnErrorFailedException: Error occurred when trying to propagate error to Observer.onError
                      at rx.observers.SafeSubscriber._onError(SafeSubscriber.java:187)
                      at rx.observers.SafeSubscriber.onError(SafeSubscriber.java:115)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.reportError(OperatorMerge.java:266)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.checkTerminate(OperatorMerge.java:818)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.emitLoop(OperatorMerge.java:579)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.emit(OperatorMerge.java:568)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.onError(OperatorMerge.java:276)
                      at rx.internal.operators.OnSubscribeMap$MapSubscriber.onError(OnSubscribeMap.java:88)
                      at rx.internal.operators.OperatorMerge$MergeSubscriber.reportError(OperatorMerge.java:266)
[... cut ...]
             at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:48)
             at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:30)
             at rx.Observable.unsafeSubscribe(Observable.java:10140)
         	a
D/Error: ERR: TOTAL BYTES WRITTEN: 49432
D/Process: killProcess, pid=10328
D/Process: com.android.internal.os.RuntimeInit$UncaughtHandler.uncaughtException:113 java.lang.ThreadGroup.uncaughtException:693 java.lang.ThreadGroup.uncaughtException:690 
Disconnected from the target VM, address: 'localhost:8602', transport: 'socket'

dariuszseweryn, 2017-02-07:
http://stackoverflow.com/questions/24718958/android-studio-ide-break-on-exception

larbirahmani, 2017-02-07:
It works now ! thank you but the reading I'm getting is strange type, how can I cast it to a string?

dariuszseweryn, 2017-02-07:
Could you elaborate more on what you have changed that made your code work? What is the strange type you are talking about?

larbirahmani, 2017-02-07:
I think it was the                 .observeOn(AndroidSchedulers.mainThread())
because it wasn't in my code.
and the strange type is                                 caractest.setText(aBytes.toString());
I get a code instead of the value of the characteristic

dariuszseweryn, 2017-02-07:
True, I did not noticed it missing. 
`aBytes` is a `byte[]`. So `byte[].toString()` will give you some odd result
Try `new String(bytes, "UTF-8");`

larbirahmani, 2017-02-07:
Yes it worked with it but it gives me only the first character, here's the code:
        subscription = device.establishConnection(this, false)
                .flatMap(rxBleConnection -> Observable.combineLatest( // use the same connection and combine latest emissions
                        rxBleConnection.readCharacteristic(mycara1), // sometimes IDE get's lost in what type is returned from an Observable - that's why I added <byte[]>
                        rxBleConnection.readCharacteristic(mycara2),
                        Pair::new // merge into a Pair
                ))
                .observeOn(AndroidSchedulers.mainThread())
                .subscribe(
                        byteArrayPair -> {
                            // here you get the latest values from notifications
                            byte[] aBytes = byteArrayPair.first;
                            byte[] bBytes = byteArrayPair.second;
                            // do your thing
                            try {
                                caractest.setText(new String(aBytes, "UTF-8"));
                                constate.setText(new String(bBytes, "UTF-8"));
                            } catch (UnsupportedEncodingException e) {
                                e.printStackTrace();
                            }

                            constate.setText(bBytes.toString());

                        },
                        throwable -> {
                            Toast.makeText(this,throwable.toString(), Toast.LENGTH_LONG).show();
                        }
                );

larbirahmani, 2017-02-08:
Yes, I'm getting only one byte, I search around and tried many tricks but non can do the work, any ideas would be very appreciated, thank you.

dariuszseweryn, 2017-02-08:
I do not know what peripheral you are talking to. If the periperal provides only one byte then you get only one byte. No tricks here. This has nothing to do with the library.

closed 2017-02-09 by dariuszseweryn