dariuszseweryn/RxAndroidBle #133: RxAndroidBle Service ThreadPoolExcution reject issue

cuihujun, 2017-02-09:
[Error.txt](https://github.com/Polidea/RxAndroidBle/files/762989/Error.txt)
### Summary
<br />
not sure why I am getting these errors, I try to implement BleService and when I call the addNotification() after 30s I will get the threadpoll reject error.
PS. the logs are attached
#### Preconditions
<br />
#### Steps to reproduce actual result
Source code:

public class BleService extends Service {
//...
    mConnectionObservable = prepareConnectionObservable();
    private Observable prepareConnectionObservable() {
        return mRxBleDevice
                .establishConnection(this, false)
                .takeUntil(disconnectTriggerSubject)
                .doOnUnsubscribe(this::clearSubscription)
                .compose(new ConnectionSharingAdapter());
    }
// ...

    mConnectionObservable
            .doOnSubscribe(new Action0() {
        @Override
        public void call() {
            Log.w(TAG, "call: start connect");
        }
    })
            .doOnNext(rxBleConnection -> Log.w(TAG, "call: Connct Sucess"))
            .doOnNext(rxBleConnection -> Log.i(TAG, "call: start find service"))
            .compose(timeoutJustFirstEmit(20, TimeUnit.SECONDS))
            .flatMap(RxBleConnection::discoverServices)
    .doOnNext(this::onDiscoverServices)
    .retryWhen(errors -> errors.flatMap((Func1<Throwable, Observable<?>>) error -> Observable.timer(5, TimeUnit.SECONDS)))
            .subscribe(rxBleConnection -> Log.i(TAG, "success"), throwable -> Log.e(TAG, "failed" + throwable.getMessage(), throwable));
//...

    public void addNotification(String servUuid, String chaUuid) {
        if (isConnected()) {
            mConnectionObservable
                    .flatMap(rxBleConnection -> rxBleConnection.setupNotification(charactoristicUuid))
                    .doOnNext(observable -> notificationHasBeenSetUp(charactoristicUuid, observable))
                    .flatMap(notificationObservable -> notificationObservable)
                    .subscribe(bytes -> {
                        onNotificationReceived(serviceUuid, charactoristicUuid, bytes);
                    }, this::onNotificationSetupFailure);
        }
    }
// ...
}
#### Actual result
<br />
#### Expected result
<br />

dariuszseweryn, 2017-02-09:
Hello,
What version of the library do you use? How do you persist the instance of `RxBleClient` so it will not be garbage collected during you use the library?

cuihujun, 2017-02-09:
hello,

thanks for the quick response

the one I used is 1.1.0.

I just got the instance form RxBleClient.create()

private RxBleDevice mRxBleDevice = null;

mRxBleDevice = RxBleClient.create(mContext).getBleDevice(macAddress);

compile "com.polidea.rxandroidble:rxandroidble:1.1.0"
compile 'com.trello:rxlifecycle:1.0'
compile 'com.trello:rxlifecycle-components:1.0'

dariuszseweryn, 2017-02-10:
In the above code you do not keep a reference to the `RxBleClient` after you create the `RxBleDevice`.
You have to keep the reference to `RxBleClient` at least as long as you plan to use anything that is obtained from it (`RxBleClient`, `RxBleConnection`, etc...)

dariuszseweryn, 2017-02-20:
Any news? Has my last comment helped in your case?

cuihujun, 2017-02-20:
thanks for checking,

I need to do BLE demo in short term, so I changed to Android BLE built-in
lib.
I will check it later.

dariuszseweryn, 2017-02-22:
Ok. Closing for now - feel free to reopen if the problem will persist.

closed 2017-02-22 by dariuszseweryn