dariuszseweryn/RxAndroidBle #134: .map() function appears to break BLE device scanning

RobLewis, 2017-02-10:
I have been trying for days to simply scan for BLE devices, without success until I made one small change. 

This code works: 

    scanSubscription = asBleClient
        .scanBleDevices( ) // returns Observable<RxBleScanResult>
        .observeOn( AndroidSchedulers.mainThread( ) )
        .doOnNext( this::addRxBleScanResult )
        .subscribe( scanResultSubscriber );

While this does not: 

    scanSubscription = asBleClient
        .scanBleDevices( )// returns Observable<RxBleScanResult>
        .observeOn( AndroidSchedulers.mainThread( ) )
        .map( scanResult -> scanResult.getBleDevice() )
        .doOnNext( this::addRxBleDevice )
        .subscribe( scanDeviceSubscriber );

#### Actual result
The version using `.map()` never finds a device. Note that the method `addRxBleScanResult` calls the same `.getBleDevice()` method as `.map()` does, but it works. In fact, `addRxBleScanResult` simply calls `.getBleDevice`, and passes the result to `addRxBleDevice`. 
<br />
#### Expected result
Both bits of code should essentially do the same thing. 
<br />

dariuszseweryn, 2017-02-13:
Hello there!
I am afraid that you may have a bug in your code or you found a bug in `RxJava` (which is unlikely). The `Observable` do not interact with each other in terms of changing their behaviour. Although I do not immediately see what may be the problem in your situation - I also do not see a reason why the issue could be in `.map() function appears to break BLE device scanning`.
Best Regards

dariuszseweryn, 2017-02-13:
@RobLewis maybe you could provide a minimal sample project that would be a showcase for the issue? (this and another one for #128 if possible)

RobLewis, 2017-02-13:
I don't know that I will have time to create a separate project, but let me add this: the complete code for the `addRxBleScanResult` method: 

```
public void addRxBleScanResult( RxBleScanResult scanResult ) {
        RxBleDevice device = scanResult.getBleDevice();
        Log.d( LOG_TAG, "addRxBleScanResult about to call addRxBleDevice with device " + device.toString() );
        addRxBleDevice( device );
    }
```
As you can see, all it does is convert the `scanResult` into a `device`, and call the same `addRxBleDevice` method that the other version of the code uses to add the device to a Vector. There is no other difference. If the problem isn't in `.map()`, then I don't know where it could be. 

As noted, I puzzled for days trying to make the original version work. As soon as I eliminated the call to `.map()`, I started getting results.

dariuszseweryn, 2017-02-13:
Do you test with a real `RxBleClient` or with a mock?

RobLewis, 2017-02-13:
This is with a real device (HP tablet running Android 5.0.1). It is scanning for prototypes of a new hardware sensor peripheral. 

P.S. On a separate topic, the peripheral needs to send data frames of up to 4K bytes. Am I correct that the new Long Read option will be useful for this? I haven't looked closely at it yet.

dariuszseweryn, 2017-02-13:
Unfortunately - I cannot reproduce your issue - The `.map()` operator has nothing to do with scanning.
You could add a `doOnNext()` after each transformation starting at `scanBleDevices()` and log what is comming to debug. 

There is no Long Read - it is Long Write. So I doubt it will be useful for you.

RobLewis, 2017-02-13:
Thanks. I will let you know if I discover anything.

dariuszseweryn, 2017-02-22:
Since there is no way that `Observable`'s along the chain can interact with each other - it could be a problem with `RxJava` but I highly doubt it. This code works perfectly for me:

```
        scanSubscription = rxBleClient
                .scanBleDevices( ) // returns Observable<RxBleScanResult>
                .observeOn( AndroidSchedulers.mainThread( ) )
                .doOnNext( this::addRxBleScanResult )
                .subscribe( scanResultSubscriber );
```
Gives result:
```
02-22 17:39:06.166 7741-7741/com.polidea.rxandroidble.sample D/RxBle#Radio:   QUEUED RxBleRadioOperationScan(1114574560)
02-22 17:39:06.166 7741-7755/com.polidea.rxandroidble.sample D/RxBle#Radio:  STARTED RxBleRadioOperationScan(1114574560)
02-22 17:39:06.181 7741-7741/com.polidea.rxandroidble.sample D/BluetoothAdapter: startLeScan(): null
02-22 17:39:06.291 7741-7752/com.polidea.rxandroidble.sample D/BluetoothAdapter: onClientRegistered() - status=0 clientIf=4
02-22 17:39:06.296 7741-7755/com.polidea.rxandroidble.sample D/RxBle#Radio: FINISHED RxBleRadioOperationScan(1114574560)
02-22 17:39:06.361 7741-7753/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=33:05:AD:46:10:08 RSSI=-89
02-22 17:39:06.381 7741-7741/com.polidea.rxandroidble.sample D/addRxBleScanResult: com.polidea.rxandroidble.RxBleScanResult@4273ff20
02-22 17:39:06.391 7741-7752/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=65:4E:45:EF:E1:C5 RSSI=-61
02-22 17:39:06.391 7741-7741/com.polidea.rxandroidble.sample D/addRxBleDevice: RxBleDeviceImpl{bluetoothDevice=null(33:05:AD:46:10:08)}
02-22 17:39:06.391 7741-7753/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=33:05:AD:46:10:08 RSSI=-88
02-22 17:39:06.391 7741-7741/com.polidea.rxandroidble.sample D/RESULT SUB: onNext: com.polidea.rxandroidble.RxBleScanResult@4273ff20
02-22 17:39:06.396 7741-7741/com.polidea.rxandroidble.sample D/addRxBleScanResult: com.polidea.rxandroidble.RxBleScanResult@427407c0
02-22 17:39:06.401 7741-7741/com.polidea.rxandroidble.sample D/addRxBleDevice: RxBleDeviceImpl{bluetoothDevice=null(65:4E:45:EF:E1:C5)}
02-22 17:39:06.401 7741-7741/com.polidea.rxandroidble.sample D/RESULT SUB: onNext: com.polidea.rxandroidble.RxBleScanResult@427407c0
02-22 17:39:06.401 7741-7741/com.polidea.rxandroidble.sample D/addRxBleScanResult: com.polidea.rxandroidble.RxBleScanResult@427411d8
02-22 17:39:06.401 7741-7741/com.polidea.rxandroidble.sample D/addRxBleDevice: RxBleDeviceImpl{bluetoothDevice=null(33:05:AD:46:10:08)}
02-22 17:39:06.401 7741-7741/com.polidea.rxandroidble.sample D/RESULT SUB: onNext: com.polidea.rxandroidble.RxBleScanResult@427411d8
02-22 17:39:06.421 7741-7752/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=33:05:AD:46:10:08 RSSI=-91
02-22 17:39:06.421 7741-7741/com.polidea.rxandroidble.sample D/addRxBleScanResult: com.polidea.rxandroidble.RxBleScanResult@42741f50
02-22 17:39:06.426 7741-7741/com.polidea.rxandroidble.sample D/addRxBleDevice: RxBleDeviceImpl{bluetoothDevice=null(33:05:AD:46:10:08)}
02-22 17:39:06.426 7741-7741/com.polidea.rxandroidble.sample D/RESULT SUB: onNext: com.polidea.rxandroidble.RxBleScanResult@42741f50
02-22 17:39:06.441 7741-7753/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=98:5A:EB:8A:3A:A9 RSSI=-63
02-22 17:39:06.446 7741-7741/com.polidea.rxandroidble.sample D/addRxBleScanResult: com.polidea.rxandroidble.RxBleScanResult@42742a38
(...)
```
And the second one using:
```
scanSubscription = rxBleClient
                .scanBleDevices( )// returns Observable<RxBleScanResult>
                .observeOn( AndroidSchedulers.mainThread( ) )
                .map( scanResult -> scanResult.getBleDevice() )
                .doOnNext( this::addRxBleDevice )
                .subscribe( scanDeviceSubscriber );
```
Gives result:
```
02-22 17:41:07.731 9993-9993/com.polidea.rxandroidble.sample D/RxBle#Radio:   QUEUED RxBleRadioOperationScan(1114599352)
02-22 17:41:07.731 9993-10025/com.polidea.rxandroidble.sample D/RxBle#Radio:  STARTED RxBleRadioOperationScan(1114599352)
02-22 17:41:07.746 9993-9993/com.polidea.rxandroidble.sample D/BluetoothAdapter: startLeScan(): null
02-22 17:41:07.816 9993-10006/com.polidea.rxandroidble.sample D/BluetoothAdapter: onClientRegistered() - status=0 clientIf=4
02-22 17:41:07.826 9993-10025/com.polidea.rxandroidble.sample D/RxBle#Radio: FINISHED RxBleRadioOperationScan(1114599352)
02-22 17:41:07.841 9993-10005/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=33:05:AD:46:10:08 RSSI=-87
02-22 17:41:07.856 9993-10061/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=4F:34:3A:32:A0:1F RSSI=-72
02-22 17:41:07.861 9993-10006/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=60:43:07:68:43:A6 RSSI=-62
02-22 17:41:07.866 9993-9993/com.polidea.rxandroidble.sample D/addRxBleDevice: RxBleDeviceImpl{bluetoothDevice=null(33:05:AD:46:10:08)}
02-22 17:41:07.866 9993-9993/com.polidea.rxandroidble.sample D/DEVICE SUB: onNext: RxBleDeviceImpl{bluetoothDevice=null(33:05:AD:46:10:08)}
02-22 17:41:07.866 9993-9993/com.polidea.rxandroidble.sample D/addRxBleDevice: RxBleDeviceImpl{bluetoothDevice=null(4F:34:3A:32:A0:1F)}
02-22 17:41:07.871 9993-9993/com.polidea.rxandroidble.sample D/DEVICE SUB: onNext: RxBleDeviceImpl{bluetoothDevice=null(4F:34:3A:32:A0:1F)}
[... cut ...]
        @Override
        public void onCompleted() {
            Log.d("DEVICE SUB", "onCompleted");
        }

        @Override
        public void onError(Throwable e) {
            Log.d("DEVICE SUB", "onError: " + e.toString());
        }

        @Override
        public void onNext(RxBleDevice device) {
            Log.d("DEVICE SUB", "onNext: " + device.toString());
        }
    };
```
Test performed on Samsung Galaxy S3 with Android 4.3.
Closing - if you will have more info - feel free to add.

closed 2017-02-22 by dariuszseweryn