dariuszseweryn/RxAndroidBle #138: Can RxBleConnection be persisted?

RobLewis, 2017-02-16:
A documentation question: 
In your code snippets, most of the BLE operations begin with: 
`device.establishConnection(context, false)`

Must the RxBleConnection be reestablished before every operation, or can it be saved in a variable and reused for multiple operations? Does a single connection serve all operations on a given device, or is a different connection required for accessing different BLE characteristics? For reading vs. writing? Is there any harm or excessive delay caused by calling `.establishConnection()` before each operation? What's the recommended practice? Clarification appreciated!

dariuszseweryn, 2017-02-20:
Q: `Must the RxBleConnection be reestablished before every operation(...)?`
A: `RxBleConnection` must be established before any operation like read / write characteristic. That is why API for these operations is on the `RxBleConnection` interface. A `RxBleConnection` is an abstraction made for a single _connection_ to a particular device (i.e. the time between the device connected and disconnected - wether it was by user unsubscribing from `RxBleDevice.establishConnection()` or by a connection error)

Q: `Can it be saved in a variable and reused for multiple operations?`
A: Yes, as long as the connection is valid [between the device is connected and disconnected (see above]. After that the `RxBleConnection` is useless [because the connection has already ended)] but it is not a recommended approach as it introduces state to manage.

Q: `Does a single connection serve all operations on a given device, or is a different connection required for accessing different BLE characteristics?`
A: A single connection can serve all operations on a given device as long as the connection is valid.

Q: `Is there any harm or excessive delay caused by calling .establishConnection() before each operation?`
A: Establishing a BLE connection is a rather long process and consumes power on both the phone and the peripheral. 

Q: `What's the recommended practice?`
A: Use a single connection as long as it feasible to your use-case. Create a single `Observable` flow  that fits your requirements. You can check this for inspiration: http://stackoverflow.com/questions/40667038/emit-an-item-one-at-a-time-interact-with-it-until-a-condition-is-met-then-cont

RobLewis, 2017-02-21:
Thanks very much! I'm still a bit confused by one thing: 
You say that it's not recommended to save the RxBleConnection in a variable because it introduces state to manage. Yet you also say to use a single connection as long as feasible because establishing a new one takes time and battery power. How do I reuse the connection without saving it somewhere? I need to interactively read and write several characteristics of the BLE device, and receive notifications from it. 
I was thinking of a method like `refreshConnection()` that would test the connection state and if necessary reconnect. It would be called before each operation. Am I on the wrong track?

RobLewis, 2017-02-22:
Here's what I have in mind. Any comments? 

```
RxBleDevice bleDevice; 
volatile RxBleConnection bleConnection;
AtomicBoolean writeCommandInProgress;
...
private Observable<RxBleConnection> getConnectionObservable() {
   if( bleDevice.getConnectionState() == RxBleConnection.RxBleConnectionState.CONNECTED ) { 
       return Observable.just( bleConnection );
   } else return bleDevice.establishConnection( asBleServiceRef, true );
}
...
public boolean writeCommand( byte[] commandBytes ) {
    if( writeCommandInProgress.compareAndSet( false, true ) ) { 
        inputSubscription = getConnectionObservable()
            .doOnNext( rxBleConnection -> bleConnection = rxBleConnection ) // save the connection
            .flatMap( rxBleConnection -> rxBleConnection.writeCharacteristic( asInputCharId, commandBytes ) )
            .subscribe( inputSubscriber );  // onCompleted() resets writeCommandInProgress
        return true;                        // command initiated
    } else return false;                    // can't issue the command because a previous one hasn't completed
}
```
Actually, this reminded me of another question: is it necessary to wait for the completion of a previous write operation before calling `writeCharacteristic()` again, or are these operations automatically queued for processing?

dariuszseweryn, 2017-02-22:
```
device.establishConnection(context, false)
                                .flatMap { connection ->
                                    Completable.concat(
                                            doSomething(connection), // Completable created here may be arbitratily long
                                            setPairingData(connection)
                                    )
                                            .andThen(Observable.just(connection))
                                }
                                .take(1)
                                .subscribe()
```
In the above example the connection is being established and after that first what will happen is the `Completable doSomething(RxBleConnection connection)` and after it `Completable setPairingData(RxBleConnection connection` will be called. Where both `Completable`s may perform basically anything. And you are still sure that the connection will not be released until both will complete.

It is safe not to wait for the completion - operations are queued internally so two operations will not be executed at the same time.

RobLewis, 2017-02-22:
Thank you for that; I will investigate. But someone with the handle "s_noopy" answered my recent StackOverflow question about `Singles` in RxAndroidBle by saying that support for them and `Completables` wouldn't be available until your 2.0 release. See [http://stackoverflow.com/questions/42280071/would-it-be-appropriate-to-use-singles-for-rxandroidble-reads-and-writes](url) Is this wrong? 
Would still appreciate your comment on my proposed approach of saving the connection and reusing it as long as the status remains CONNECTED. 
And did you have something specific in mind for `setPairingData()`? Can I get more info about this?

RobLewis, 2017-02-22:
It occurs to me I may have a misunderstanding about BLE. Is a connection only desired at the moment that there is data to be exchanged? In my use case I sometimes have a constant stream of data packets coming from a peripheral device.

dariuszseweryn, 2017-02-22:
`s_noopy` is my SO handle. 

`Single`s and `Completable`s are not supported in the API of the library as the change would need a major version bump for it. But you can still use them in your part of the application.

Your proposal should work but it is introducing state which may lead hard to find errors (such as race conditions).

`setPairingData()` could also be named `doYourSecondThing()`. The point is that everything that is using the connection is also plugged into `Observable` chain - so it is safe to create a single flow of data.

`Is a connection only desired at the moment that there is data to be exchanged?` - false - you as a developer decide how long you want to persist the connection

RobLewis, 2017-02-23:
Your example is very clever—thanks! 
One final question: `doSomething(connection)` and `setPairingData(connection)` look like simple method calls, but the docs say `concat()` arguments are Completables. Wouldn't I build each one with something like `Completable.fromCallable( connection -> ??? )`. But Callables don't accept parameters, so how do I pass the connection to them? Do I perhaps want an implementation of Completable.OnSubscribe? Could it be as simple as 
`new Completable( connection -> { <code using the connection> } )`

dariuszseweryn, 2017-02-23:
First of all you can use it the same way as with any other invocations:
```
        final Object object = new Object();
        
        new Runnable() {
            @Override
            public void run() {
                object.doSomething();
            }
        };
```
`Runnable` is not accepting any parameters - yet you can use references from the outside.
And remember that `Observable` has a method `.toCompletable()`. So it is basically up to you how you will wire-up the flow using `Observable`s, `Single`s or `Completable`s.

closed 2017-02-24 by dariuszseweryn