dariuszseweryn/RxAndroidBle #142: RxAndroidBle on Android Wear

jcamins, 2017-02-26:
### Summary

When I use the example scanning code from the website, I get a LOCATION_SERVICES_DISABLED error, even though when I go into the developer options, location says it is turned on. However, per the documentation (https://developer.android.com/training/articles/wear-location-detection.html#Connect), regular location services should not be used on Android Wear devices, and there does not seem to be any way to disable RxAndroidBle's location services check.

jcamins, 2017-02-26:
I confirmed that I could make the scan work by simply commenting out the location services check, but that doesn't really seem like something that should be upstreamed.

uKL, 2017-02-27:
Hey! Thanks for trying the library on Android Wear. Frankly speaking, we didn't test in on the Wear. Validators you are mentioning are strictly for Android. We will talk about it internally how we can resolve the problem.

If you have any ideas please don't hesitate to mention them.

dariuszseweryn, 2017-03-01:
Hello @jcamins 
I think you that you have mixed the Location options available. The location needed (or supposed) to be on is under normal Settings - not under the Developer Options.
The application does not really use the Location Services - it only checks wether they are on.
Why commenting out the check has helped is described here #106 but for now we cannot be certain wether this check is needed or not for a particular device - so it is always on as this may lead to tracking app user's location. 
If you will have an idea how to solve the problem without using whitelisting / blacklisting - feel free to speak up.
Best Regards

jcamins, 2017-03-01:
@DariuszAniszewski the only location option I can find on Android Wear is under Developer Options. Which makes sense, since Android Wear doesn't support location services at all.

So I guess the fact that you closed this means that you're officially not supporting Android Wear? That's a shame.

dariuszseweryn, 2017-03-01:
I am @dariuszseweryn ;)
I am developing on Sony SmartWatch 3 - quite old but that is the only one I have at the moment. When I go to the Settings - just under Bluetooth switch - I see Location - this one has to be on.
Seems to work on Android Wear. I will check with other watches but I do not expect a different behaviour. Feel free to provide more information if you think that something is wrong.

dariuszseweryn, 2017-03-01:
@jcamins You could add what watch do you use so I could check what is your case.

jcamins, 2017-03-01:
@dariuszseweryn sorry about that. GitHub autocompleted for me, apparently to the wrong person. My Asus ZenWatch 2 does not have a Location option in the main settings screen, and the one in the Developer Options doesn't seem to do anything (and doesn't seem like a reasonable thing to ask users to find, anyway). Your watch would seem to contradict the documentation, which says that location services are not available on Android Wear. Do you have Android Wear 2.0? Maybe support for location services is changing with AW 2.0.

dariuszseweryn, 2017-03-01:
`Sony SmartWatch 3` is definitely Android Wear 1.0 (API 23)
I am trying to get access `LG Watch Urbane` - there is a chance to grab one tomorrow.
I have also checked the emulator with API 23 - it indeed does not have the `Location` setting - need to dig deeper.

jcamins, 2017-03-02:
How do I check the API version? I have Android Wear 1.5, but I don't know how to check API version.

jcamins, 2017-03-02:
Also, does your Sony SmartWatch 3 require location services? Maybe we could just check whether the user is on Wear when deciding whether or not to throw if the services are not enabled?

dariuszseweryn, 2017-03-02:
If it is not visible in the About Settings page - you can connect your watch to the computer and the Android Studio lists the current API level - then compare with wikipedia Android Wear topic.

`Sony SmartWatch 3` does not require location services to be turned on to get BLE scan results though the situation is the same as in #106 and I did not found any official guides for BLE scanning on Android Wear yet.

jcamins, 2017-03-02:
Ah, right. It wasn't in the settings, but according to Android Studio, ZenWatch 2 is also API 23.

dariuszseweryn, 2017-03-02:
It seems that `Sony SmartWatch 3` is having a GPS receiver. It would explain why `Location` setting is available.

Still - we should find some official guidelines about when we should allow scan on Android Wear - google groups discussion would be sufficient. @jcamins if you have a moment - you could try to find something. I have different project at the moment on my plate.

Best Regards

jcamins, 2017-03-02:
I couldn't find anything about BLE on Android Wear online. Probably because AW 1 is intended to be used in concert with a phone.

dariuszseweryn, 2017-03-02:
I do not have an idea. I am trying to reach someone from Google to shed some light.

LadyViktoria, 2017-03-13:
You need Location permission for scanning BT devices.
My advice is to use The Scanner from setting Menü. Boundcon BT device. And in your App get The bounded devices and Connect to Them. 
Appart from that habe a Look at this
https://github.com/NightscoutFoundation/xDrip/blob/master/wear/src/main/java/com/eveningoutpost/dexdrip/LocationPermissionActivity.java
Maybe ist is what you are looking for

jcamins, 2017-03-13:
@LadyViktoria the issue here is not location permissions, which I was able to grant, but rather location services, which are not necessarily available at all on Android Wear.

dariuszseweryn, 2017-03-23:
@jcamins I have just pushed a commit that should fix your issue with Android Wear. Feel free to give feedback and reopen if necessary.

closed 2017-03-23 by dariuszseweryn

MGasztold, 2019-07-30:
Does anybody know a modern (2017/2018/2019) android wear BUDGET smart watch for which it is possible to develop apps leveraging Bluetooth Low Energy?

I am confused because I struggle to find anything. Anybody can share some experience and point some particular easy to buy smart watch model? Thank you