dariuszseweryn/RxAndroidBle #143: onConnectionStateChange not being called

skynet67, 2017-02-26:
Hi. I have found that sometimes after switching bluetooth state (enable -> disable -> enable) I was not able to reconnect to my device. The problem hides inside the RxBleRadioOperationConnect task, since it can't stop and release the semaphore from RxBleRadioImpl. 

This happens becase we don't receive a new state from onConnectionStateChange, so 
then `rxBleGattCallback.getOnConnectionStateChange().doOnNext()` will never be executed.

After some reasearches I've found that other people has similar issue with receiving a new state:
http://stackoverflow.com/questions/35103701/ble-android-onconnectionstatechange-not-being-called
https://code.google.com/p/android/issues/detail?id=223582

With a such fix `rxBleGattCallback.getOnConnectionStateChange().timeout(30, TimeUnit.SECONDS)` I was able to reconnect.

uKL, 2017-02-27:
Hey!

The fix you've proposed seems to bo only a partial workaround. It won't behave correctly when `autoConnect` flag will be set to true as the connection may take very long time.

Are you able to reproduce the issue? It would be helpful to get some ADB logs with log level set to debug

```
RxBleClient.setLogLevel(RxBleLog.DEBUG);
```

dariuszseweryn, 2017-02-27:
The code you use would also be appreciated. For instance if you use `autoConnect = true` then it is possible that the callback will happen after several (unspecified) minutes or even never.

skynet67, 2017-02-27:
Hi there. 
I use 'autoConnect = false'. 
Simple log: https://gist.github.com/skynet67/9781916bd0f4ad02bc13ef183981e4d8
Full log: https://gist.github.com/skynet67/372eaa1b55c260ab1650e32345f3f138

uKL, 2017-02-27:
Could you also paste how you use the `RxBleClient` instance?

skynet67, 2017-02-27:
```
@Provides
    @Singleton
    RxBleClient provideRxBleClient(Application application) {
        return RxBleClient.create(application);
    }
```
```
  public Observable<Peripheral> scan(UUID... services) {
        return bleClient.scanBleDevices(services)
                .doOnSubscribe(() -> lastFoundDevices.clear())
                .map(result -> peripheralDeviceForMacAddress(result.getBleDevice().getMacAddress()))
                .doOnNext(device -> lastFoundDevices.add(device))
                .cast(Peripheral.class);
    }

public PeripheralDevice peripheralDeviceForMacAddress(String macAddress) {
        PeripheralDevice device = deviceHashMap.get(macAddress);
        if (device == null) {
            device = new PeripheralDevice(bleClient.getBleDevice(macAddress), context);
            deviceHashMap.put(macAddress, device);
        }
        return device;
    }

```

Than I try to connect.

```
 private Observable<RxBleConnection> getConnectionObservable(boolean isReconnect) {
        return Observable.defer(() -> {
            final AtomicReference<Throwable> receivedError = new AtomicReference<>(null);
            return device
                    .establishConnection(context, isReconnect)
                    .compose(Rx.mainThread())
                    // Set the device state to "CONNECTING" while connection isn't established yet
                    .doOnSubscribe(() -> lastConnectionAttemptError = null)
                    // If connection was successful - set the device state to "CONNECTED"
                    .doOnNext(connectionSubject::onNext)
                    ....
                    .doAfterTerminate(() -> handleDisconnect(receivedError.get()));
        });
    }
```

uKL, 2017-02-27:
For now, from the logs, I can see that after the connection is established (with device 00:07:80:EA:A3:89) you are setting up a notification and it is successful.

```
02-27 14:50:06.893 23106-23118/com.my_app D/BluetoothGatt: onSearchComplete() = Device=00:07:80:EA:A3:89 Status=0
02-27 14:50:06.893 23106-23118/com.my_app D/RxBle#BluetoothGatt: onServicesDiscovered status=0
02-27 14:50:06.895 23106-25327/com.my_app D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(44899208)
02-27 14:50:06.896 23106-25327/com.my_app D/BluetoothGatt: setCharacteristicNotification() - uuid: d1b41cde-dbf5-4acf-8679-ecb8b4dca6fc enable: true
02-27 14:50:06.897 28614-28624/? D/BtGatt.GattService: registerForNotification() - address=00:07:80:EA:A3:89 enable: true
02-27 14:50:06.897 28614-28745/? I/bt_btif: HAL bt_gatt_callbacks->client->register_for_notification_cb
02-27 14:50:06.898 28614-28745/? D/BtGatt.GattService: onRegisterForNotifications() - address=null, status=0, registered=1, charUuid=d1b41cde-dbf5-4acf-8679-ecb8b4dca6fc
02-27 14:50:06.904 23106-25327/com.my_app D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(119800995)
02-27 14:50:06.905 23106-24563/com.my_app D/RxBle#Radio: FINISHED RxBleRadioOperationServicesDiscover(62693809)
02-27 14:50:06.906 23106-24563/com.my_app D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicWrite(44899208)
02-27 14:50:06.943 28614-28745/? I/bt_btif: HAL bt_gatt_callbacks->client->write_characteristic_cb
02-27 14:50:06.943 23106-23193/com.my_app D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=1ed50000-dbf5-4acf-8679-ecb8b4dca6fc status=0
02-27 14:50:06.944 23106-24563/com.my_app D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicWrite(44899208)
02-27 14:50:06.945 23106-24563/com.my_app D/RxBle#Radio:  STARTED RxBleRadioOperationDescriptorWrite(119800995)
02-27 14:50:07.041 28614-28745/? I/bt_btif: HAL bt_gatt_callbacks->client->write_descriptor_cb
02-27 14:50:07.043 23106-23117/com.my_app D/RxBle#BluetoothGatt: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
02-27 14:50:07.047 23106-24563/com.my_app D/RxBle#Radio: FINISHED RxBleRadioOperationDescriptorWrite(119800995)
```

Then the adapter is being disabled and the disconnect operation never finishes. I'll investigate more.

One more question. Where does the "My MyApp: DISCONNECTED" log come from?

skynet67, 2017-02-27:
This is our internal event that has nothing in common with BleState. So if device.getConnectionState() != RxBleConnection.RxBleConnectionState.CONNECTED and our currentState == CONNECTED, then we assume that device has been disconnected.

dariuszseweryn, 2017-02-27:
@skynet67 From the logs I see that the disconnect operation is started but never finished. It should timeout in about 10 seconds but you say that the connect operation is the faulty one. Could you elaborate?

Btw. I wonder if using of the Gatt Server is messing with the BLE stack somehow.
```
02-27 14:50:08.805 4216-21339/? D/BluetoothGattServer: onServiceAdded() - service=badbadba-dbad-badb-adba-badbadbadbadstatus=0
```

skynet67, 2017-02-27:
I don't know why disconnect operation is missing, but it should be there. 
another logs:
```
02-27 21:54:50.189 28285-28285/ D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(9103522)
02-27 21:54:58.602 28285-28430/ D/RxBle#Radio: FINISHED RxBleRadioOperationDisconnect(98404472)
02-27 21:54:58.603 28285-28430/ D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(9103522)
02-27 21:54:50.192 4216-4216/? D/BluetoothManager: getConnectedDevices
02-27 21:54:50.192 28614-17457/? D/BtGatt.GattService: endServiceDeclaration()
02-27 21:54:50.192 28614-17457/? D/BtGatt.GattService: continueServiceDeclaration() - srvcHandle=0
02-27 21:54:50.192 28614-17457/? D/BtGatt.GattService: continueServiceDeclaration() - next entry type=1
02-27 21:54:50.194 28614-28745/? D/BtGatt.GattService: onServiceAdded() UUID=badbadba-dbad-badb-adba-badbadbadbad, status=0, handle=46
```

I found only one reference to "badbad":
BADBADBADBAD	"Bad bad bad bad", Burroughs large systems "uninitialized" memory (48-bit words)

dariuszseweryn, 2017-03-03:
@skynet67 Current `1.2.0-SNAPSHOT` release should have a fix for this issue.
Best Regards

closed 2017-03-03 by dariuszseweryn

RobLewis, 2017-03-03:
Any idea when 1.2.0 official release might be?

dariuszseweryn, 2017-03-03:
I have found a potential compatibility issue with projects using Dagger dependency injection. Once we will solve it - I will make a release. I would like to do it in the next two weeks.