dariuszseweryn/RxAndroidBle #146: BleGattException{status=1, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_WRITE'}}

lukasblue, 2017-03-08:
Hello. I'm not quite sure if it's not related to previous issue "BleGattException status=1"

I am connecting to the device like this : 

`connectionObservable = _device.establishConnection(context, false)
                .takeUntil(disconnectTriggerSubject)
                .observeOn(AndroidSchedulers.mainThread())
                .doOnUnsubscribe(this::clearConnectionSubscription)
                .compose(new ConnectionSharingAdapter());

        connectionObservable.subscribe(this::onConnectionReceived, this::onConnectionFailure);`

everything looks great, I can read and write characteristics 
example : 

`   public void writeCharacteristic()
    {
        //subscriptionUpdateCharacteristicsInitialization = connectionObservable
        connectionObservable
                .flatMap(rxBleConnection -> rxBleConnection.writeCharacteristic(char_uuid, value))
                .subscribe(
                        characteristicValue -> {
                            Log.d("neurron", "CHARACTERISTIC INITIALIZATION VALUE CONFIRMED!");
                            triggerDisconnect();
                        },
                        throwable -> {
                            // Handle an error here.
                            Log.d("neurron", "CHARACTERISTIC INITIALIZATION ERROR!!! "+throwable.toString());
                        }
                );
    }
`

after some time I need to reboot ble device and connect to it for a secound time. There is no problem with establishing connection (I am doing this in the same way like for the first time (above)), but when I'm trying to write characteristic (the same way) I'm getting an error "BleGattException{status=1, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_WRITE'}}"

I will be appreciate for any suggestion.

dariuszseweryn, 2017-03-08:
Hello. 
If you would check the `BleGattException` and read the javadoc you could then check what is the `status=1` about at the page: https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h 

It seems that the Android OS is caching services from the previous connection (which are not valid anymore after the reboot as the handles may have changed - no logs to confirm that though).

What can you do in such a situation:
1. Turn the bluetooth off and on again - it makes the Android clear all caches
2. Tinker with a hidden method `BluetoothGatt.refresh()` - to get access to the raw `BluetoothGatt` you would need to use the `1.2.0-SNAPSHOT` version of the library and create a custom operation or access it through reflection.
3. Make the peripheral you are using to have exactly the same handles to services / characteristics after the reboot.

The snapshot version has a known bug at this moment - it does not work with projects that are using Dagger 2.

lukasblue, 2017-03-09:
Thank you very much for extremely quick response. 

You have right, after turning off and on bluetooth on device everything works prefectly. Could you please tell me how can I get 1.2.0-SNAPSHOP. Maybe thats silly, but I cannot find such version. The last one is 1.1.0. From your answer I suggest that this version (1.2.0-SNAPSHOT) will give me the access to BluetoothGatt and I will be able to refresh it via reflection. Is it possible to give me some example code of that (accessing gatt) ?

LadyViktoria, 2017-03-09:
repositories {
        maven { url 'https://oss.sonatype.org/content/repositories/snapshots/' }
}

dependencies {
    compile 'com.polidea.rxandroidble:rxandroidble:1.2.0-SNAPSHOT'
}

LadyViktoria, 2017-03-09:
i am also interestd in that gatt things.
i am still using 1.2.0 snapshot. 
i use a passive device that connects automatically to my phone.
after a while i get an error that GATT is null and cound not estaböish the connection

lukasblue, 2017-03-09:
Thanks a lot. That's working property. 
Could you please give me any tip how to access BluetoothGatt - to refresh it ?

dariuszseweryn, 2017-03-09:
Checkout `RxBleConnection.queue(RxBleRadioOperationCustom<T> operation)` This gives you access to the `BluetoothGatt`

lukasblue, 2017-03-14:
Thank you very much. That's working perfectly.

lukasblue, 2017-03-14:
There is still one more issue in my code. To perform this custom method i have to subscribe on it. My code looks like this : 

`    private static class CustomGattRefreshOperation implements RxBleRadioOperationCustom<Void>
    {
        CustomGattRefreshOperation() {
        }

        @NonNull
        @Override
        public Observable<Void> asObservable(BluetoothGatt bluetoothGatt,
                                               RxBleGattCallback rxBleGattCallback,
                                               Scheduler scheduler) throws Throwable {
            refreshDeviceCache(bluetoothGatt);
            return null ;
        }
(...)
}`

`_connection.queue(new CustomGattRefreshOperation())
                .subscribe(result -> {
                    Log.d("tag","done");
                }, throwable -> {
                    Log.d("tag", "error "+throwable.toString());
                });`

so obvious error in this case is : 
java.lang.IllegalArgumentException: The custom operation asObservable method must return a non-null observable

maoueh, 2017-03-14:
@lukasblue You need to return a valid `Observable`, simplest way is to use `Observable.fromCallable(...)`:

```
@NonNull
@Override
public Observable<Void> asObservable(BluetoothGatt bluetoothGatt,
                                                                   RxBleGattCallback rxBleGattCallback,
                                                                   Scheduler scheduler) throws Throwable {
    return Observable.fromCallable(() -> refreshDeviceCache(bluetoothGatt));
}
```

At the moment `asObservable` is called, the library is not yet executing your operation. It simply want a way to do it later which is why you return an `Observable`, to defer execution.

In your previous code, you're refreshing when `asObservable` was called which is wrong. By using the construct above, the refresh is not perform when `asObservable` is called, it's being done later on when library decides it's now the first on the queue, meaning it's ready to execute.

@dariuszseweryn Do you think we would need a simpler abstraction that would have a `run` and and who would implement `asObservable` as:

```
@NonNull
@Override
public Observable<Void> asObservable(BluetoothGatt bluetoothGatt,
                                                                   RxBleGattCallback rxBleGattCallback,
                                                                   Scheduler scheduler) throws Throwable {
    return Observable.fromCallable(() -> run(...));
}
```

?

dariuszseweryn, 2017-03-20:
I do not think that having a `run()` abstract method would be a good idea since a lot of potential use-cases will be asynchronous and only a few synchronous. I would like to keep the API as simple (small) as possible. It may be a good idea to add your example of showing the synchronous use-case using `Observable.fromCallable()` in the readme.

Since there was no further response from @lukasblue — I am closing.

closed 2017-03-20 by dariuszseweryn