dariuszseweryn/RxAndroidBle #148: crash report

gzhhong, 2017-03-10:
### Summary
<br />
03-10 12:07:20.145 26185-26185/com.orange.scannertest D/BluetoothGatt: connect() - device: 48:FB:72:EA:9F:65, auto: false
03-10 12:07:20.145 26185-26185/com.orange.scannertest D/BluetoothGatt: registerApp()
03-10 12:07:20.146 26185-26185/com.orange.scannertest D/BluetoothGatt: registerApp() - UUID=bcc00f61-f9ff-4bc0-80e6-e91cdf4aeef3
03-10 12:07:20.147 26185-26204/com.orange.scannertest D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
03-10 12:07:21.101 26185-26185/com.orange.scannertest D/com.orange.nsd**: start the scan of nsd on wifi of service _presence._tcp
03-10 12:07:21.277 26185-26202/com.orange.scannertest I/art: Background partial concurrent mark sweep GC freed 18618(1432KB) AllocSpace objects, 0(0B) LOS objects, 39% free, 10MB/17MB, paused 1.495ms total 112.213ms
03-10 12:07:21.305 26185-26204/com.orange.scannertest D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=48:FB:72:EA:9F:65
03-10 12:07:21.312 26185-26204/com.orange.scannertest W/BluetoothGatt: Unhandled exception in callback
                                                                       rx.exceptions.OnErrorNotImplementedException: Task rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker@12f52a1e rejected from java.util.concurrent.ThreadPoolExecutor@366c42ff[Terminated, pool size = 0, active threads = 0, queued tasks = 0, completed tasks = 0]
                                                                           at rx.internal.util.InternalObservableUtils$ErrorNotImplementedAction.call(InternalObservableUtils.java:386)
                                                                           at rx.internal.util.InternalObservableUtils$ErrorNotImplementedAction.call(InternalObservableUtils.java:383)
                                                                           at rx.internal.util.ActionSubscriber.onError(ActionSubscriber.java:44)
                                                                           at rx.observers.SafeSubscriber._onError(SafeSubscriber.java:153)
                                                                           at rx.observers.SafeSubscriber.onError(SafeSubscriber.java:115)
                                                                           at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:60)
                                                                           at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:30)
                                                                           at rx.Observable.subscribe(Observable.java:10236)
                                                                           at rx.Observable.subscribe(Observable.java:10203)
                                                                           at rx.Observable.subscribe(Observable.java:10008)
                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onConnectionStateChange(RxBleGattCallback.java:71)
                                                                           at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:181)
                                                                           at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
                                                                           at android.os.Binder.execTransact(Binder.java:446)
                                                                        Caused by: java.util.concurrent.RejectedExecutionException: Task rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker@12f52a1e rejected from java.util.concurrent.ThreadPoolExecutor@366c42ff[Terminated, pool size = 0, active threads = 0, queued tasks = 0, completed tasks = 0]
                                                                           at java.util.concurrent.ThreadPoolExecutor$AbortPolicy.rejectedExecution(ThreadPoolExecutor.java:2011)
                                                                           at java.util.concurrent.ThreadPoolExecutor.reject(ThreadPoolExecutor.java:793)
                                                                           at java.util.concurrent.ThreadPoolExecutor.execute(ThreadPoolExecutor.java:1339)
                                                                           at java.util.concurrent.Executors$DelegatedExecutorService.execute(Executors.java:583)
                                                                           at rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker.schedule(ExecutorScheduler.java:79)
                                                                           at rx.internal.operators.OperatorObserveOn$ObserveOnSubscriber.schedule(OperatorObserveOn.java:188)
                                                                           at rx.internal.operators.OperatorObserveOn$ObserveOnSubscriber$1.request(OperatorObserveOn.java:145)
                                                                           at rx.Subscriber.setProducer(Subscriber.java:209)
                                                                           at rx.Subscriber.setProducer(Subscriber.java:205)
                                                                           at rx.internal.operators.OperatorObserveOn$ObserveOnSubscriber.init(OperatorObserveOn.java:139)
                                                                           at rx.internal.operators.OperatorObserveOn.call(OperatorObserveOn.java:75)
                                                                           at rx.internal.operators.OperatorObserveOn.call(OperatorObserveOn.java:40)
                                                                           at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:44)
                                                                           at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:30) 
                                                                           at rx.Observable.subscribe(Observable.java:10236) 
                                                                           at rx.Observable.subscribe(Observable.java:10203) 
                                                                           at rx.Observable.subscribe(Observable.java:10008) 
                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onConnectionStateChange(RxBleGattCallback.java:71) 
                                                                           at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:181) 
                                                                           at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70) 
                                                                           at android.os.Binder.execTransact(Binder.java:446) 
#### Preconditions
<br />
#### Steps to reproduce actual result
1.
2.
3.
#### Actual result
<br />
#### Expected result
<br />

dariuszseweryn, 2017-03-10:
Hello.
From the readme: 
```
Obtaining the client

It's your job to maintain single instance of the client. You can use singleton, scoped Dagger component or whatever else you want.
```
You have to persist a reference to the `RxBleClient` object as long as you use anything related to the library.

Best Regards

closed 2017-03-10 by dariuszseweryn

gzhhong, 2017-03-13:
aha, you are right. Thank you so much!