dariuszseweryn/RxAndroidBle #149: BleGattException{macAddress=CC:78:AB:8A:30:06, status=133 (0x85 -> https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h), bleGattOperationType=BleGattOperation{description='CONNECTION_STATE'}}

sanjay4one, 2017-03-15:
### Summary
Hello I am trying to connect my ble devices by using RxAndroidBle Sample app but every time it is throwing the below exception.
BleGattException{macAddress=CC:78:AB:8A:30:06, status=133 (0x85 -> https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h), bleGattOperationType=BleGattOperation{description='CONNECTION_STATE'}}

Any help would be appreciated.
Thanks in advance.
<br />
#### Preconditions
<br />
#### Steps to reproduce actual result
1.
2.
3.
#### Actual result
<br />
#### Expected result
<br />

dariuszseweryn, 2017-03-20:
Hello,

Sorry for the delay - I was on vacation.
The `status=133` exception is known as `GATT_ERROR`. It may happen in multiple situations:

1. When the peripheral is out of range / turned off at the moment of establishing the connection
2. When the Android BLE stack is not working correctly (sometimes turning off and on both Bluetooth Adapter and WiFi helps)
3. When a peripheral does not conform correctly to BLE specification
4. When the phone BLE drivers are not good enough
5. Other (check google)

This error cannot be prevented on Android unfortunately — or at least I do not know how. 
Best Regards.

closed 2017-03-20 by dariuszseweryn

aamnapm, 2019-07-06:
@dariuszseweryn  hi i have ble device 
when i connect ble device i have three authentication.
my first and second authentication work correctely but in thrice auth in response i have problem  @sanjay4one 
what am i doing?