dariuszseweryn/RxAndroidBle #152: Are there plans to support Rx-BluetoothGattServer?

brightredchilli, 2017-03-28:
### Summary
Are there plans to support Rx-BluetoothGattServer?

Hi, I'm working on a project that starts a ble server. As far as I can tell, this library does not wrap any of BluetoothGattServer. Are there plans for this, and if so, would it be interesting for me to try to get a PR in of something that I have in place? I based my work heavily on how this library is structured.

This is how I imagine the interface would look like:

```
public interface RxBleServer {

    /***
     * Returns all writes that happen on a particular characteristic UUID. No validation
     * if you subscribe on a characteristic that is not actually a write characteristic.
     * Assembles values if they are written in chunks.
     */
    public Observable<byte[]> getCharacteristicWrites(UUID characteristicUUID);

    /***
     * Send an indication on a characteristic.
     */
    public Observable<Integer> indicateCharacteristic(BluetoothGattCharacteristic characteristic,
                                                      byte[] value);
}
```

dariuszseweryn, 2017-03-29:
Hello @brightredchilli 
Thanks for your interest. I would definitely like a functionality like this wrapped in a reactive approach.
This idea would need a separate module in my opinion (i.e. `rxandroidbleserver`) as most apps would need either central or peripheral functionality but rather not both.
It is interesting tough how Android would handle both roles at once — but that is a different topic.
Either way - a green light from me.

brightredchilli, 2017-03-30:
Alright, that sounds great - i imagine we'd need to maintain a separate dev branch while the server side stuff gets up to speed - its probably not going to have all of the layers and bells and whistles that the client side has for a bit. I'll start off with 'getWriteCharacteristic'(get all ble writes to a characteristic on this gatt server), in my opinion this has the least dependencies and is the easiest to understand.

dariuszseweryn, 2017-03-31:
A good idea from @uKL: prepare the server module so it could be easily plugged in instead of an `RxBleDeviceMock`. To do that there will be a need for an Interface for all Android components used in the server.

dustedrob, 2017-05-08:
Is there any ongoing work for this. Would love to collaborate?

kobronson, 2017-06-28:
Hi @dariuszseweryn 

is there any plan to release RxAndroidBle version with server support? If yes, then when it is expected to arrive?

dariuszseweryn, 2017-06-28:
Hello @kobronson 

We have a plan (as you see this issue being opened) but it is not decided yet in what version of the library (or a different project) it will be available nor when will it arrive. There are things that have greater priority at the moment.

External pull requests are welcomed though. :)

uKL, 2018-03-16:
Just to mention that the API may be shared with the mock #370

EskelCz, 2018-10-16:
I could really use this, how much work would a simple implementation take? Maybe I can fund a Bounty on it.