dariuszseweryn/RxAndroidBle #154: Zip Notification results with write commands

AmitShah, 2017-03-31:
### Summary
I have a custom piece of hardware that acknowledges writes to characteristics via notifications.  Is it possible to perform a zip operation between a writeCharacteristic and notification?  I leveraged the CharacteristicOperationExampleActivity.java and I would like to accomplish something like the following:

` 
notificationObservable = connectionObservable
                    //setupNotification emits an Observerbale that emits an observerable on complete
                    .flatMap(rxBleConnection -> rxBleConnection.setupNotification(UUID.fromString(...)))
                    .doOnNext(notificationObservable -> runOnUiThread(this::notificationHasBeenSetUp));
Observable.from(new String[]{"AT$INFO=?"}).concatMap(cmd->
           
                Observable.zip(
                    connectionObservable.flatMap(
                                rxBleConnection ->
                                        rxBleConnection.writeCharacteristic(
                                                characteristicUuid,
                                                HexString.hexToBytes(convertStringToHex(cmd)))),
                        notificationObservable.flatMap(no->no),
                        Pair::new)

            ).subscribe(resultOfCommands->{
                //convertHexToString(HexString.bytesToHex((byte[])resultOfCommands));

            }, error->{

            });
`
However, I never receive any bytes from the notificationObservable.  I am sort of new to javarx, and any help is greatly appreciated.

dariuszseweryn, 2017-03-31:
Please post this question on Stackoverflow instead of here as this is a question about how to implement a solution - not an issue in the library. The knowledge that will be in the answer will be easier to reach for other people when it will be available on Stackoverflow.

dariuszseweryn, 2017-03-31:
@AmitShah When posting on Stackoverflow - remember to add a tag `rxandroidble`

closed 2017-03-31 by dariuszseweryn