dariuszseweryn/RxAndroidBle #155: Internal error when setting up notification.

marcbaldwin, 2017-03-31:
### Summary
`BleCannotSetCharacteristicNotificationException` when setting up notification on characteristic.

#### Steps to reproduce actual result
1. Connect to BLE device
2. Discover services
3. Get characteristic
4. Setup notification for characteristic

```
device.establishConnection(context, true)
      .flatMap(connection -> connection.discoverServices()
            .flatMap(services -> Observable.zip(
                                services.getCharacteristic(UART_UUID, TX_UUID),
                                services.getCharacteristic(UART_UUID, RX_UUID),
                                (tx, rx) -> {
                                     ...
                                     connection.setupNotification(rx)
                                     ...
                                }
                        )
            )
      )
```

#### Actual result
Exception: BleCannotSetCharacteristicNotificationException

#### Expected result
Characteristic notification observable to be set up.

The exception is caused as `BluetoothGatt.setCharacteristicNotification` returns false, due to the fact `mClientIf` has not been initialised. `mClientIf` is initialised in the `onClientRegistered` callback.

If I reconnect to the device then the notification setup succeeds.

Any help with this would be greatly appreciated.

Thanks,

 Marc

dariuszseweryn, 2017-03-31:
Hello @marcbaldwin 
Could you add what phone / OS you use?
Additionally please set `RxBleLog.setLogLevel(RxBleLog.VERBOSE)` and paste the logs from the application.

dariuszseweryn, 2017-04-10:
Since there is no contact and so far the described situation is working as designed (`onError()` invoked when the notification cannot be set) – I am closing.

closed 2017-04-10 by dariuszseweryn