dariuszseweryn/RxAndroidBle #156: BleDisconnectedException when connect to deivce sometimes

liunewshine, 2017-04-01:
### Summary
Continuing establishConnection then unsubscribe，sometimes connect success，sometimes thow BleDisconnectedException
#### Steps to reproduce actual result
Continuing establishConnection then unsubscribe
#### Actual result
sometimes connect success，sometimes thow BleDisconnectedException
#### Expected result
always connect success
[image]
[image]

dariuszseweryn, 2017-04-03:
Hello @liunewshine
 Excuse me if I did not understand you correctly. Do you have issue that `RxBleDevice.establishConnection()` does sometimes emit a `BleDisconnectedException`?

dariuszseweryn, 2017-04-10:
Since there is no contact and the described situation is working as designed (`onError()` invoked when the connection cannot be established) – I am closing.

closed 2017-04-10 by dariuszseweryn