dariuszseweryn/RxAndroidBle #160: Mock setupNotification overwrites Gatt value

blakopf, 2017-04-04:
### Summary
Why does the setupNotification(...) method overwrite the value of the BluetoothGattCharacteristic?

In detail, look at https://github.com/Polidea/RxAndroidBle/blob/master/mockrxandroidble/src/main/java/com/polidea/rxandroidble/mockrxandroidble/RxBleConnectionMock.java#L430

`
private void setCharacteristicNotification(UUID notificationCharacteristicUUID, boolean enable) {
   writeCharacteristic(notificationCharacteristicUUID, new byte[]{(byte) (enable ? 1 : 0)}).subscribe();
}`

dariuszseweryn, 2017-04-04:
Hello @blakopf 
To be honest — I do not see a reason — this may be a bug. Why do you ask?

blakopf, 2017-04-05:
Because it messes up my values.

I setup my mock with various services and characteristics with values.
If I connect to the mock later and setupNotifications(...) for a UUID, the value of the characteristic for this UUID becomes 1 (and 0 if I unsubscribe from the notification):

`new byte[]{(byte) (enable ? 1 : 0)}`

I think this is a bug and the value should not be overwritten.

dariuszseweryn, 2017-04-05:
It seems so. The fix has just been landed and should be available shortly in `1.3.0-SNAPSHOT`.
Best Regards.

closed 2017-04-05 by dariuszseweryn

blakopf, 2017-04-05:
That was fast, thank you very much.