dariuszseweryn/RxAndroidBle #162: Problems with unsubscribe

paulonova, 2017-04-05:
Hi !
I'm doing an app that works as a remote control to a ventilator using RxAndroidBle. I have a problem with the "unsubscribe" because when I  use             .flatMap(rxBleConnection ->  rxBleConnection.writeCharacteristic(Uuids.UUID_RX, flaktCommandConcat.getBytes())) 

and after that I use "subscription.unsubscribe();"  the writeCharacteristics doesn´t work because the "unsubscribe" runs always first and the connection disconect before the data  was sent..

What I need is:
- When I click the button i want to connect to the ventilator
- Then send all values
- And then disconnect.
ps: If I repeat the procedure, it will need to do the same thing over and over again.

Can some one help me with some idea?
I tryed to use .delay(1000, Time.MILISECONDS) and it worked but it  took a long time to send the information to the ventiator...

Please some one help me! I´m on a deadLine...

Thanks
Paulo

This is my code:

public void writeRxCharacteristics(String flaktCommandConcat){

        rxBleDevice = rxBleClient.getBleDevice(Uuids.DEVICE_ADDRESS);

        subscription = rxBleDevice.establishConnection(true) //false
                .observeOn(AndroidSchedulers.mainThread())
                .flatMap(rxBleConnection -> rxBleConnection.createNewLongWriteBuilder()
                .setCharacteristicUuid(Uuids.UUID_RX)
                .setBytes(flaktCommandConcat.getBytes())
                .build())

                .subscribe(
                        byteArray -> {
                            Log.d("CharacteristicValue","WRITE: " + Arrays.toString(byteArray));
                        },

                        throwable -> {
                            Log.d("CharacteristicValue","Throwable: " + throwable.toString());
                            rxBleActivity.onScanFailure(throwable, getContext());
                        }
                );

        rxBleDevice.observeConnectionStateChanges()
                .observeOn(AndroidSchedulers.mainThread())
                .delay(1000, TimeUnit.MILLISECONDS)
                .subscribe(
                        rxBleConnectionState -> {
                            Log.d("RxBleConnectionState", " CON_STATUS: " + rxBleConnectionState);
                            disconnect();

                        },
                        throwable -> {
                            Log.d("ConnectionStateChanges","Throwable: " + throwable.toString());

                        }
                );
    }

   public void disconnect() {
        if (subscription != null && !subscription.isUnsubscribed()) {
            subscription.unsubscribe();
            subscription = null;
        }
        Log.d("CONNECTION2", " CON_STATUS: " + rxBleDevice.getConnectionState().toString());
    }

uKL, 2017-04-05:
Hey,

this is an implementation question which is more suitable to be posted on StackOverflow. Please post a question and tag with `rxandroidble` tag. 

Thanks!

closed 2017-04-05 by uKL

paulonova, 2017-04-05:
ok!