dariuszseweryn/RxAndroidBle #163: How to get an existing connection?

jaychang0917, 2017-04-05:
Suppose the ble device is paired and connected by user manually, how can i get the existing connection to perform read/write instead call `establishConnection` to create a new connection? Thanks!

uKL, 2017-04-05:
How do you see the connection to be established otherwise than by calling the establishConnection method? This question looks to be a good candidate to be posted to StackOverflow with *rxandroidble* tag.

jaychang0917, 2017-04-05:
In settings -> Bluetooth -> paired device, it shows connected

uKL, 2017-04-05:
This shows you the Classic Bluetooth connection status, which is independent of Bluetooth Low Energy. In this case, just use establishConnection.

jaychang0917, 2017-04-05:
oh, I get it! btw, if i call `establishConnection` two times, it will create two connections or reuse the existing connection?

uKL, 2017-04-05:
1. If you'd do this when another connection with the same device is active you'll get an error.
2. Check ConnectionSharingAdapter and our example project.

closed 2017-04-05 by jaychang0917