dariuszseweryn/RxAndroidBle #164: Sometimes BleDisconnectedException is thrown when writeCharacteristic

jaychang0917, 2017-04-06:
```java
bleConnection.writeCharacteristic(A_UUID, data)
      .subscribeOn(Schedulers.io())
      .observeOn(AndroidSchedulers.mainThread())
      .subscribe(
        charValue -> Log.d(TAG, "success"),
        error -> Log.d(TAG, "error"));
```

Sometimes it throws `com.polidea.rxandroidble.exceptions.BleDisconnectedException`, have any idea?

dariuszseweryn, 2017-04-06:
If the `bleConnection` got disconnected during the write - it is totally legit to emit `BleDisconnectedException` as the write cannot complete then.

jaychang0917, 2017-04-06:
So i need to monitor the connection state using `observeConnectionStateChanges `?

dariuszseweryn, 2017-04-06:
A good idea is to have only a single `.subscribe()` in the flow related to BLE.
You can also monitor when the `.establishConnection()` will emit `BleDisconnectedException`
`.observerConnectionStateChanges()` is rather for notifying the UI mostly - it was not meant to be used as an input to other actions.

jaychang0917, 2017-04-06:
What does single `.subscribe()` means?  Do you mean I need to chain `establishConnection ` and `writeCharacteristic ` when I want to write data? So `observerConnectionStateChanges ` can't monitor ble connection state, right?

dariuszseweryn, 2017-04-06:
Yes, this is what I personally do - I construct a chained flow of `Observable`s with a single subscribe because then I am sure that I will get just a single notification that something went wrong.
`.observeConnectionStateChanges()` is emitting only the connection states - It does not emit errors that may have triggered the disconnect.

jaychang0917, 2017-04-06:
Since i have a method that will call writeCharacteristic several times in 1s, if i chain with `establishConnection `, is that ok?

jaychang0917, 2017-04-06:
And one thing i want to make sure, does `unsubscribe` the subscription created by `establishConnection` means close the connection? Do i need `unsubscribe` the subscription created by `writeCharacteristic `?

dariuszseweryn, 2017-04-06:
Chaining should be fine as long as you try to establish only one connection at any given time.

Unsubscribing from `.establishConnection()` is closing it. It is described in the javadoc:
```
     * The connection is automatically disconnected (and released) when resulting Observable is unsubscribed.
     * On the other hand when the connections is interrupted by the device or the system, the Observable will be unsubscribed as well
     * following BleDisconnectedException or BleGattException emission.
```
Generally a good idea is to unsubscribe from `Subscriptions` whenever you are no longer interested in the result.

jaychang0917, 2017-04-06:
Thanks your details explanation and advice. Thank you very much!

closed 2017-04-06 by jaychang0917

RobLewis, 2017-12-05:
I'm seeing a writeCharacteristic() attempt cause an error/unsubscription and can't figure out why: [StackOverflow](https://stackoverflow.com/questions/47644540/ble-device-writecharacteristic-failing)