dariuszseweryn/RxAndroidBle #165: Access to low level bluetooth callback in custom operations

uKL, 2017-04-06:
In order to improve throughput more, I'd like to have more direct access (skipping Rx overhead) to BluetoothGattCallback.

Let's discuss how.

dariuszseweryn, 2017-04-06:
First idea - add a callback listener for every possible callback of `BluetoothGattCallback`

uKL, 2017-04-06:
This was my first idea, with a few assumptions:

1. Only one callback listener can be plugged in at time
1. It sticks to a custom operation paradigm
1. The callback listener is disposed (null) at the end of the custom operation automatically to prevent leaks.

What do you think?

dariuszseweryn, 2017-04-06:
Why `1. Only one callback listener can be plugged in at time`?

uKL, 2017-04-06:
To simplify the whole thing and avoid list iteration. In the scope of a single custom operation, I think it will be enough. What's your opinion?

dariuszseweryn, 2017-04-06:
Ah, one at a time in this meaning. That's reasonable. I first thought of a Callback Listener per callback type (one callback for `onCharacteristicWrite()` other for `onCharacteristicChanged()` and so on) and that one could not set both at the same time.

uKL, 2017-04-06:
Nope, I'm thinking about some abstract class implementing **BluetoothGattCallback** providing default, empty methods as templates. The user would be able to implement whichever callback required and only that one.

closed 2017-06-29 by dariuszseweryn

dariuszseweryn, 2017-07-04:
Is available in `1.4.0-SNAPSHOT`

RobLewis, 2017-09-13:
This sounds interesting. Is there any documentation of how it fits into the `RxAndroidBle` overall scheme? Usage example?

uKL, 2017-09-18:
Hey Rob,
all you need can be found in `RxBleGattCallback#setNativeCallback` Javadoc. The BluetoothGattCallback interface is documented in Android’s documentation.