dariuszseweryn/RxAndroidBle #167: BLE  device is (usually) not found in scan

RobLewis, 2017-04-06:
The prototype "Aeroscope" BLE device I'm writing an app for just had a firmware update. Previously, the RxAndroidBle library found 2 different prototype samples very reliably and quickly during the initial scan for devices. Now it almost never finds the device, even after scanning for a full minute. I'd blame this on the new firmware (I'm waiting to hear back from the developers), but iOS apps like LightBlue and BLExplr are still able to see the device. 

I've tried both 1.1.0 and 1.2.0 versions of the library, with the same result. The attached log is from 1.2.0. Below are sections of the log that seem relevant, It appears that the device ("Jack's Aeroscope") is seen at one point, but then there is a class cast exception. I'm continuing to analyze but wanted to submit this before the end of today. 

```
04-05 16:04:17.422 7947-7947/io.aeroscope.aeroscope D/AeroscopeBluetoothServc: Entered scanForAeroscopes()
04-05 16:04:17.435 7947-7947/io.aeroscope.aeroscope D/BluetoothAdapter: isEnabled
04-05 16:04:17.492 7947-7947/io.aeroscope.aeroscope D/AeroscopeBluetoothServc: AsScanSubscriber onStart() called
04-05 16:04:17.502 7947-7947/io.aeroscope.aeroscope D/AeroscopeBluetoothServc: Exiting scanForAeroscopes() after subscribing
04-05 16:04:17.502 7947-7947/io.aeroscope.aeroscope D/MainActivity: Finished startScan()
04-05 16:04:17.524 7947-7947/io.aeroscope.aeroscope D/BluetoothAdapter: startLeScan(): null
04-05 16:04:17.529 7947-7947/io.aeroscope.aeroscope D/BluetoothAdapter: 985258297: getState(). Returning 12
04-05 16:04:17.533 7947-7947/io.aeroscope.aeroscope D/BluetoothAdapter: 985258297: getState(). Returning 12
04-05 16:04:17.535 7947-7975/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3d23050 requestNextVsync
04-05 16:04:17.536 7947-7947/io.aeroscope.aeroscope D/BluetoothAdapter: 985258297: getState(). Returning 12
04-05 16:04:17.541 7947-7975/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3d23050 latestVsyncEvent 558588505802034
04-05 16:04:17.553 7947-7975/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3d23050 requestNextVsync
04-05 16:04:17.557 7947-7964/io.aeroscope.aeroscope D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=1
04-05 16:04:17.561 7947-7947/io.aeroscope.aeroscope I/BluetoothLeScanner: startRegisteration: mLeScanClients={android.bluetooth.BluetoothAdapter$2@1ed478df=android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper@e4618a}
04-05 16:04:17.565 7947-7975/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3d23050 latestVsyncEvent 558588522487966
04-05 16:04:17.636 7947-7975/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3d23050 requestNextVsync
04-05 16:04:17.641 7947-7975/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3d23050 latestVsyncEvent 558588606307844
...
04-05 16:04:39.777 7947-8056/io.aeroscope.aeroscope D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=52:6A:43:7C:9F:06, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=null, mManufacturerSpecificData={76=[12, 14, 0, -48, -49, 114, -28, -122, -41, -14, 91, 25, 23, -122, 101, 8]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-79, mTimestampNanos=1120853243854871}
04-05 16:04:50.955 7947-7966/io.aeroscope.aeroscope D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=52:6A:43:7C:9F:06, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=null, mManufacturerSpecificData={76=[12, 14, 0, -46, -49, 115, 11, 90, 25, -37, 78, 84, 62, -119, -97, 49]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-79, mTimestampNanos=1120864424254641}
04-05 16:05:06.252 7947-7964/io.aeroscope.aeroscope D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=52:6A:43:7C:9F:06, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=null, mManufacturerSpecificData={76=[12, 14, 0, -44, -49, 6, 4, 102, -112, 96, -31, 103, -111, 75, 81, 50]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-78, mTimestampNanos=1120879720660488}
04-05 16:05:17.250 7947-8056/io.aeroscope.aeroscope D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=52:6A:43:7C:9F:06, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=null, mManufacturerSpecificData={76=[12, 14, 0, -42, -49, -52, 35, 106, 16, 9, 95, 53, 11, 47, 125, -69]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-79, mTimestampNanos=1120890719279181}
04-05 16:05:17.498 7947-8054/io.aeroscope.aeroscope D/BluetoothAdapter: stopLeScan()
04-05 16:05:17.501 7947-8054/io.aeroscope.aeroscope D/BluetoothAdapter: 985258297: getState(). Returning 12
04-05 16:05:17.503 7947-8054/io.aeroscope.aeroscope D/BluetoothAdapter: 985258297: getState(). Returning 12
04-05 16:05:17.503 7947-8054/io.aeroscope.aeroscope I/BluetoothLeScanner: startRegisteration: mLeScanClients={android.bluetooth.BluetoothAdapter$2@1ed478df=android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper@e4618a} ,callback=android.bluetooth.BluetoothAdapter$2@1ed478df
04-05 16:05:17.526 7947-8054/io.aeroscope.aeroscope D/AeroscopeBluetoothServc: AsScanSubscriber onCompleted() called; unsubscribed
04-05 16:05:17.528 7947-8054/io.aeroscope.aeroscope D/AeroscopeBluetoothServc: AsScanSubscriber onCompleted() broadcast asDeviceVector
04-05 16:05:17.528 7947-7947/io.aeroscope.aeroscope D/MainActivity: Got foundDevices: []
04-05 16:05:17.530 7947-7947/io.aeroscope.aeroscope D/MainActivity: We are done scanning.
...
[logs_8_04-05.txt](https://github.com/Polidea/RxAndroidBle/files/902784/logs_8_04-05.txt)

04-05 16:08:35.273 7947-7964/io.aeroscope.aeroscope D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=6B:CA:B3:49:12:E8, mScanRecord=ScanRecord [mAdvertiseFlags=26, mServiceUuids=null, mManufacturerSpecificData={76=[16, 2, 11, 0]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-48, mTimestampNanos=1121088741861116}
04-05 16:08:58.739 7947-7966/io.aeroscope.aeroscope D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=CC:C6:B5:9F:9A:F7, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=[f9541234-91b3-bd9a-f077-80f2a6e57d00], mManufacturerSpecificData={}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=Jack's Aeroscope������], mRssi=-58, mTimestampNanos=1121112203939117}
04-05 16:08:58.741 7947-7947/io.aeroscope.aeroscope D/BluetoothDevice: mAddress: CC:C6:B5:9F:9A:F7
04-05 16:08:58.766 7947-7947/io.aeroscope.aeroscope D/AeroscopeBluetoothServc: AsScanSubscriber onError() received: io.aeroscope.aeroscope.AeroscopeBluetoothService cannot be cast to io.aeroscope.aeroscope.AeroscopeDisplay
                                                                               java.lang.ClassCastException: io.aeroscope.aeroscope.AeroscopeBluetoothService cannot be cast to io.aeroscope.aeroscope.AeroscopeDisplay
                                                                                   at io.aeroscope.aeroscope.-$Lambda$11.$m$0(Unknown Source)
                                                                                   at io.aeroscope.aeroscope.-$Lambda$11.call(Unknown Source)
                                                                                   at rx.internal.util.ActionObserver.onNext(ActionObserver.java:39)
                                                                                   at rx.internal.operators.OnSubscribeDoOnEach$DoOnEachSubscriber.onNext(OnSubscribeDoOnEach.java:96)
                                                                                   at rx.observers.SerializedObserver.onNext(SerializedObserver.java:91)
                                                                                   at rx.observers.SerializedSubscriber.onNext(SerializedSubscriber.java:94)
                                                                                   at rx.internal.operators.OperatorTakeTimed$TakeSubscriber.onNext(OperatorTakeTimed.java:62)
                                                                                   at rx.internal.operators.OperatorTake$1.onNext(OperatorTake.java:76)
                                                                                   at rx.internal.operators.OperatorObserveOn$ObserveOnSubscriber.call(OperatorObserveOn.java:224)
                                                                                   at rx.android.schedulers.LooperScheduler$ScheduledAction.run(LooperScheduler.java:107)
[... cut ...]
04-05 16:08:58.924 7947-7975/io.aeroscope.aeroscope D/GraphicBuffer: unregister, handle(0xa38b7ac0) (w:401 h:59 s:416 f:0x1 u:0x000f02)
04-05 16:08:58.927 7947-7975/io.aeroscope.aeroscope D/GraphicBuffer: register, handle(0xa38b7ac0) (w:401 h:59 s:416 f:0x1 u:0x000f02)
04-05 16:09:00.778 7947-7975/io.aeroscope.aeroscope D/OpenGLRenderer: Flushing caches (mode 0)
04-05 16:09:00.778 7947-7975/io.aeroscope.aeroscope D/OpenGLRenderer: Flushing caches (mode 0)
04-05 16:09:00.779 7947-7975/io.aeroscope.aeroscope D/GraphicBuffer: unregister, handle(0xa38b7ac0) (w:401 h:59 s:416 f:0x1 u:0x000f02)
```

RobLewis, 2017-04-06:
I should add that the "Unknown scanning error" message near the end comes from my adaptation of your code to handle scan errors. Which means that it is _not_ a `BleScanException`. 

The connection scanner code is: 

    public void scanForAeroscopes() { // entered here OK
        Log.d( LOG_TAG, "Entered scanForAeroscopes()" ); 
        rxBleDeviceVector.clear( ); // empty the vectors of discovered devices
        asDeviceVector.clear( );
        scanSubscriber = new AsScanSubscriber<>( ); // subscriber to RxBleScanResult
        scanSubscription = asBleClient
                .scanBleDevices( asServiceIdArray ) // filter by Service UUID
                .subscribeOn( Schedulers.io() )  
                .observeOn( AndroidSchedulers.mainThread( ) )
                .take( MAX_AEROSCOPES ) // have tried values of 2 and 1
                .take( SCAN_TIME, TimeUnit.SECONDS )  // limit the length of the scan (initially 60L)
                .doOnNext( this::addRxBleScanResult ) // put a found device in the Vectors
                .subscribe( scanSubscriber ); 
        Log.d( LOG_TAG, "Exiting scanForAeroscopes() after subscribing" ); 
    }

dariuszseweryn, 2017-04-06:
Hello @RobLewis 
The library only wraps what it gets from the native Android API. If the Android API stopped to scan your device after a firmware update there is nothing the library can do. It seems that the firmware you have uploaded has changed the Advertisement Interval to a value that is not useful for your phone. Unfortunately I cannot find any guidelines for Android system in that matter.
As for the `ClassCastException` it looks like the problem lies somewhere in your code as these classes are nowhere in the library.

RobLewis, 2017-04-06:
Thanks. 
Also: I don't know if it could be relevant, but I just switched from retroLambda to Jack.

RobLewis, 2017-04-06:
I don't know what the Advertisement Interval is or where to find it. 
Could the device name have anything to do with it? This new sample has a custom name ("Jack's Aeroscope")

dariuszseweryn, 2017-04-06:
Jack / Retrolambda — this should not make any difference since the library if you use version `1.2.0`. You can make a clean build just to be sure.

The device name is not the issue. Advertisement Interval is a value set on your peripheral (Aeroscope) in the firmware. Your firmware developer will know about it.

RobLewis, 2017-04-06:
It appears that the class cast exception is an irrelevant and unimportant bug, and the real problem seems to be this: 

```

04-05 16:08:35.273 7947-7964/io.aeroscope.aeroscope D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=6B:CA:B3:49:12:E8, mScanRecord=ScanRecord [mAdvertiseFlags=26, mServiceUuids=null, mManufacturerSpecificData={76=[16, 2, 11, 0]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-48, mTimestampNanos=1121088741861116}
04-05 16:08:58.739 7947-7966/io.aeroscope.aeroscope D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=CC:C6:B5:9F:9A:F7, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=[f9541234-91b3-bd9a-f077-80f2a6e57d00], mManufacturerSpecificData={}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=Jack's Aeroscope������], mRssi=-58, mTimestampNanos=1121112203939117}
04-05 16:08:58.741 7947-7947/io.aeroscope.aeroscope D/BluetoothDevice: mAddress: CC:C6:B5:9F:9A:F7
...
Caused by: rx.exceptions.OnErrorThrowable$OnNextValue: OnError while emitting onNext value: com.polidea.rxandroidble.RxBleScanResult.class
    at rx.exceptions.Exceptions.throwOrReport(Exceptions.java:190)
    at rx.internal.operators.OnSubscribeDoOnEach$DoOnEachSubscriber.onNext(OnSubscribeDoOnEach.java:98)
    at rx.observers.SerializedObserver.onNext(SerializedObserver.java:91) 
    at rx.observers.SerializedSubscriber.onNext(SerializedSubscriber.java:94) 
    at rx.internal.operators.OperatorTakeTimed$TakeSubscriber.onNext(OperatorTakeTimed.java:62) 
    at rx.internal.operators.OperatorTake$1.onNext(OperatorTake.java:76) 
    at rx.internal.operators.OperatorObserveOn$ObserveOnSubscriber.call(OperatorObserveOn.java:224) 
    at rx.android.schedulers.LooperScheduler$ScheduledAction.run(LooperScheduler.java:107) 
    at android.os.Handler.handleCallback(Handler.java:815) 
    at android.os.Handler.dispatchMessage(Handler.java:104) 
    at android.os.Looper.loop(Looper.java:194) 
    at android.app.ActivityThread.main(ActivityThread.java:5576) 
    at java.lang.reflect.Method.invoke(Native Method) 
    at java.lang.reflect.Method.invoke(Method.java:372) 
    at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:956) 
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:751) 
```
This scan found 2 BLE devices. The MAC address `CC:C6:B5:9F:9A:F7` is the one we are interested in. But something went wrong when the Subscriber tried to emit the device via `onNext()`, and instead `onError()` was called. (There is also a `doOnNext()` method in the Subscription, but it is not being called.) 

Does this suggest anything? It seems unlikely, but could the garbage characters that follow the device name ("Jack's Aeroscope") be a problem?

RobLewis, 2017-04-07:
Grasping at straws here: any chance the newer version of RxJava messed up something? 

The hardware developers have informed me that the Advertising Interval is usually 220 ms, but increases to 1.5 sec after 48 hours of no connection (to save battery). After a connection it goes back to 220 ms for 48 hours. I believe this is a new feature of the recent firmware update.

dariuszseweryn, 2017-04-07:
I don't expect a newer version of RxJava to cause problems — it is heavily tested.
The problem may be with how the `RxBleScanResult` is consumed in your code — I would look there.
As for the Advertising Interval I am no expert in how this should be set to get the best behaviour on Android devices. I have just seen that some settings are making specific phones to not scan the device at all. Every change in Advertising Interval should be tested in my opinion.

RobLewis, 2017-04-09:
I had been putting a 60-second timeout on the scan, but I increased that and found that in my trials the device was found—the longest observed delay this time was 40 seconds, and sometimes the device was found quickly. I suspect what may have happened was that the device has been in its 1.5 second Advertising Interval low power mode, in which it seems to take a long time to scan. (Once connected, it reverts to the standard 220 ms Advertising Interval for 48 hours.) 

However, I'm getting a new error in the log. Apparently I am trying to read and write the device too soon after attempting to connect with it. My write attempt came 7ms later. How long should a connection take to establish? 

But see what happens next: 

```
04-09 13:04:15.368 5129-5129/io.aeroscope.aeroscope D/AeroscopeDevice: Write error in Input message to Aeroscope: null
04-09 13:04:15.675 5129-5156/io.aeroscope.aeroscope D/GraphicBuffer: register, handle(0xac184ee0) (w:800 h:1280 s:800 f:0x1 u:0x000f02)
04-09 13:04:15.678 5129-5129/io.aeroscope.aeroscope D/ActivityThread: ACT-STOP_ACTIVITY_HIDE handled : 0 / android.os.BinderProxy@2effaaf1
04-09 13:04:15.714 5129-5129/io.aeroscope.aeroscope D/AeroscopeBluetoothServc: Returned from super.onUnbind()
04-09 13:04:15.715 5129-5129/io.aeroscope.aeroscope D/ActivityThread: SVC-UNBIND_SERVICE handled : 0 / BindServiceData{token=android.os.BinderProxy@3fb46635 intent=Intent { cmp=io.aeroscope.aeroscope/.AeroscopeBluetoothService }}
04-09 13:04:15.716 5129-5129/io.aeroscope.aeroscope D/ActivityThread: SVC-Destroying service: io.aeroscope.aeroscope.AeroscopeBluetoothService@a0d163d
04-09 13:04:15.716 5129-5129/io.aeroscope.aeroscope D/ActivityThread: SVC-STOP_SERVICE handled : 0 / android.os.BinderProxy@3fb46635
04-09 13:04:15.719 5129-5129/io.aeroscope.aeroscope E/ActivityThread: Service io.aeroscope.aeroscope.AeroscopeBluetoothService has leaked IntentReceiver com.polidea.rxandroidble.RxBleAdapterStateObservable$1$1@3743daee that was originally registered here. Are you missing a call to unregisterReceiver()?
    android.app.IntentReceiverLeaked: Service io.aeroscope.aeroscope.AeroscopeBluetoothService has leaked IntentReceiver com.polidea.rxandroidble.RxBleAdapterStateObservable$1$1@3743daee that was originally registered here. Are you missing a call to unregisterReceiver()?
        at android.app.LoadedApk$ReceiverDispatcher.<init>(LoadedApk.java:944)
        at android.app.LoadedApk.getReceiverDispatcher(LoadedApk.java:715)
        at android.app.ContextImpl.registerReceiverInternal(ContextImpl.java:1742)
        at android.app.ContextImpl.registerReceiver(ContextImpl.java:1722)
        at android.app.ContextImpl.registerReceiver(ContextImpl.java:1716)
        at android.content.ContextWrapper.registerReceiver(ContextWrapper.java:488)
        at com.polidea.rxandroidble.RxBleAdapterStateObservable$1.call(RxBleAdapterStateObservable.java:61)
        at com.polidea.rxandroidble.RxBleAdapterStateObservable$1.call(RxBleAdapterStateObservable.java:47)
        at rx.internal.operators.OnSubscribeCreate.call(OnSubscribeCreate.java:72)
        at rx.internal.operators.OnSubscribeCreate.call(OnSubscribeCreate.java:32)
        at rx.Observable.unsafeSubscribe(Observable.java:10140)
        at rx.internal.operators.OnSubscribeFilter.call(OnSubscribeFilter.java:45)
        at rx.internal.operators.OnSubscribeFilter.call(OnSubscribeFilter.java:30)
        at rx.Observable.unsafeSubscribe(Observable.java:10140)
        at rx.internal.operators.OnSubscribeMap.call(OnSubscribeMap.java:48)
        at rx.internal.operators.OnSubscribeMap.call(OnSubscribeMap.java:33)
        at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:48)
        at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:30)
        at rx.Observable.unsafeSubscribe(Observable.java:10140)
        at rx.internal.operators.OperatorMerge$MergeSubscriber.onNext(OperatorMerge.java:248)
        at rx.internal.operators.OperatorMerge$MergeSubscriber.onNext(OperatorMerge.java:148)
        at rx.internal.operators.OnSubscribeFromArray$FromArrayProducer.fastPath(OnSubscribeFromArray.java:76)
        at rx.internal.operators.OnSubscribeFromArray$FromArrayProducer.request(OnSubscribeFromArray.java:58)
        at rx.Subscriber.setProducer(Subscriber.java:211)
        at rx.internal.operators.OnSubscribeFromArray.call(OnSubscribeFromArray.java:32)
        at rx.internal.operators.OnSubscribeFromArray.call(OnSubscribeFromArray.java:24)
        at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:48)
        at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:30)
        at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:48)
        at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:30)
        at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:48)
        at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:30)
        at rx.Observable.unsafeSubscribe(Observable.java:10140)
        at rx.internal.operators.OnSubscribeMap.call(OnSubscribeMap.java:48)
        at rx.internal.operators.OnSubscribeMap.call(OnSubscribeMap.java:33)
        at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:48)
        at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:30)
        at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:48)
[... cut ...]
04-09 13:04:15.762 5129-6259/io.aeroscope.aeroscope D/AeroscopeDevice: AsConnSubscriber made connection com.polidea.rxandroidble.internal.connection.RxBleConnectionImpl@33c070f6
04-09 13:04:15.764 5129-5129/io.aeroscope.aeroscope D/AeroscopeDevice: getConnectionObservable used existing connection
04-09 13:04:15.771 5129-5157/io.aeroscope.aeroscope D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(937594675)
```

What does "Are you missing a call to unregisterReceiver()?" mean? After outputting this error the app seems to run OK, but I'd like to know what's going on.

dariuszseweryn, 2017-04-10:
> However, I'm getting a new error in the log. Apparently I am trying to read and write the device too soon after attempting to connect with it. My write attempt came 7ms later. How long should a connection take to establish?
Current architecture allows you to perform `.writeCharacteristic()` only after the `RxBleConnection` has been emitted - so it is safe to use.

> What does "Are you missing a call to unregisterReceiver()?" mean?
It means that the context that was used to register a `BroadcastReceiver` - which is internally used by `RxBleAdapterStateObservable` - has finished but the receiver was not unregistered (unsubscribed) creating a memory leak.

RobLewis, 2017-04-10:
So there's no information about how long it normally takes to establish a connection to a device found in a scan? No upper limit? Does it make sense to do a 
`while( !isConnected() ) { };` 
wait loop (perhaps with a timeout)? 
A brief pause in the UI is acceptable, but a long one (more than a few seconds) isn't. 
Any advantage to a `BlockingObservable`? Other solutions? 
(I think I'd prefer to avoid restructuring my code to put all subsequent actions in `establishConnection`'s `onNext()` handler.)

RobLewis, 2017-04-10:
Currently I'm using `.subscribeOn( Schedulers.io() )` for `scanBleDevices()`. Is this a good idea?

dariuszseweryn, 2017-04-11:
> So there's no information about how long it normally takes to establish a connection to a device found in a scan? No upper limit?

Generally there is no info - it will vary. `.establishConnection(false)` is guaranteed to fail in 35 seconds (usually 30).

> Does it make sense to do a `while( !isConnected() ) { };` wait loop (perhaps with a timeout)? Any advantage to a BlockingObservable? Other solutions?

No sense at all in my opinion, no and yes. `RxJava` is all about managing asynchronous code - learn to embrace it - http://www.grahamlea.com/2014/07/rxjava-threading-examples/ or tons of other learning sites available.

> Currently I'm using .subscribeOn( Schedulers.io() ) for scanBleDevices(). Is this a good idea?

Does not matter to much as every BLE method is called on the UI thread in the end and callbacks are always on background threads. It is not a bad idea at least. ;)

closed 2017-04-18 by dariuszseweryn

RobLewis, 2017-04-24:
Regarding this: 

_What does "Are you missing a call to unregisterReceiver()?" mean?_

> It means that the context that was used to register a BroadcastReceiver - which is internally used by RxBleAdapterStateObservable - has finished but the receiver was not unregistered (unsubscribed) creating a memory leak.

The only thing I can relate this to is that I set up a subscription to `observeConnectionStateChanges( )` in an Activity's `onStart()` method and don't unsubscribe it anywhere. Would this explain it? 

This brings up a more general question: I'm not clear on how the various Observables should be managed over the Activity lifecycle. As I interpret Google's documentation, in `onPause()` you should pause tasks that have high CPU usage. Is it OK to leave low-demand operations like monitoring ConnectionStateChanges running? Any general guidelines would be appreciated. 

Also, what is the recommended technique for handling the loss of a BLE connection, such as by moving a peripheral device out of Bluetooth range?

dariuszseweryn, 2017-04-24:
> The only thing I can relate this to is that I set up a subscription to observeConnectionStateChanges( ) in an Activity's onStart() method and don't unsubscribe it anywhere. Would this explain it?

Yes

> This brings up a more general question: I'm not clear on how the various Observables should be managed over the Activity lifecycle. As I interpret Google's documentation, in onPause() you should [...]

Whenever you are no longer interested in an `Observable` - unsubscribe. Android may kill background Activities at any time.

> Also, what is the recommended technique for handling the loss of a BLE connection, such as by moving a peripheral device out of Bluetooth range?

It depends on the application.

RobLewis, 2017-04-24:
Again, trying to interpret Android docs: a "paused" Activity isn't really in the background, it's just suspended momentarily for things like answering an incoming phone call (and it can remain visible). In that case, I think I would like to leave things like ConnectionStateChange monitoring enabled. 
Apparently it is possible for a paused Activity to be killed, but the app will first be notified by a call to `onStop()` (followed by `onDestroy()`). So it would appear that if I unsubscribe in `onStop()` I should be OK. True?

dariuszseweryn, 2017-04-24:
Yup

RobLewis, 2017-04-24:
Under what conditions (if any) does ConnectionStateChange call the Subscriber's `onError()` and `onCompleted()` methods?

dariuszseweryn, 2017-04-24:
Check `RxBleDeviceImpl` - it's encapsulated and easy to follow. TLDR; - never.