dariuszseweryn/RxAndroidBle #168: writeCharacteristic write wrong value sometimes

jaychang0917, 2017-04-07:
```java
bleConnection.writeCharacteristic(LIGHT_A_UUID, data)
      .subscribeOn(Schedulers.io())
      .observeOn(AndroidSchedulers.mainThread())
      .subscribe(
        charValue -> {
          Log.d(TAG, "Change eye color success");
        },
        error -> {
          closeBleConnection();
        });
```

The above function is used to change a light color, sometimes sending the same byte data but results different light color, have any idea?

uKL, 2017-04-07:
We need more data. Did you try enabling logs? Please have a look at `RxBleLog` class and it's static methods.

Do you mean that the data that is sent out is different that you provide to a write function?

dariuszseweryn, 2017-04-07:
Hello @jaychang0917 - do you reuse your `byte[] data`?

dariuszseweryn, 2017-04-18:
No response - closing.

closed 2017-04-18 by dariuszseweryn