dariuszseweryn/RxAndroidBle #169: Update JavaDocs

IgorGanapolsky, 2017-04-11:
### Summary
Hello,
thank you for creating this advanced library for BLE scanning based on RxJava.  I am trying to look through your code to get inspirations on how to use it in my own project, and to understand it better.  I wondering - can you add more JavaDocs throughout your code?  At least on the public methods.  As it stands, the code is very sophisticated and not trivial to follow.  Furthermore, RxJava has a high learning curve, which adds necessity to have better documentation throughout the codebase.
<br />
#### Preconditions
Look through the existing code.
<br />
#### Expected result
There need to be JavaDocs on public members.
<br />

dariuszseweryn, 2017-04-11:
Hello @IgorGanapolsky 
Thanks for the insight. I will add your request to my `TODOs`. Besides: which `public` methods do not have `JacaDoc`? I am trying to keep at least the API as documented as possible.

dariuszseweryn, 2017-04-11:
The interface of `OperationPriorityFifoBlockingQueue` is just a wrapper for a `PriorityBlockingQueue` intended to use with operations — it is a quite simple class. `RxBleRadioImpl` is basically a worker which is responsible for serializing operations and `RxBleClientImpl` is the implementation of the entry point to the library. Anything in particular is hard to comprehend? On which things should I concentrate describing?

IgorGanapolsky, 2017-04-11:
@dariuszseweryn Essentially putting what you just said in a JavaDoc comment for class declarations would be great.  It may seem intuitive and straightforward to smart engineers like you :), but believe me - this code is very non-trivial for newbies to BLE and those who not experts in RxJava.  Additionally, each public method having a JavaDoc comment would make it a lot clearer to follow.

dariuszseweryn, 2017-04-11:
Skip the flattery. ;) I know that diving into code that was written by others is not easy. I will try to add more comments and JavaDoc when I will have a moment.

RobLewis, 2017-04-12:
I sympathize. The project I'm on has involved simultaneously learning Android, RxJava, and BLE. I think "drinking from a firehose" is an apt way to put it :-)  Additional documentation is always a welcome thing.

kenwdelong, 2017-05-09:
Sorry, but also, are the javadocs hosted anywhere?

dariuszseweryn, 2017-05-10:
@kenwdelong they are available in the `.aar` you get from the Maven repository.