dariuszseweryn/RxAndroidBle #170: Upgrade gradle 3.5

passsy, 2017-04-13:
After upgrading to gradle 3.5 `./gradlew wrapper --gradle-version 3.5 --distribution-type all` checkstyle doesn't work anymore. I tried fixing it but couldn't find a solution :(

```
Pascals-MBP:rxandroidble pascalwelsch$ ./gradlew test
Jack is disabled, but one of the plugins you are using supports Java 8 language features.
Jack is disabled, but one of the plugins you are using supports Java 8 language features.
:mockrxandroidble:preBuild UP-TO-DATE
:mockrxandroidble:preDebugBuild UP-TO-DATE
:mockrxandroidble:checkDebugManifest
:mockrxandroidble:preDebugAndroidTestBuild UP-TO-DATE
:mockrxandroidble:preDebugUnitTestBuild UP-TO-DATE
:mockrxandroidble:preReleaseBuild UP-TO-DATE
:mockrxandroidble:preReleaseUnitTestBuild UP-TO-DATE
:mockrxandroidble:prepareComArtemzinRxjavaProguardRules1290Library
:mockrxandroidble:prepareIoReactivexRxandroid121Library
:rxandroidble:checkstyle FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':rxandroidble:checkstyle'.
> Unable to create a Checker: configLocation {rxandroidble/checkstyle.xml}, classpath {null}.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 2.033 secs
```

dariuszseweryn, 2017-04-13:
Hello @passsy 
Well... Another thing to dig in. :) Currently I have a different project on my plate and after that I am going to finish the work on the API 21 scanning. 
Feel free to dig deeper into this as I am not sure when I will have a moment.

uKL, 2017-04-16:
Fixed in master.

closed 2017-04-16 by uKL