dariuszseweryn/RxAndroidBle #171: Connection not established but still getting RxBleConnection with CONNECTED state

newaxis, 2017-04-13:
### Summary
I try basic thing: connect to device. To connect I need to press and hold Pair button on my device (custom project). When Pair button is not pressed I get RxBleConnection in onNext() and immediatly after I get BleGattException. When I press Pair on the device exceptions stop and I get only one RxBleConnection. Question is why I get RxBleConnection with Connected status whed device is not connected? :)

#### Steps to reproduce actual result
Code to reproduce this situation:
```
bleDevice = App.getRxBleClient(LoginActivity.this).getBleDevice("24:0A:C4:05:77:A2");
        connectionObservable = bleDevice.establishConnection(false)
                .retryWhen(new Func1<Observable<? extends Throwable>, Observable<?>>() {
                    @Override
                    public Observable<?> call(Observable<? extends Throwable> observable) {
                        return observable.flatMap(new Func1<Throwable, Observable<?>>() {
                            @Override
                            public Observable<?> call(Throwable throwable) {
                                Log.i(TAG, "call: error: " + throwable.getClass().getSimpleName());
                                return throwable instanceof BleException ? Observable.just(null) : Observable.error(throwable);
                            }
                        });
                    }
                })
                .compose(new ConnectionSharingAdapter())
                .observeOn(AndroidSchedulers.mainThread());
        connectionObservable.subscribe(new BaseObserver<RxBleConnection>(TAG) {
            @Override
            public void onNext(RxBleConnection item) {
                Log.i(TAG, "onNext1: " + bleDevice.getConnectionState().toString());
            }
        });
```

#### Actual result
```
04-13 13:48:04.370 23880-23895/TAG I/LoginActivity:call:error:BleGattException
04-13 13:48:04.735 23880-23880/TAG I/LoginActivity:onNext1:RxBleConnectionState{CONNECTED}
04-13 13:48:05.071 23880-24574/TAG I/LoginActivity:call:error:BleGattException
04-13 13:48:05.478 23880-23880/TAG I/LoginActivity:onNext1:RxBleConnectionState{CONNECTED}
04-13 13:48:05.815 23880-23895/TAG I/LoginActivity:call:error:BleGattException
04-13 13:48:06.289 23880-23880/TAG I/LoginActivity:onNext1:RxBleConnectionState{CONNECTED}
04-13 13:48:06.621 23880-23896/TAG I/LoginActivity:call:error:BleGattException
04-13 13:48:07.126 23880-23880/TAG I/LoginActivity:onNext1:RxBleConnectionState{CONNECTED}
04-13 13:48:07.462 23880-23895/TAG I/LoginActivity:call:error:BleGattException
04-13 13:48:07.528 23880-23880/TAG I/LoginActivity:onNext1:RxBleConnectionState{CONNECTED}
04-13 13:48:08.118 23880-23896/TAG I/LoginActivity:call:error:BleGattException
04-13 13:48:08.461 23880-23880/TAG I/LoginActivity:onNext1:RxBleConnectionState{CONNECTED}
04-13 13:48:08.797 23880-23896/TAG I/LoginActivity:call:error:BleGattException
04-13 13:48:12.901 23880-23880/TAG I/LoginActivity:onNext1:RxBleConnectionState{CONNECTED}
04-13 13:48:13.214 23880-23895/TAG I/LoginActivity:call:error:BleGattException
04-13 13:48:13.663 23880-23880/TAG I/LoginActivity:onNext1:RxBleConnectionState{CONNECTED}
04-13 13:48:13.968 23880-24574/TAG I/LoginActivity:call:error:BleGattException
04-13 13:48:14.387 23880-23880/TAG I/LoginActivity:onNext1:RxBleConnectionState{CONNECTED}
04-13 13:48:14.689 23880-23895/TAG I/LoginActivity:call:error:BleGattException
04-13 13:48:15.136 23880-23880/TAG I/LoginActivity:onNext1:RxBleConnectionState{CONNECTED}
04-13 13:48:15.516 23880-24574/TAG I/LoginActivity:call:error:BleGattException
04-13 13:48:20.521 23880-23880/TAG I/LoginActivity:onNext1:RxBleConnectionState{CONNECTED}
```
#### Expected result
Not getting RxBleConnection with Connected status when device is not connected.

dariuszseweryn, 2017-04-13:
Hello. Could you call `RxBleLog.setLogLevel(RxBleLog.VERBOSE)` and paste that logs?

dariuszseweryn, 2017-04-26:
No response - closing.

closed 2017-04-26 by dariuszseweryn