dariuszseweryn/RxAndroidBle #173: How to find cause of scan error

RobLewis, 2017-04-15:
### Summary
Scan finds device, then crashes
<br />
#### Preconditions
Scan for devices with a single Service UUID filter, use .take(1) and .take(120L) to limit scan
<br />
#### Steps to reproduce actual result
Code that does the scan
```
    scanSubscription = asBleClient
            .scanBleDevices( asServiceIdArray )          // argument must be an array, apparently
            .take( 1 )                     
            .take( 120L, TimeUnit.SECONDS )   
            .observeOn( AndroidSchedulers.mainThread() ) 
            .subscribe(
                    this::addRxBleScanResult,           
                    this::handleRxBleScanError,         
                    this::completeRxBleScan              
            );
```

#### Expected result
When the first device is found, `addRxBleScanResult()` should be called, then since the `.take(1)` has been satisfied, `completeRxBleScan()` should be called. 
<br />

#### Actual result
Fatal program error. The log mentions a ClassCastException (for 2 of my classes), but I am unable to find where or why this is taking place. The log indicates that the device was found, but none of the `.subscribe()` methods are called. The last thing before the crash is `startRegisteration`:

```
04-14 14:31:47.412 15670-15687/io.aeroscope.aeroscope D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=CC:C6:B5:9F:9A:F7, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=[f9541234-91b3-bd9a-f077-80f2a6e57d00], mManufacturerSpecificData={}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=Jack's Aeroscope������], mRssi=-57, mTimestampNanos=1892890531330833}
04-14 14:31:47.415 15670-15670/io.aeroscope.aeroscope D/BluetoothDevice: mAddress: CC:C6:B5:9F:9A:F7
04-14 14:31:47.425 15670-15670/io.aeroscope.aeroscope D/BluetoothAdapter: stopLeScan()
04-14 14:31:47.426 15670-15670/io.aeroscope.aeroscope D/BluetoothAdapter: 523560835: getState(). Returning 12
04-14 14:31:47.427 15670-15670/io.aeroscope.aeroscope D/BluetoothAdapter: 523560835: getState(). Returning 12
04-14 14:31:47.427 15670-15670/io.aeroscope.aeroscope I/BluetoothLeScanner: startRegisteration: mLeScanClients={android.bluetooth.BluetoothAdapter$2@3ab9d939=android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper@38c5992c} ,callback=android.bluetooth.BluetoothAdapter$2@3ab9d939
04-14 14:31:47.447 15670-15670/io.aeroscope.aeroscope E/AndroidRuntime: FATAL EXCEPTION: main
                                                                        Process: io.aeroscope.aeroscope, PID: 15670
                                                                        java.lang.IllegalStateException: Fatal Exception thrown on Scheduler.Worker thread.
                                                                            at rx.android.schedulers.LooperScheduler$ScheduledAction.run(LooperScheduler.java:114)
                                                                            at android.os.Handler.handleCallback(Handler.java:815)
                                                                            at android.os.Handler.dispatchMessage(Handler.java:104)
                                                                            at android.os.Looper.loop(Looper.java:194)
                                                                            at android.app.ActivityThread.main(ActivityThread.java:5576)
                                                                            at java.lang.reflect.Method.invoke(Native Method)
                                                                            at java.lang.reflect.Method.invoke(Method.java:372)
                                                                            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:956)
                                                                            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:751)
                                                                         Caused by: rx.exceptions.OnErrorFailedException: Error occurred when trying to propagate error to Observer.onError
                                                                            at rx.observers.SafeSubscriber._onError(SafeSubscriber.java:187)
                                                                            at rx.observers.SafeSubscriber.onError(SafeSubscriber.java:115)
                                                                            at rx.exceptions.Exceptions.throwOrReport(Exceptions.java:216)
                                                                            at rx.observers.SafeSubscriber.onNext(SafeSubscriber.java:139)
                                                                            at rx.internal.operators.OperatorObserveOn$ObserveOnSubscriber.call(OperatorObserveOn.java:224)
                                                                            at rx.android.schedulers.LooperScheduler$ScheduledAction.run(LooperScheduler.java:107)
                                                                            at android.os.Handler.handleCallback(Handler.java:815) 
                                                                            at android.os.Handler.dispatchMessage(Handler.java:104) 
                                                                            at android.os.Looper.loop(Looper.java:194) 
                                                                            at android.app.ActivityThread.main(ActivityThread.java:5576) 
                                                                            at java.lang.reflect.Method.invoke(Native Method) 
                                                                            at java.lang.reflect.Method.invoke(Method.java:372) 
                                                                            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:956) 
                                                                            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:751) 
```

<br />

uKL, 2017-04-15:
Hey!

Are you sure that the log is complete? I don't see any ClassCast you've mentioned.

Could you also post your `addRxBleScanResult` method? It looks like it's throwing an exception.

RobLewis, 2017-04-15:
Log attached. 
[logs_13_04-14.txt](https://github.com/Polidea/RxAndroidBle/files/923879/logs_13_04-14.txt)

Below is the method. However, I can't see in the log where it's being called. 

```
    private void addRxBleScanResult( RxBleScanResult scanResult ) {
        RxBleDevice device = scanResult.getBleDevice();
        Log.d( LOG_TAG, "addRxBleScanResult about to call addRxBleDevice with device " + device.toString() );
        // TODO: add each device to the displayed "found" list when it's found(?)
        addRxBleDevice( device );
    }
```
The method it calls is well tested and has never had a problem: 

```
    private boolean addRxBleDevice( RxBleDevice device ) {
        Log.d( LOG_TAG, "Entering addRxBleDevice with device " + device.getName() );
        if( !rxBleDeviceVector.contains( device ) ) {
            Log.d( LOG_TAG, "Adding RxBleDevice " + device.getName() + " to Vector" );
            addAeroscopeDevice( new AeroscopeDevice( device, this ) ); 
            return rxBleDeviceVector.add( device ); // true
        } else {
            Log.d( LOG_TAG, "RxBleDevice " + device.getName() + " already in rxBleDeviceVector");
            return false;
        }
    }
```

uKL, 2017-04-15:
Could you also post 
`AeroscopeDevice` and `handleRxBleScanError`. I'd say it's somewhere there. First, there is a failure in onNext handling, then another one in onError handling.

RobLewis, 2017-04-15:
AeroscopeDevice attached (sorry it is a mess; I'm in the middle of redoing it)

Here's `handleRxBleScanError()`: 
   ```
 private void handleRxBleScanError( Throwable scanError ) {
        nowScanning.set( false );        // reset "scan active" flag (synchronized) (error ends scan)
        errorScanning.set( true );       // set the "scan error" flag
        Log.d( LOG_TAG, "scanSubscription onError() received: " + scanError.getMessage(), scanError );
        handleScanError( scanError, callingContext ); // TODO: test
    }
```

OK, this is where I suspected the problem was days ago, but nothing is getting logged. As you will see below, `handleScanError()` is an adaptation of your own code, and I'm trying to send it a context so the Toast can be displayed. Since this method exists in the class `AeroscopeBluetoothService`, which doesn't have a UI, I am trying to send it the context of the `MainActivity` in the variable `callingContext`. In `MainActivity`, the `onServiceConnected()` routine has this: 
`asBleServiceRef.setCallingContext( MainActivity.this );`

Maybe `handleScanError()` is trying to cast AeroscopeBluetoothService to MainActivity, causing the ClassCastException, but why isn't anything getting logged? 

Here's the routine: 
```
    static void handleScanError( Throwable scanError, Context context ) {
        Log.d( LOG_TAG, "Entering handleScanError: " + scanError.getMessage() );
        Log.d( LOG_TAG, "Context passed to handleScanError is: " + context.getClass().getName() );  // debugging (pushed)
        if (scanError instanceof BleScanException ) {
            switch ( ((BleScanException) scanError).getReason() ) {
                case BleScanException.BLUETOOTH_NOT_AVAILABLE:
                    Log.d( LOG_TAG, "handleScanError: Bluetooth is not available" );
                    Toast.makeText( context, "Bluetooth is not available", Toast.LENGTH_SHORT ).show( );
                    break;
                case BleScanException.BLUETOOTH_DISABLED:
                    Log.d( LOG_TAG, "handleScanError: Bluetooth is disabled" );
                    Toast.makeText( context, "Enable bluetooth and try again", Toast.LENGTH_SHORT ).show( );
                    break;
                case BleScanException.LOCATION_PERMISSION_MISSING:
                    Log.d( LOG_TAG, "handleScanError: Location permission missing" );
                    Toast.makeText( context,
                            "On Android 6.0 location permission is required. Implement Runtime Permissions", Toast.LENGTH_SHORT ).show( );
                    break;
                case BleScanException.LOCATION_SERVICES_DISABLED:
                    Log.d( LOG_TAG, "handleScanError: Location services disabled" );
                    Toast.makeText( context, "Location services needs to be enabled on Android 6.0", Toast.LENGTH_SHORT ).show( );
                    break;
                case BleScanException.BLUETOOTH_CANNOT_START:
                default:
                    Log.d( LOG_TAG, "handleScanError: Bluetooth cannot start" );
                    Toast.makeText( context, "Unable to start scanning (reason unknown)", Toast.LENGTH_SHORT ).show( );
            } // switch
        } else { // exception was not a BleScanException
            Log.d( LOG_TAG, "handleScanError: Unknown scanning error" );
            Toast.makeText( context, "Error scanning for Aeroscopes: " + scanError.getMessage(), Toast.LENGTH_SHORT ).show( );
        }
        Log.d( LOG_TAG, "handleScanError Finished" );
    }

```
Really appreciate your help on this! 

[AeroscopeDevice.txt](https://github.com/Polidea/RxAndroidBle/files/923910/AeroscopeDevice.txt)

uKL, 2017-04-15:
Well, I think it may be something with the lambdas. Are you using retrolambda?

My concern is the line `addAeroscopeDevice( new AeroscopeDevice( device, this ) ); `

1. Correct me if I'm wrong, is this method within a service? `this` should be an AeroscopeDisplay. Is it?
2. If so, please try replacing this with an explicit `ClassName.this` or remove lambda at all (just to verify replace with an anonymous class)

RobLewis, 2017-04-15:
I was using retrolambda until you released 1.2.0 of the library, when I switched to Jack (which I know is deprecated and eventually will go away).

RobLewis, 2017-04-15:
addAeroscopeDevice immediately follows in the code: 

```
    private boolean addRxBleDevice( RxBleDevice device ) {
        Log.d( LOG_TAG, "Entering addRxBleDevice with device " + device.getName() );
        if( !rxBleDeviceVector.contains( device ) ) {
            Log.d( LOG_TAG, "Adding RxBleDevice " + device.getName() + " to Vector" );
            addAeroscopeDevice( new AeroscopeDevice( device, this ) ); 
            return rxBleDeviceVector.add( device ); // true
        } else {
            Log.d( LOG_TAG, "RxBleDevice " + device.getName() + " already in rxBleDeviceVector");
            return false;
        }
    }
    
    // Add an Aeroscope device to the Vector (returns true if it was not already present)
    // static because there's just 1 vector
    private boolean addAeroscopeDevice( AeroscopeDevice asDevice ) {
        Log.d( LOG_TAG, "Entering addAeroscopeDevice with device " + asDevice.bleDevice.getName() );
        if( !asDeviceVector.contains( asDevice ) ) {
            Log.d( LOG_TAG, "Adding AeroscopeDevice " + asDevice.bleDevice.getName( ) + " to Vector" );
            return asDeviceVector.add( asDevice ); // true
        } else { // don't think we should ever see this(?)
            Log.d( LOG_TAG, "AeroscopeDevice " + asDevice.bleDevice.getName() + " already in asDeviceVector");
            return false;
        }
    }
```
They are both in `AeroscopeBluetoothService`. The `this` is there to pass a reference to the Service to the AeroscopeDevice constructor, which it saves in a variable to allow the Device to call the Service methods.

RobLewis, 2017-04-25:
I have now seen what may be another variation of this "problem with lambdas". First, could you elaborate on what you mean by this? Is it that the lambdas aren't being compiled correctly? Is Jack failing? Is there something wrong with my logic? With release 1.2.0 of your library? Explain please. 
In any case, my code to scan for BLE devices has this: 
```
                .take( secondsToScan, TimeUnit.SECONDS )    
                .take( numberToFind )      
```
What seems to be happening is that `.take( numberToFind )` is satisfied, causing the Subscriber's `onCompleted()` to be called. Previously, this was a simple method reference: `this::completeRxBleScan`. AFAIK, it never had a problem until the recent 1.2.0 library update and/or the switch from retrolambda to Jack. Now I am seeing this in the log: 
```
04-24 13:56:39.015 22783-22783/io.aeroscope.aeroscope D/AeroscopeBluetoothServc: Adding AeroscopeDevice Jack's Aeroscope to Vector
04-24 13:56:39.017 22783-22783/io.aeroscope.aeroscope E/AndroidRuntime: FATAL EXCEPTION: main
                                                                        Process: io.aeroscope.aeroscope, PID: 22783
                                                                        java.lang.IllegalStateException: Fatal Exception thrown on Scheduler.Worker thread.
                                                                            at rx.android.schedulers.LooperScheduler$ScheduledAction.run(LooperScheduler.java:114)
                                                                            at android.os.Handler.handleCallback(Handler.java:815)
                                                                            at android.os.Handler.dispatchMessage(Handler.java:104)
                                                                            at android.os.Looper.loop(Looper.java:194)
                                                                            at android.app.ActivityThread.main(ActivityThread.java:5576)
                                                                            at java.lang.reflect.Method.invoke(Native Method)
                                                                            at java.lang.reflect.Method.invoke(Method.java:372)
                                                                            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:956)
                                                                            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:751)
                                                                         Caused by: java.lang.IncompatibleClassChangeError: Class 'io.aeroscope.aeroscope.-$Lambda$10' does not implement interface 'rx.functions.Action0' in call to 'void rx.functions.Action0.call()' (declaration of 'java.lang.reflect.ArtMethod' appears in /system/framework/core-libart.jar)
                                                                            at rx.internal.util.ActionSubscriber.onCompleted(ActionSubscriber.java:49)
                                                                            at rx.observers.SafeSubscriber.onCompleted(SafeSubscriber.java:79)
                                                                            at rx.internal.operators.OperatorObserveOn$ObserveOnSubscriber.checkTerminated(OperatorObserveOn.java:281)
                                                                            at rx.internal.operators.OperatorObserveOn$ObserveOnSubscriber.call(OperatorObserveOn.java:216)
                                                                            at rx.android.schedulers.LooperScheduler$ScheduledAction.run(LooperScheduler.java:107)
                                                                            at android.os.Handler.handleCallback(Handler.java:815) 
                                                                            at android.os.Handler.dispatchMessage(Handler.java:104) 
                                                                            at android.os.Looper.loop(Looper.java:194) 
                                                                            at android.app.ActivityThread.main(ActivityThread.java:5576) 
                                                                            at java.lang.reflect.Method.invoke(Native Method) 
                                                                            at java.lang.reflect.Method.invoke(Method.java:372) 
                                                                            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:956) 
                                                                            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:751) 
```
The key line would seem to be this: 
                                                                         `Caused by: java.lang.IncompatibleClassChangeError: Class 'io.aeroscope.aeroscope.-$Lambda$10' does not implement interface 'rx.functions.Action0' in call to 'void rx.functions.Action0.call()' (declaration of 'java.lang.reflect.ArtMethod' appears in /system/framework/core-libart.jar)`

Of course I can't tell, but I suspect that `-$Lambda$10` is the above method reference (I'm guessing that this is a system-created name for an inner class that implements the method reference.) 
I've replaced the method reference with an anonymous inner class and am now testing. So far it looks OK. 
Could you help me understand what's going on here?

RobLewis, 2017-05-03:
Perhaps someone smarter than me can explain this, but it appears at least some of these errors were due to the fact that several of my Subscribers' `onError( Throwable t )` methods were throwing `RuntimeException`, which I now understand is a no-no.

dariuszseweryn, 2017-05-12:
As it turned out - the cause was `jack` which was not regenerating lambdas as it should. Clean build solved the problem.

closed 2017-05-12 by dariuszseweryn