dariuszseweryn/RxAndroidBle #176: close or unsuscribe connection when the device drops the connection

ingalvaror, 2017-04-16:
### Summary
<br />hi im connecting to a device and sending commands without problems, but when the devices is disconnected by itself, gets stuck and i cant connect again and cant see the device when i scan again, so what i have to do to reset the bluetooth stack?  can i unsuscribe connectionObservable or reset RxBleDevice  or call triggerdisconnect in the onWriteFailure method? Thanks

    connectionObservable =  bleDevice
            .establishConnection(getApplicationContext(), false)
            .takeUntil(disconnectTriggerSubject)
            .doOnUnsubscribe(this::triggerDisconnect)
            .compose(new ConnectionSharingAdapter());

private void triggerDisconnect() {
disconnectTriggerSubject.onNext(null);
}

        connectionObservable
                .flatMap(rxBleConnection->rxBleConnection.writeCharacteristic(characteristicUuid, color))
                .observeOn(AndroidSchedulers.mainThread())
                .subscribe(bytes->{
                    onWriteSuccessPass();
                }, this::onWriteFailure);

uKL, 2017-04-18:
Hey!

The connection observable unsubscribes itself when the Bluetooth device is disconnected. If it didn't it means that the Android system didn't recognize that fact, that the connection was lost. 

What's your connection supervision timeout on the device? Maybe it's too big.

ingalvaror, 2017-04-18:
hi! thanks for your response! the time when the connections drops is around 1 minute, when i cancel the connection with triggerDisconnect  its ok, but when the connection drops itself i cant connect the device again, i have a question, in my observable i have this: 
takeUntil(disconnectTriggerSubject)
        .doOnUnsubscribe(this::triggerDisconnect)

Is it necessary to use both methods or just use the takeuntil? do i have any error in the code that i wrote?

uKL, 2017-04-19:
It's enough to have `.takeUntil(disconnectTriggerSubject)` as a disconnect trigger. Alternatively you can store the subscription and call `.unsubscribe()` on it.

The connection is automatically closed once the observable is unsubscribed, inversely the observable is unsubscribed (with an error being emitted) when the system recognizes that the connection is dead.

Please post verbose logs from that situation.

ingalvaror, 2017-04-19:
thanks for your response, i just want to be sure that the code is ok, this is running in a background service, i dont know if i have to write any additional code in this case, the proccess is: i start to scan, when the device is detected, try to connect, if gets connected i send a password through writecharacteristic  and then i send a command to turn on a light, then i send a command to turn off, the device suddenly disconnect itself and i try to connect again, this time the connection was done again but in some cases it seems that the connection is not closing. Another thing is sometimes when i connect again, the method to send password is calling a couple of times, i dont know if this is related to the problem, if the observable is not closing or restarting or something, thanks, this is the code, please tell me if im doing something wrong or if there is a better method to do this, Thanks: 

**///establish the connection**

```
connectionObservable =  bleDevice
        .establishConnection(getApplicationContext(), false)
        .takeUntil(disconnectTriggerSubject)
.retryWhen(new Func1<Observable<? extends Throwable>, Observable<?>>() {
                    @Override
                    public Observable<?> call(Observable<? extends Throwable> errorNotification) {
                        return errorNotification
                                .flatMap(new Func1<Throwable, Observable<?>>() {
                                    @Override
                                    public Observable<?> call(Throwable throwable) {

                                        return Observable.timer(10, TimeUnit.SECONDS);
                                    }
                                });
                    }
                })
        .compose(new ConnectionSharingAdapter());
        connectionObservable.subscribe(rxBleConnection -> {
            Log.d(getClass().getSimpleName(), "Hey, connection has been established!");
            sendPass();
        }, this::onConnectionFailure);

**the connection is done, so i send pass to be able to write**

        if (isConnected()) {

            connectionObservable
                    .flatMap(rxBleConnection -> rxBleConnection.writeCharacteristic(characteristicUuid, password))
                    .observeOn(AndroidSchedulers.mainThread())
                    .subscribe(bytes -> {
                        onWriteSuccessPass();
                    }, this::onWriteFailure);
        }
```

**send the command to turn off or turn on (this is called multiple times depending of the gps location)**

    public void changeLight(light) {
        if (light.equals("on")) {

            sendCommand(on);

        } else if (light.equals("off")) {

            sendCommand(off);
        }
    }

    public void sendCommand(byte[] bytesColor) {
        if (isConnected()) {
            connectionObservable
                    .flatMap(rxBleConnection -> rxBleConnection.writeCharacteristic(characteristicUuid, bytesColor))
                    .observeOn(AndroidSchedulers.mainThread())
                    .subscribe(bytes -> {
                        onWriteSuccessPermission();
                    }, this::onWriteFailure);
        }
    }

so i dont know if calling the observable again to change the light or send the pass after reconnection is related with the problem, here is the logs:

```
04-19 10:37:45.975 21669-21669/? D/RxBle#Radio:   QUEUED RxBleRadioOperationScan(213265115)
04-19 10:37:45.978 21669-21715/? D/RxBle#Radio:  STARTED RxBleRadioOperationScan(213265115)
04-19 10:37:46.072 1794-2350/? W/Conscrypt: Could not set socket write timeout: null
[... 117 lines cut ...]
04-19 10:39:57.593 21669-21715/? D/RxBle#Radio:  STARTED RxBleRadioOperationServicesDiscover(59864543)
04-19 10:39:57.604 21669-21669/? D/BluetoothGatt: discoverServices() - device: C8:A1:2F:4E:F7:03
04-19 10:39:57.604 9897-9953/? D/BtGatt.GattService: discoverServices() - address=C8:A1:2F:4E:F7:03, connId=7
04-19 10:39:57.605 9897-9914/? D/BtGatt.GattService: onSearchCompleted() - connId=7, status=0
04-19 10:39:57.607 21669-21681/? D/BluetoothGatt: onSearchComplete() = Device=C8:A1:2F:4E:F7:03 Status=0
04-19 10:39:57.607 21669-21681/? D/RxBle#BluetoothGatt: onServicesDiscovered status=0
04-19 10:39:57.612 21669-21855/? D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(86063017)
04-19 10:39:57.622 21669-21855/? D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(70034789)
04-19 10:39:57.624 21669-21855/? D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(106710088)
04-19 10:39:57.626 21669-21855/? D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(260918983)
04-19 10:39:57.632 21669-21715/? D/RxBle#Radio: FINISHED RxBleRadioOperationServicesDiscover(59864543)
04-19 10:39:57.632 21669-21715/? D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicWrite(86063017)
04-19 10:39:57.635 21669-21695/? D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=6e400002-b5a3-f393-e0a9-e50e24dcca9e status=0
04-19 10:39:57.638 21669-21715/? D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicWrite(86063017)
04-19 10:39:57.638 21669-21715/? D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicWrite(70034789)
04-19 10:39:57.643 21669-21680/? D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=6e400002-b5a3-f393-e0a9-e50e24dcca9e status=0
04-19 10:39:57.647 21669-21715/? D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicWrite(70034789)
[... cut ...]
04-19 10:39:57.691 21669-21690/? D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=6e400002-b5a3-f393-e0a9-e50e24dcca9e status=0
04-19 10:39:57.696 21669-21715/? D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicWrite(106710088)
04-19 10:39:57.696 21669-21715/? D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicWrite(260918983)
04-19 10:39:57.711 21669-21695/? D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=6e400002-b5a3-f393-e0a9-e50e24dcca9e status=0
04-19 10:39:57.715 21669-21715/? D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicWrite(260918983)

```

uKL, 2017-04-20:
What I see in the logs

1. You start the scan looking for the device
2. Once the device is found you establish connection
3. Under the hood, there is a service discovery leading to a write operation
4. There are two writes to a characteristic (6e400002-b5a3-f393-e0a9-e50e24dcca9e)
5. After the second write, I see something strange, it looks like there is another scan turned on just before the connection is disconnected.

Are you sure the code you've posted is complete? I don't see any scannings.

My other questions suggestions:
1. Please post `onWriteSuccessPass`
2. Please post complete code, including the scan.
3. Does it work without `retryWhen`?
4. Please add `doOnError` log between establishConnection and takeUntil

ingalvaror, 2017-04-20:
Thanks for your response! you are very kind! the retryWhen method is not really necessary, I use it to test the connection authomatically, but I just need to connect and send commands during certain changes in the position of the gps, so I dont need to be connected all the time, the problem that I have is the weird behavior of the disconnect sometimes and it seems that the observable keeps the state of what I have sent before because it calls the method onWriteFailure several times after it is disconnected by itself. I have two questions, do i need .observeOn(AndroidSchedulers.mainThread()) if this is running on a service? i change the ui but with a broadcast, the other question is, which must have the onError method ? this is my code of the service:

**creation of the service:**

```
    private PublishSubject<Void> disconnectTriggerSubject = PublishSubject.create();

 @Override
    public void onCreate() {
        super.onCreate();

        characteristicUuid = UUID.fromString(characteristicID);
        rxBleClient = MyApplication.getRxBleClient(this);
        RxBleClient.setLogLevel(RxBleLog.DEBUG);
        pairAndSendCommand();
        listener = new LocationListener() {
            @Override
            public void onLocationChanged(Location location) {

                if (!userGoesAway) {
                    float distance = location.distanceTo(placeLocation);
                    if (distance > radioPlace) {
                        userGoesAway = true;
                        turnOff();
                    }
                }
            }
            @Override
            public void onStatusChanged(String s, int i, Bundle bundle) {}

            @Override
            public void onProviderEnabled(String s) {}

            @Override
            public void onProviderDisabled(String s) { }
        };

        locationManager = (LocationManager) getApplicationContext().getSystemService(Context.LOCATION_SERVICE);
    }

    @Override
    public int onStartCommand(final Intent intent, final int flags, final int startId) {
        super.onStartCommand(intent, flags, startId);

        Log.d(TAG, "onStartCommand");

        if (intent != null && intent.getStringExtra("stop") != null) {

            triggerDisconnect();
            if (isScanning())
                scanSubscription.unsubscribe();

            if (locationManager != null) {
                locationManager.removeUpdates(listener);
            }
        }

        return START_NOT_STICKY;
    }

    @Override
    public void onTaskRemoved(Intent rootIntent) {

        triggerDisconnect();

        if (isScanning())
            scanSubscription.unsubscribe();

        if (locationManager != null) {
            locationManager.removeUpdates(listener);
[... 162 lines cut ...]
            locationManager.removeUpdates(listener);
        }

        userGoesAway = false;

    }

    private void onConnectionFailure(Throwable throwable) {
        System.out.println("Connection error: " + throwable);
    }

    private void onWriteFailure(Throwable throwable) {
        System.out.println("Write error: " + throwable.getMessage());
    }

    private void onWriteFailureOff(Throwable throwable) {
        System.out.println("Write error: " + throwable);
        userGoesAway = false;
    }

    private void triggerDisconnect() {
        disconnectTriggerSubject.onNext(null);
    }

```

ingalvaror, 2017-04-26:
hi, could you review the code?

uKL, 2017-04-27:
I'm sorry but I'm currently involved in some other tasks. If you'll find any issues in the library please submit an issue. I suggest that you can get more RxJava support from the community at StackOverflow.

closed 2017-04-27 by uKL