dariuszseweryn/RxAndroidBle #177: Can't

SsIDit, 2017-04-19:
Can't connect to device again after disconnect.
<br />
The first time I connect to device everything is ok. Then I unsubscribe and when trying to connect again I get BleDisconnectedException and BleGattCannotStartException.
<br/>
Restarting bluetooth doesn't help. Only full recompile of app. When I connect without library, I don't have such problem.

**Log:**
04-19 17:49:24.197 4175-4175/com.nsofta.resmartbulb D/com.nsofta.resmartbulb.ble.SingleController: State changed: CONNECTING
04-19 17:49:24.303 4175-4175/com.nsofta.resmartbulb D/BluetoothGatt: connect() - device: 04:A3:16:A7:AD:5D, auto: false
04-19 17:49:24.303 4175-4175/com.nsofta.resmartbulb D/BluetoothGatt: registerApp()
04-19 17:49:24.303 4175-4175/com.nsofta.resmartbulb D/BluetoothGatt: registerApp() - UUID=e0225aec-28f1-4110-89bb-b397dad8a97c
04-19 17:49:24.305 4175-4186/com.nsofta.resmartbulb D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
04-19 17:49:25.533 4175-4187/com.nsofta.resmartbulb D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=04:A3:16:A7:AD:5D
04-19 17:49:25.536 4175-4417/com.nsofta.resmartbulb D/com.nsofta.resmartbulb.ble.SingleController: State changed: CONNECTED
04-19 17:49:25.552 4175-4417/com.nsofta.resmartbulb E/com.nsofta.resmartbulb.ble.SingleController: BleDisconnectedException{bluetoothDeviceAddress='04:A3:16:A7:AD:5D'}
04-19 17:49:25.565 4175-4175/com.nsofta.resmartbulb D/BluetoothGatt: discoverServices() - device: 04:A3:16:A7:AD:5D
04-19 17:49:26.660 4175-4253/com.nsofta.resmartbulb D/BluetoothGatt: onSearchComplete() = Device=04:A3:16:A7:AD:5D Status=0
04-19 17:49:26.661 4175-4417/com.nsofta.resmartbulb D/com.nsofta.resmartbulb.ble.SingleController: 
04-19 17:49:32.188 4175-4417/com.nsofta.resmartbulb E/com.nsofta.resmartbulb.ble.SingleController: BleDisconnectedException{bluetoothDeviceAddress='04:A3:16:A7:AD:5D'}
04-19 17:49:32.272 4175-4417/com.nsofta.resmartbulb E/com.nsofta.resmartbulb.ble.SingleController: BleDisconnectedException{bluetoothDeviceAddress='04:A3:16:A7:AD:5D'}
04-19 17:49:32.286 4175-4175/com.nsofta.resmartbulb E/com.nsofta.resmartbulb.ble.SingleController: BleGattCannotStartException{macAddress=04:A3:16:A7:AD:5D, bleGattOperationType=BleGattOperation{description='CHARACTERISTIC_WRITE'}}
04-19 17:49:32.303 4175-4175/com.nsofta.resmartbulb E/com.nsofta.resmartbulb.ble.SingleController: BleGattCannotStartException{macAddress=04:A3:16:A7:AD:5D, bleGattOperationType=BleGattOperation{description='CHARACTERISTIC_WRITE'}}

dariuszseweryn, 2017-04-19:
Hello @SsIDit 
Please be so kind to show your code and call `RxBleLog.setLogLevel(RxBleLog.VERBOSE)` before capturing logs.

SsIDit, 2017-04-20:
Hi, I did this before I posted this logs

Code:
```
    @Provides
    @Singleton
    public RxBleClient provideBleClient(Context context) {
        RxBleClient rxBleClient = RxBleClient.create(context);
        if (BuildConfig.DEBUG) {
            rxBleClient.setLogLevel(RxBleLog.VERBOSE);
        }
        return rxBleClient;
    }
```

dariuszseweryn, 2017-04-20:
Then you are not posting logs from a debug build as there are clearly no logs from the library. Without it I cannot help you.
I will also need your code that is touching all interaction with the library.

SsIDit, 2017-04-20:
Sorry for being so noob. I didn't check VERBOSE filter flag while watching logs. Here's full log:

`D/BluetoothAdapter: stopLeScan()
D/BluetoothAdapter: STATE_ON
D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(168601508)
D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(168601508)
V/RxBle#BleConnectionCompat: Connecting without reflection
D/BluetoothGatt: connect() - device: 04:A3:16:A7:AD:5D, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=df2acb44-52b4-4441-8cea-8aa69061a6f7
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=04:A3:16:A7:AD:5D
D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
D/RxBle#Radio:   QUEUED RxBleRadioOperationServicesDiscover(11251770)
D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(168601508)
D/RxBle#Radio:  STARTED RxBleRadioOperationServicesDiscover(11251770)
D/BluetoothGatt: discoverServices() - device: 04:A3:16:A7:AD:5D
D/BluetoothGatt: onSearchComplete() = Device=04:A3:16:A7:AD:5D Status=0
D/RxBle#BluetoothGatt: onServicesDiscovered status=0
D/RxBle#Radio: FINISHED RxBleRadioOperationServicesDiscover(11251770)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(8621495)
D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicWrite(8621495)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(109083823)
D/RxBle#Radio:  REMOVED RxBleRadioOperationCharacteristicWrite(109083823)
D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicWrite(119559651)
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000ffe9-0000-1000-8000-00805f9b34fb status=0
D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicWrite(119559651)
D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(41808438)
D/RxBle#Radio:  STARTED RxBleRadioOperationDisconnect(41808438)
V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@6a86e28
V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@2374741
D/RxBle#Radio:   QUEUED RxBleRadioOperationScan(266842456)
D/BluetoothManager: getConnectionState()
D/BluetoothManager: getConnectedDevices
D/BluetoothGatt: cancelOpen() - device: 04:A3:16:A7:AD:5D
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=04:A3:16:A7:AD:5D
D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=0
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=5
D/RxBle#Radio: FINISHED RxBleRadioOperationDisconnect(41808438)
D/RxBle#Radio:  STARTED RxBleRadioOperationScan(266842456)
D/BluetoothAdapter: startLeScan(): null
D/BluetoothAdapter: STATE_ON
D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
D/RxBle#Radio: FINISHED RxBleRadioOperationScan(266842456)
D/BluetoothAdapter: stopLeScan()
D/BluetoothAdapter: STATE_ON
D/RxBle#Radio:   QUEUED RxBleRadioOperationScan(45418503)
D/RxBle#Radio:  STARTED RxBleRadioOperationScan(45418503)
D/RxBle#Radio:   QUEUED RxBleRadioOperationScan(108427424)
D/BluetoothAdapter: startLeScan(): null
D/BluetoothAdapter: STATE_ON
D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=6
D/BluetoothAdapter: stopLeScan()
D/BluetoothAdapter: STATE_ON
D/RxBle#Radio: FINISHED RxBleRadioOperationScan(45418503)
D/RxBle#Radio:  STARTED RxBleRadioOperationScan(108427424)
D/BluetoothAdapter: startLeScan(): null
D/BluetoothAdapter: STATE_ON
D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=6
D/RxBle#Radio: FINISHED RxBleRadioOperationScan(108427424)
D/BluetoothAdapter: stopLeScan()
D/BluetoothAdapter: STATE_ON
D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(232084268)
D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(232084268)
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(29412271)
V/RxBle#BleConnectionCompat: Connecting without reflection
D/BluetoothGatt: connect() - device: 04:A3:16:A7:AD:5D, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=70c2ac66-68f2-4146-bb05-696c14c2c50b
[... 300 lines cut ...]
        unsubscribeLast();
        commandSubscription = connection.writeCharacteristic(dataCharacterisctic, new byte[]{0x56, 0, 0, 0, level, (byte) 0x0F, (byte) 0xAA})
                .subscribe(onSuccess, onThrowable);
    }

    //abort previous command
    private void unsubscribeLast() {
        if (commandSubscription != null && !commandSubscription.isUnsubscribed()) {
            commandSubscription.unsubscribe();
        }
    }

    private String getConnectionState(RxBleConnection.RxBleConnectionState state) {
        try {
            Field field = RxBleConnection.RxBleConnectionState.class.getDeclaredField("description");
            field.setAccessible(true);
            return (String) field.get(state);
        } catch (IllegalAccessException e) {
            Logger.logError(LOG_TAG, e);
        } catch (NoSuchFieldException e) {
            Logger.logError(LOG_TAG, e);
        }
        return "undefined";
    }
}`

SsIDit, 2017-04-20:
It seems to me, that problem is in discovering services. From documentation of library:

> Result of the discovery is cached internally so consecutive calls won't trigger BLE operation

Are there any ways to force clearing services cache?

dariuszseweryn, 2017-04-20:
Force – currently not unless you will do it yourself in a custom operation.

I see that you are not managing the state of the connection and you may be using an old `RxBleConnection` before new is established.

SsIDit, 2017-04-20:
I just moved 
```
if (serviceDiscoverSubscription != null && !serviceDiscoverSubscription.isUnsubscribed()) {
                        serviceDiscoverSubscription.unsubscribe();
                    }
                    if (connStateSubscription != null && !connStateSubscription.isUnsubscribed()) {
                        connStateSubscription.unsubscribe();
                    }
```

in doOnUnsubscribe method of my establishConnection() subscription and everything worked.
Thank you so much! I could't solve this issue for a week. You've made me happy. Issue is solved.

dariuszseweryn, 2017-04-20:
Glad to help.

closed 2017-04-20 by dariuszseweryn