dariuszseweryn/RxAndroidBle #178: Stuck with Connecting state

kanat, 2017-04-21:
### Summary
Stuck with Connecting state
#### Preconditions
Im just calling `device.establishConnection(false)`
#### Steps to reproduce actual result
1. `device.establishConnection(false)`
2. Unsubcribe
3. `device.establishConnection(false)`
4. Repeat agin from 1 N times
#### Actual result
CONNECTING state

`ConnectionManager: (Key-1-1:2184) [connect] RxBleDeviceImpl{bluetoothDevice=StarLine BLE(F8:5B:60:89:CD:8A)} (autoConnect='false')`

And I get

`DeviceManager: (Key-1-1:2184) [start] Failed: BleGattCallbackTimeoutException{macAddress=null, bleGattOperationType=BleGattOperation{description='CONNECTION_STATE'}}`

I dont understand why mac address is null

#### Expected result

CONNECTED state
<br />

dariuszseweryn, 2017-04-21:
Nice one. It seems to be a race condition in the `RxBleRadioOperationConnect`. I am writing a fix.

kanat, 2017-04-24:
Is it possible to configure timeout value for ble operation?
Now it takes about 35 seconds to get BleGattCallbackTimeoutException.

dariuszseweryn, 2017-04-24:
From connection observable you can unsubscribe at any moment and it should be handled by the library.

kanat, 2017-04-25:
Unsubscribing is not helping. 

The same result.

kanat, 2017-04-25:
As Understand the problem occurs here in RxBleRadioOperationDisconnect (line 62):

```
if (bluetoothGatt == null) {
    RxBleLog.w("Disconnect operation has been executed but GATT instance was null.");
    onCompleted();
}
```

After calling `onCompleted` the method `releaseRadio` is never called.

dariuszseweryn, 2017-04-25:
True. Another thing that did not manifest before.

timmiller17, 2017-04-25:
I think I am struggling with this as well.  How is the fix coming along?

dariuszseweryn, 2017-04-25:
Fix for the original cause is ready and waiting on a private PR to be reviewed by @uKL but he has a quite full plate at the moment. I hope that he will have a moment tomorrow to check it out. The fix for the manifestation (Disconnect) I will make tomorrow.

kanat, 2017-04-25:
Is there any way to push my own **ConnectionModule** into the RxAndroidBle Graph?
(Just to provide fixed versions of operations, until official release will be published)

dariuszseweryn, 2017-04-25:
Unfortunately - There is no possibility to do this. You would have to compile your own version from the sources.

kanat, 2017-04-25:
What a pity.
That would be nice to have)

kanat, 2017-04-25:
> It seems to be a race condition in the RxBleRadioOperationConnect

@dariuszseweryn could you tell what was the problem at RxBleRadioOperationConnect?

dariuszseweryn, 2017-04-25:
`RxBleGattCallback` is emitting `onConnectionStateChange()` before the subscription chain is fully established.

dariuszseweryn, 2017-04-26:
@kanat I have pushed fixes for all issues found by you. These should be available in the `1.3.0-SNAPSHOT` artifact shortly. Could you give it a try?

kanat, 2017-04-26:
@dariuszseweryn yes, sure, I can try it with real device. 
Where/How can I get `1.3.0-SNAPSHOT` artifact?

dariuszseweryn, 2017-04-26:
See https://github.com/Polidea/RxAndroidBle
`Snapshot` paragraph

kanat, 2017-04-26:
Actually I've tried it with my own fixes, seems it works :)
Anyway I'll try `1.3.0-SNAPSHOT` as well.

kanat, 2017-04-26:
@dariuszseweryn It works, thanks.

kanat, 2017-04-26:
@dariuszseweryn, is it possible to get frozen version of `1.3.0-SNAPSHOT`, to avoid situation where my own app can be affected by new changes which can arrive with a modified SNAPSHOT?
Or maybe it's better to compile RxAndroidBle by my own, until the new release will be published?

dariuszseweryn, 2017-04-26:
Great news. I will try to publish `1.2.1` tomorrow. You are welcome.

Your own version will be more bulletproof. SNAPSHOTs tend to change but regression is rarely an issue.

closed 2017-04-26 by dariuszseweryn

dariuszseweryn, 2017-04-27:
@kanat I have just released `1.2.1`
It should be available shortly.

kanat, 2017-04-27:
@dariuszseweryn Great news, thanks!