dariuszseweryn/RxAndroidBle #179: Not using the new Android Scan API?

chuks, 2017-04-22:
### It looks like this library is using pre 5.0 Android BLE Scan API
<br />
#### 
<br />
#### Steps to reproduce actual result
1. Create A sample activity with this library
2. Start scan
3. Look at logs in Android Studio. 
#### Actual result 
- You'll see logs calling BluetoothAdapter: startLeScan() (which is long deprecated)
<br />
#### Expected result
- Expected to see the new Android BLE API, using BluetoothLeScanner
<br />

dariuszseweryn, 2017-04-22:
Hello @chuks 
If you would check out other issues before posting you would see #16 and #74 
This feature has currently the highest priority.
Best Regards

closed 2017-04-22 by dariuszseweryn

chuks, 2017-04-22:
Great, just reading those! I'm also happy to help, if you need any help. This is a great library. I had written one myself for internal projects, and was looking to open source it when I found this one, and I figured instead of releasing of mine, I could help you with yours, if needed. Do you have a timeline on when you think you'll release the new 1.3.0 version?