dariuszseweryn/RxAndroidBle #182: DaggerClientComponent class is undefined

ilya-payrange, 2017-04-27:
### Summary
DaggerClientComponent class is undefined
<br />
#### Preconditions
Pull latest RxAndroidBle
<br />
#### Steps to reproduce actual result
1.Pull latest RxAndroidBle
2.import into Android Studio project
3.Compile
#### Actual result
Error:(12, 32) error: cannot find symbol class DaggerClientComponent
on 
`import com.polidea.rxandroidble.DaggerClientComponent;`
<br />
#### Expected result
Should compile ok.
<br />

uKL, 2017-04-27:
Did you try compiling from the console?

dariuszseweryn, 2017-04-27:
@ilya-payrange `1.2.0` or `1.3.0-SNAPSHOT`?

ilya-payrange, 2017-04-27:
Resolved.

It was on Android Studio 2.4 preview, with Java 8 enabled (as sample apps require Java 8 and Jack is deprecated).

Studio 2.4 and toolchain barked at apt. I got it build ok replacing 
`apt rootProject.ext.libs.dagger_compiler`
with 
`annotationProcessor rootProject.ext.libs.dagger_compiler`
in RxAndroidBle build.gradle.

I wonder if there's a way to do the same without messing with build.gradle (so updates/pulls will need no 
patching)?

dariuszseweryn, 2017-04-28:
Nothing that I can know from top of my head.

closed 2017-05-12 by dariuszseweryn

KomoriWu, 2017-09-13:
How do you solve this problem

dariuszseweryn, 2017-09-13:
@KomoriWu currently the project uses `annotationProcessor` by default. If you have problems making it to work you can check out [this topic](https://stackoverflow.com/questions/42692525/how-to-enable-annotation-processor-in-android-studio-2-3) or [other similar](https://www.google.pl/search?q=android+studio+annotation+processor+not+working)

farmanvu, 2017-11-02:
@uKL and @dariuszseweryn. Thanks for your outstanding job of making RxAndroidBLE open lib.

 Issue: I have added the lib as a MODULE to my Android project but I am facing the error i.e DaggerClientComponent class is undefined and the class is not there in the lib files.
I am using Android Studio 3.0
and 
// Dagger dependencies
    compile 'com.google.dagger:dagger:2.10'
    annotationProcessor 'com.google.dagger:dagger-compiler:2.10'

How can I solve the issue, I need yours guidance in the respect.

dariuszseweryn, 2017-11-02:
Hello @farmanvu 
Have you tried to [google the problem](https://www.google.pl/search?q=annotationprocessor+android+studio+3.0)?
I once had to [enable `annotationProcessor` in Android Studio 2.3 which I use](https://stackoverflow.com/questions/42692525/how-to-enable-annotation-processor-in-android-studio-2-3) (I am not on 3.0 yet because of incompatibility of groovy plugin #302)
If you solve the issue feel free to report back