dariuszseweryn/RxAndroidBle #183: Gatt didn't colsed after ConnectionSubscription.unsubscribe

sycinsider, 2017-04-27:
I unsubscribe from connection subscription then I see in logs D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(257136686), but i cant see log:D/BluetoothGatt: close() after this;

D/RxBle#Radio:  STARTED RxBleRadioOperationServicesDiscover(59117293)
D/BluetoothGatt: discoverServices() - device: 00:0E:0B:00:00:0B
D/BluetoothGatt: onSearchComplete() = Device=00:0E:0B:00:00:0B Status=0
D/RxBle#BluetoothGatt: onServicesDiscovered status=0
D/RxBle#Radio: FINISHED RxBleRadioOperationServicesDiscover(59117293)
E/onFoundService: write now
D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000ffe1-0000-1000-8000-00805f9b34fb enable: true
D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(19773138)
D/RxBle#Radio:  STARTED RxBleRadioOperationDescriptorWrite(19773138)
D/RxBle#BluetoothGatt: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicLongWrite(19378408)
D/RxBle#Radio: FINISHED RxBleRadioOperationDescriptorWrite(19773138)
D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicLongWrite(19378408)
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000ffe1-0000-1000-8000-00805f9b34fb status=0
D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000ffe1-0000-1000-8000-00805f9b34fb enable: false
D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(40037033)
E/clearConnectionSubscription:: disconnect
D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(257136686)

dariuszseweryn, 2017-04-27:
As you can see the last started operation is: `D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicLongWrite(19378408)` which points to problem with the Long Write Operation. Your `WriteOperationAckStrategy` has a problematic implementation perhaps?

sycinsider, 2017-04-27:
Here is the code that i write long data,Where is the problem?

mWriteSubscribe = mRxBleConnectionObservable
                    .flatMap(rxBleConnection -> Observable.combineLatest(
                            rxBleConnection.setupNotification(notify),
                            rxBleConnection.setupNotification(notify),
                            Pair::new
                    ), (rxBleConnection, callbackObservablePair) -> { 
                        Observable<byte[]> deviceCallback0 = callbackObservablePair.first;
                        Observable<byte[]> deviceCallback1 = callbackObservablePair.second;
                        return rxBleConnection.createNewLongWriteBuilder() 
                                .setBytes(sendData) 
                                .setCharacteristicUuid(write)
                                .setWriteOperationAckStrategy(booleanObservable -> Observable.zip(
                                        // so we zip three observables
                                        deviceCallback0, 
                                        deviceCallback1, 
                                        booleanObservable,
                                        (callback0, callback1, aBoolean) -> aBoolean 
                                ))
                                .build();
                    })
                    .flatMap(observable -> observable)
                    .take(1) 
                    .doOnSubscribe(this::writeRunnable)
                    .subscribe(
                            bytes -> {
                                onWriteSuccess(); 
 },
                           this::onWriteFailure);

dariuszseweryn, 2017-04-27:
Maybe one of the `deviceCallback0` or `deviceCallback1` is not notifying?
What is `notify`? Do you create two notifications with the exact same `UUID`?

sycinsider, 2017-04-27:
I have modified your sample about it and i create two notifications with the exact same UUID;What should I do?

dariuszseweryn, 2017-04-27:
First of all you should try to understand what you are trying to achieve and why (if) you want to use a custom ACK Strategy.

Try using:
```
mWriteSubscribe = mRxBleConnectionObservable
                    .flatMap(rxBleConnection -> rxBleConnection.setupNotification(notify),
                     (rxBleConnection, notificationObservable) -> { 
                        Log.d("XXX", "Creating Long Write Operation";
                        return rxBleConnection.createNewLongWriteBuilder() 
                                .setBytes(sendData) 
                                .setCharacteristicUuid(write)
                                .setWriteOperationAckStrategy(booleanObservable -> Observable.zip(
                                        notificationObservable.doOnNext(ignoredBytes -> Log.d("XXX", "notification")), 
                                        booleanObservable.doOnNext(ignoredBytes -> Log.d("XXX", "chunk written")),
                                        (notification, aBoolean) -> aBoolean 
                                ))
                                .build();
                    })
                    .flatMap(observable -> observable)
                    .take(1) 
                    .doOnSubscribe(this::writeRunnable)
                    .subscribe(
                            bytes -> {
                                onWriteSuccess(); 
                            },
                            this::onWriteFailure
                    );
```
And check logs. It is most likely that there will not be any `notification` in the logs. In which case it is most likely that your peripheral is not notifying anything.

dariuszseweryn, 2017-05-12:
No further contact from @sycinsider. 
Assuming the problem was solved.
Best Regards.

closed 2017-05-12 by dariuszseweryn