dariuszseweryn/RxAndroidBle #186: Mysterious crash apparently from ClassCastException

RobLewis, 2017-05-02:
### Summary
Have been struggling to debug this error for weeks. 
<br />
#### Preconditions
Android MainActivity that calls a second AeroscopeDisplay activity
<br />
#### Steps to reproduce actual result
1.I can't really say
2.
3.
#### Actual result
See log (especially the exception dump at the end)
<br />
#### Expected result
No crash. 
You have said before that problems could be caused by lambdas, but haven't supplied any details. See, for example, this: @https://github.com/Polidea/RxAndroidBle/issues/173
[logs_26_05-1.txt](https://github.com/Polidea/RxAndroidBle/files/969230/logs_26_05-1.txt)

<br />

yohanan, 2017-05-02:
```rx.exceptions.OnErrorFailedException: Error occurred when trying to propagate error to Observer.onError```

do you have a line somewhere that calls onError?  in particular, do you have one that is something similar to the following?

```Observer.onError(Exceptions.propagate(throwable));```

RobLewis, 2017-05-02:
Thanks, @yohanan—I really appreciate the help. 

I actually have several `onError()` handlers in my code, for the various Subscribers I'm using. Many of them are lambdas or method references, and most don't do anything except log the error and maybe set a flag. A few re-throw the error as a RuntimeException (maybe I should be looking more closely at these?) I don't use the `.propagate()` construct as I'm not aware of it. 

As I noted, Polidea responded to my earlier issue (@#173) by saying it was "probably caused by a lambda", but I haven't been able to find out what they mean by that. Are lambdas not compiled correctly (I'm using Jack)? Do I have a logic error? 

Another head scratcher is the log saying the ultimate cause is a Class Cast Exception, because one of my classes can't be cast to a different one. 

I'm no expert at reading logs, but it seems to me that a line like this: 
`at io.aeroscope.aeroscope.-$Lambda$13.$m$0(Unknown Source)
`
is saying that the routine causing the error is known by the internally-generated name `-$Lambda$13.$m$0`, which isn't much help. (I'm guessing that the compiler is creating inner classes to implement the lambdas, and giving them names like this. It would be interesting if there was some kind of compiler output that listed stuff like this.)

yohanan, 2017-05-02:
a colleague of mine was struggling last night with what appears to be the same issue; the stack traces look very similar.

in our case, we were mistakenly calling ```onError(Exceptions.propagate(throwable))``` because of a misinterpretation on how ```Exceptions.propagate()``` worked: we thought it **returned** an exception, but it actually _**throws**_ one.  so, we were effectively throwing an exception in something designed to pass along one; hence, the error.

there is a good chance your issue has nothing to do with the RxAndroidBle library.  there is a posting on Stack Overflow that discusses this more generally --- not specific to ```Exeptions.propagate()```:
 [RxJava: Error occurred when trying to propagate error to Observer.onError](http://stackoverflow.com/questions/31482682/rxjava-error-occurred-when-trying-to-propagate-error-to-observer-onerror).

this all points to throwing an exception in an ```onError()``` method, so i'd focus on those in your code.

as a suggestion related to the lambdas, if you're having trouble isolating the issue because of them, it might help to convert temporarily to anonymous inner classes.  this can easily be done in Android Studio by navigating to the lambda and pressing ```alt+return``` then selecting _Replace lambda with anonymous class_ from the popup menu.  you can convert back in a similar manner.

RobLewis, 2017-05-02:
OK, grasping at straws (or perhaps threads?) here, but the log indicates that the last operation on the thread that crashed (the main thread 30032-30032) was an attempt to write Characteristic 1237: 
```
05-01 12:55:01.906 30032-30032/io.aeroscope.aeroscope D/BluetoothGatt: writeCharacteristic() - uuid: 00001237-0000-1000-8000-00805f9b34fb
05-01 12:55:01.906 30032-30032/io.aeroscope.aeroscope D/BluetoothDevice: mAddress: CC:C6:B5:9F:9A:F7
05-01 12:55:01.999 30032-30051/io.aeroscope.aeroscope D/BluetoothGatt: onCharacteristicWrite() - Device=CC:C6:B5:9F:9A:F7 UUID=00001237-0000-1000-8000-00805f9b34fb Status=0 srvcType=0 srvcInstId=0 charInstId=0
05-01 12:55:01.999 30032-30051/io.aeroscope.aeroscope D/BluetoothDevice: mAddress: CC:C6:B5:9F:9A:F7
05-01 12:55:01.999 30032-30051/io.aeroscope.aeroscope D/BluetoothDevice: mAddress: CC:C6:B5:9F:9A:F7
05-01 12:55:01.999 30032-30051/io.aeroscope.aeroscope D/BluetoothDevice: mAddress: CC:C6:B5:9F:9A:F7
05-01 12:55:01.999 30032-30051/io.aeroscope.aeroscope D/BluetoothDevice: mAddress: CC:C6:B5:9F:9A:F7
05-01 12:55:02.000 30032-30051/io.aeroscope.aeroscope D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=00001237-0000-1000-8000-00805f9b34fb status=0
05-01 12:55:02.003 30032-30058/io.aeroscope.aeroscope D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicWrite(722505227)
05-01 12:55:02.003 30032-30058/io.aeroscope.aeroscope D/RxBle#Radio:  STARTED RxBleRadioOperationDescriptorWrite(89438770)
05-01 12:55:02.005 30032-30032/io.aeroscope.aeroscope E/AndroidRuntime: FATAL EXCEPTION: main
                                                                        Process: io.aeroscope.aeroscope, PID: 30032
                                                                        java.lang.IllegalStateException: Fatal Exception thrown on Scheduler.Worker thread.
                                                                            at rx.android.schedulers.LooperScheduler$ScheduledAction.run(LooperScheduler.java:114)
                                                                            at android.os.Handler.handleCallback(Handler.java:815)
                                                                            at android.os.Handler.dispatchMessage(Handler.java:104)
                                                                            at android.os.Looper.loop(Looper.java:194)
                                                                            at android.app.ActivityThread.main(ActivityThread.java:5576)
                                                                            at java.lang.reflect.Method.invoke(Native Method)
                                                                            at java.lang.reflect.Method.invoke(Method.java:372)
                                                                            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:956)
                                                                            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:751)
                                                                         Caused by: rx.exceptions.OnErrorFailedException: Error occurred when trying to propagate error to Observer.onError
                                                                            at rx.observers.SafeSubscriber._onError(SafeSubscriber.java:187)
                                                                            at rx.observers.SafeSubscriber.onError(SafeSubscriber.java:115)
                                                                            at rx.exceptions.Exceptions.throwOrReport(Exceptions.java:216)
                                                                            at rx.observers.SafeSubscriber.onNext(SafeSubscriber.java:139)
                                                                            at rx.internal.operators.OperatorObserveOn$ObserveOnSubscriber.call(OperatorObserveOn.java:224)
                                                                            at rx.android.schedulers.LooperScheduler$ScheduledAction.run(LooperScheduler.java:107)
                                                                            at android.os.Handler.handleCallback(Handler.java:815) 
                                                                            at android.os.Handler.dispatchMessage(Handler.java:104) 
                                                                            at android.os.Looper.loop(Looper.java:194) 
                                                                            at android.app.ActivityThread.main(ActivityThread.java:5576) 
                                                                            at java.lang.reflect.Method.invoke(Native Method) 
                                                                            at java.lang.reflect.Method.invoke(Method.java:372) 
                                                                            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:956) 
                                                                            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:751) 
(etc.)
```
Here's that code: 
```
    void copyFpgaRegisterBlockToAeroscope() {
        try {
            this.getConnectionObservable()
[... cut ...]
I'm going to try removing the `.observeOn( AndroidSchedulers.mainThread() )` because it doesn't seem necessary. Also I'm adding the commented-out `.doOnNext()` to see if I can get it to log anything; currently none of the `Log.d()` statements are working. 

Final thought: is `catch( Exception e )` possibly a problem? (I only added it to try to find this problem). Is it possible that an error from `writeCharacteristic(`) isn't an Exception but some other Throwable? Should I handle this differently? Or remove it?

RobLewis, 2017-05-02:
This is very promising @yohanan -- thanks bigly! The StackOverflow post you referenced points to [another one](http://stackoverflow.com/questions/24860942/observer-onerror-firing-off-inconsistently), which also looks helpful.

dariuszseweryn, 2017-05-12:
As it turned out - the cause was `jack` which was not regenerating lambdas as it should. Clean build solved the problem.

closed 2017-05-12 by dariuszseweryn