dariuszseweryn/RxAndroidBle #187: Document characteristics of various Observables

RobLewis, 2017-05-02:
### Summary
Documentation request
<br />
I frequently have questions about the behavior of the various Observables produced by RxAndroidBle, and can't find the answers in the JavaDoc or anywhere else. 

For example, does the `readCharacteristic` Observable emit a single value and then stop? Does it call `onCompleted()` after emitting the value? What about the notification Observables? 

I suggest a table showing, for each Observable, whether it is single-valued, multi-valued, or infinite, hot or cold, and describing under what circumstances its `onError()` and `onCompleted()` methods are invoked. If this already exists somewhere, kindly point me at it. 
<br />

dariuszseweryn, 2017-05-17:
Added in Readme.

closed 2017-05-17 by dariuszseweryn

RobLewis, 2017-05-17:
Very helpful—thank you!