dariuszseweryn/RxAndroidBle #188: Direct connection is sometimes timeouted for 20s

misevcik, 2017-05-04:
### Summary
I've got following connection workflow:
1.) Start application
2.) Scan
3.) If device in range, stop scan. Direct connect to device

Sometimes (1 out if 10-20 attempts) connection is timeouted for 20s. Device I'm connecting to is in range and visible by scan. 
I got following error:
05:16:31.276 17413-17413/rxble E/BleClient: onConnectionFailure BleGattException{macAddress=00:16:A4:11:52:F1, status=133 (0x85 -> https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h), bleGattOperationType=BleGattOperation{description='CONNECTION_STATE'}}

Next connection attempt is always successful. 

#### Preconditions
Unknown
#### Steps to reproduce actual result
1.Unknown
#### Actual result
Sometimes connection is time outed.
#### Expected result
Connection is always successful.

###Log:
05:16:10.828 17413-17413/rxble I/MAIN: START
05:16:10.854 17413-17413/rxble D/BleScanner: Scan :true
05:16:10.856 17413-17413/rxble D/RxBle#Radio:   QUEUED RxBleRadioOperationScan(175046017)
05:16:10.857 17413-17626/rxble D/RxBle#Radio:  STARTED RxBleRadioOperationScan(175046017)
05:16:10.858 17413-17413/rxble I/FragmentContainer: CREATED
05:16:10.859 17413-17413/rxble D/ActivityThread: add activity client record, r= ActivityRecord{e168603 token=android.os.BinderProxy@a8a46b2 {rxble/rxble.MainActivity}} token= android.os.BinderProxy@a8a46b2
05:16:10.859 17413-17413/rxble I/MAIN: onResume
05:16:10.863 17413-17413/rxble I/MainActivity: onPause
05:16:10.871 17413-17413/rxble D/BluetoothAdapter: startLeScan(): null
05:16:10.871 17413-17413/rxble I/BluetoothAdapter: getBluetoothLeScanner
05:16:10.872 17413-17413/rxble I/BluetoothLeScanner: startScan 3 is called
05:16:10.874 17413-17413/rxble D/BluetoothAdapter: STATE_ON
05:16:10.881 17413-17425/rxble D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
05:16:10.884 17413-17626/rxble D/RxBle#Radio: FINISHED RxBleRadioOperationScan(175046017)
05:16:10.948 17413-17413/rxble I/MAIN: onResume
05:16:10.993 17413-17413/rxble D/Device: Created {name:myDevice, address:00:16:A4:11:52:F1}
05:16:10.994 17413-17413/rxble D/BleScanner: Scan :true
05:16:10.994 17413-17413/rxble D/BleScanner: Scan :true
05:16:11.125 17413-17413/rxble I/BleScanner: addScanResult:{myDevice:00:16:A4:11:52:F1}
05:16:11.125 17413-17413/rxble D/BleScanner: Scan :false
05:16:11.126 17413-17413/rxble D/BluetoothAdapter: stopLeScan()
05:16:11.126 17413-17413/rxble I/BluetoothAdapter: getBluetoothLeScanner
05:16:11.127 17413-17413/rxble I/BluetoothLeScanner: stopScan is called
05:16:11.127 17413-17413/rxble D/BluetoothAdapter: STATE_ON
05:16:11.136 17413-17413/rxble D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(21531100)
05:16:11.137 17413-17626/rxble D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(21531100)
05:16:11.141 17413-17413/rxble I/BluetoothDevice: connectGatt: 2
05:16:11.141 17413-17413/rxble D/BluetoothGatt: connect() - device: 00:16:A4:11:52:F1, auto: false
05:16:11.141 17413-17413/rxble D/BluetoothGatt: registerApp()
05:16:11.141 17413-17413/rxble D/BluetoothGatt: registerApp() - UUID=a19129cb-2387-40a0-b04b-bf4eb0b58aad
05:16:11.146 17413-17448/rxble D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
05:16:11.414 17413-17413/rxble D/ActivityThread: Remove activity client record, r= ActivityRecord{dc3da4d token=android.os.BinderProxy@bc675e4 {rxble/rxble.SplashScreen}} token= android.os.BinderProxy@bc675e4
05:16:31.266 17413-17424/rxble D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=00:16:A4:11:52:F1
05:16:31.267 17413-17424/rxble D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=133
05:16:31.270 17413-17626/rxble D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(21531100)
05:16:31.271 17413-17424/rxble D/RxBle#RadioOperationConnect: No subscribers, finishing operation
05:16:31.273 17413-17424/rxble D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(228502169)
05:16:31.274 17413-17626/rxble D/RxBle#Radio:  STARTED RxBleRadioOperationDisconnect(228502169)
05:16:31.276 17413-17413/rxble D/BleClient: onConnectionUnsubscribed 
05:16:31.276 17413-17413/rxble E/BleClient: onConnectionFailure BleGattException{macAddress=00:16:A4:11:52:F1, status=133 (0x85 -> https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h), bleGattOperationType=BleGattOperation{description='CONNECTION_STATE'}}
05:16:31.280 17413-17413/rxble D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(162350442)
05:16:31.282 17413-17413/rxble D/BluetoothManager: getConnectionState()
05:16:31.282 17413-17413/rxble D/BluetoothManager: getConnectedDevices
05:16:31.296 17413-17413/rxble D/BluetoothGatt: close()
05:16:31.296 17413-17413/rxble D/BluetoothGatt: unregisterApp() - mClientIf=5
05:16:31.299 17413-17626/rxble D/RxBle#Radio: FINISHED RxBleRadioOperationDisconnect(228502169)
05:16:31.300 17413-17626/rxble D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(162350442)
05:16:31.325 17413-17413/rxble I/BluetoothDevice: connectGatt: 2
[... 22 lines cut ...]
05:16:31.753 17413-17635/rxble V/BleClient: onNotificationHasBeenSetUp
05:16:31.753 17413-17635/rxble D/Device: onDeviceHwConnectionEvent {state=2 hasProtocol=null address=00:16:A4:11:52:F1}
05:16:31.757 17413-17635/rxble D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(25762952)
05:16:31.760 17413-17626/rxble D/RxBle#Radio: FINISHED RxBleRadioOperationDescriptorWrite(133504918)
05:16:31.760 17413-17626/rxble D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicWrite(25762952)
05:16:31.854 17413-17471/rxble D/RxBle#BluetoothGatt: onCharacteristicChanged characteristic=569a2000-b87f-490c-92cb-11ba5ea5167c
05:16:31.856 17413-17471/rxble D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=569a2001-b87f-490c-92cb-11ba5ea5167c status=0
05:16:31.858 17413-17626/rxble D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicWrite(25762952)
05:16:31.872 17413-17413/rxble I/VersionBase: authorize RandValue:639219, sent:false
05:16:31.873 17413-17413/rxble D/Utils: codeUserIdByte {userID:11011512249623163464, codeUserId:[17, 1, 21, 18, 36, -106, 35, 22, 52, 100], outData:[17, 1, 21, 18, 36, -106, 35, 22, 52, 100]
05:16:31.875 17413-17413/rxble D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(121249229)
05:16:31.876 17413-17413/rxble D/BleClient: Write success
05:16:31.876 17413-17626/rxble D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicWrite(121249229)
05:16:31.954 17413-17424/rxble D/RxBle#BluetoothGatt: onCharacteristicChanged characteristic=569a2000-b87f-490c-92cb-11ba5ea5167c
05:16:31.955 17413-17424/rxble D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=569a2001-b87f-490c-92cb-11ba5ea5167c status=0
05:16:31.958 17413-17626/rxble D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicWrite(121249229)
05:16:31.975 17413-17413/rxble I/VersionBase: getNumberOfUsers
05:16:31.977 17413-17413/rxble D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(124338663)
05:16:31.978 17413-17413/rxble D/BleScanner: Scan :false
05:16:31.979 17413-17626/rxble D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicWrite(124338663)
05:16:31.983 17413-17413/rxble D/BleClient: Write success
05:16:32.153 17413-17448/rxble D/RxBle#BluetoothGatt: onCharacteristicChanged characteristic=569a2000-b87f-490c-92cb-11ba5ea5167c
05:16:32.154 17413-17448/rxble D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=569a2001-b87f-490c-92cb-11ba5ea5167c status=0
05:16:32.155 17413-17626/rxble D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicWrite(124338663)
05:16:32.192 17413-17413/rxble D/BleClient: Write success

dariuszseweryn, 2017-05-10:
Hello @misevcik 
Unfortunately this status is returned by the native Android BLE Stack and we can do nothing to fix it (or at least I do not know what could be done).
Best Regards

closed 2017-05-10 by dariuszseweryn

ardmn, 2017-05-22:
@misevcik Are You have solve your problem ?