dariuszseweryn/RxAndroidBle #189: Notifications arriving in bursts on 3 different threads

RobLewis, 2017-05-06:
### Summary
Notifications from BLE peripheral are being received in bursts of 6, on 3 different threads
<br />
#### Preconditions
Subscribe to notifications on characteristics 0x1235 and 0x1239
<br />
#### Steps to reproduce actual result
1. Subscribe to notifications on 2 different characteristics
2. See log
3.
#### Actual result
Logged packet arrivals occur mostly on one thread (9709), but occasionally on a different one (9710). After about 1 second, a third thread (9888) begins delivering occasional packets. 6 packets are received quickly (1-2 ms apart), then there is a ~40 ms delay, and the process repeats.  
<br />
#### Expected result
Continuous reception of packets. 
<br />
I have actually subscribed to notifications on 2 different characteristics. The other one (0x1239) is low traffic. Later in the log (not shown), it sends 4 packets on the same 9709 and 9710 threads as the 0x1235 characteristic. 
I realize that the "bursty" transmission could be coming from the peripheral, but this seems unlikely and I have no real way to test it. 
Question: could multiple executions of the `setupNotification()` code on the same characteristic cause these multiple threads? If so, is there a downside to this? 
Since there is some significant processing to be done on each arriving packet, I have subscribed to notifications on `Schedulers.computation()`, which I read somewhere dedicates a CPU core to the operation. Code: 
```
dataNotifSub = selectedScope.getConnectionObservable( )
    .subscribeOn( Schedulers.computation( ) )  
    .flatMap( rxBleConnection -> rxBleConnection.setupNotification( asDataCharId ) )
    .doOnNext( notificationObservable -> runOnUiThread( this::dataNotificationHasBeenSetUp ) )
    .doOnError( setupThrowable -> runOnUiThread( this::dataNotificationSetupError ) )
    .flatMap( notificationObservable -> notificationObservable )
    .timestamp( )
    .doOnUnsubscribe( () -> Log.d( LOG_TAG, "doOnUnsubscribe() invoked by dataNotifSub" ) )
    .subscribe( this::onDataNotificationReceived, this::onDataNotificationError ); 
```

Log excerpt (with extraneous messages filtered) below: 

```
05-05 17:08:52.060 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.061 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.062 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.064 9692-9710/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.065 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.066 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.067 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.111 9692-9710/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.113 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.114 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.115 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.116 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.117 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.160 9692-9710/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.161 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.164 9692-9710/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.165 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.166 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.167 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.257 9692-9710/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.260 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.262 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.263 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.264 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.265 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.307 9692-9710/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.308 9692-9710/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.310 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.312 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.313 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.315 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.358 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.359 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:52.360 9692-9709/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
[... cut ...]
05-05 17:08:53.267 9692-9710/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:53.270 9692-9710/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
05-05 17:08:53.271 9692-9710/io.aeroscope.aeroscope D/BluetoothGatt: onNotify() - Device=CC:C6:B5:9F:9A:F7 UUID=00001235-0000-1000-8000-00805f9b34fb srvcType=0 srvcInstId=0 charInstId=0
```

dariuszseweryn, 2017-05-15:
Hello @RobLewis 
Multiple executions of `setupNotifications()` are using the same `Observable` so the notifications are setup only once.
The logs you see are from the native Android BLE stack and that is exactly as Android behaves. It is using multiple background threads for working with BLE. The library then moves all callbacks to a different thread so the Android threads won't be occupied for too long (which could break the BLE stack).
Does it answer your question?

RobLewis, 2017-05-15:
*Sigh* I have so much to learn. Thank you.

closed 2017-05-15 by RobLewis