dariuszseweryn/RxAndroidBle #190: Untraceable, seemingly random crashes in RxBleScan

RobLewis, 2017-05-08:
### Summary
Sometimes, but not always, the app crashes after a scan
<br />
#### Preconditions
Scan for BLE devices
<br />
#### Steps to reproduce actual result
1. Start a scan
2. Scan finishes
3. Long and complex stack trace
#### Actual result
Crash
<br />
#### Expected result
No crash
<br />

I have been chasing this problem for weeks, and thought I had fixed it more than once. At first I thought it was the `.observeOn( AndroidSchedulers.mainThread())`, so I removed that, and it worked for a while. Then I thought maybe it was the `.take()` operators, so I removed those. Again, it seemed to work. I have now removed everything that isn't essential, and it's still blowing up. 

The last item in the stack trace "caused by" chain is this: 
`Caused by: java.lang.ClassCastException: io.aeroscope.aeroscope.AeroscopeBluetoothService cannot be cast to io.aeroscope.aeroscope.AeroscopeDisplay`
Maybe that is a clue. The above code resides in a Bound Service called AeroscopeBluetoothService, and is called by the Activity AeroscopeDisplay. The crash seems to have something to do with propagating an error; could that somehow cause the ClassCastException? 
And is there a chance that switching to the newer API 21 scanning API may help this? I'm getting desperate. 

Code: 
```
public void scanForAeroscopes( int numberToFind, long secondsToScan ) {  
        Log.d( LOG_TAG, "Entered new scanForAeroscopes()" );
        rxBleDeviceVector.clear( );
        asDeviceVector.clear( );
        nowScanning.set( true );                             // set "scan active" flag (synchronized)
        errorScanning.set( false );                          // clear the "scan error" flag
        scanSubscription = asBleClient
                .scanBleDevices( asServiceIdArray )          // argument must be an array, apparently
                //.take( numberToFind )                                // removed
                //.take( secondsToScan, TimeUnit.SECONDS )   // removed
                //.observeOn( AndroidSchedulers.mainThread() ) // removed
                .subscribe(
                        this::addRxBleScanResult, 
                        this::handleRxBleScanError,    
                        this::completeRxBleScan              // (was being called when .take() fired)
                );
    }

    public void scanForAeroscopes( ) {  //  no-arg overload
        scanForAeroscopes( MAX_AEROSCOPES, SCAN_TIME ); 
    }

    private void addRxBleScanResult( RxBleScanResult scanResult ) {
        RxBleDevice device = scanResult.getBleDevice();
        Log.d( LOG_TAG, "addRxBleScanResult about to call addRxBleDevice with device " + device.toString() );
        addRxBleDevice( device );
        completeRxBleScan();  // (have to fake a completion because the now-removed .take() was only thing causing scan to complete)
    }

    private void handleRxBleScanError( Throwable scanError ) {
        nowScanning.set( false );  
        errorScanning.set( true ); 
        Log.d( LOG_TAG, "scanSubscription onError() received: " + scanError.getMessage(), scanError );
    }

    private void completeRxBleScan() {   
        Log.d( LOG_TAG , "completeRxBleScan was called by scanSubscription; unsubscribing..." );
        scanSubscription.unsubscribe();
        nowScanning.set( false );  
        errorScanning.set( false ); 
        // Broadcast the results back to MainActivity
        Intent scanIntent = new Intent( "scanResults" );
        scanIntent.putExtra( "io.aeroscope.aeroscope.DeviceVector", asDeviceVector ); 
        LocalBroadcastManager.getInstance( getApplicationContext() ).sendBroadcast( scanIntent ); 
[... 138 lines cut ...]
                                                                            at rx.internal.operators.OnSubscribeRefCount$2.onNext(OnSubscribeRefCount.java:120)
                                                                            at rx.internal.operators.OperatorPublish$PublishSubscriber.dispatch(OperatorPublish.java:585)
                                                                            at rx.internal.operators.OperatorPublish$PublishSubscriber.onNext(OperatorPublish.java:283)
                                                                            at rx.internal.operators.OnSubscribeMap$MapSubscriber.onNext(OnSubscribeMap.java:77)
                                                                            at rx.internal.operators.OperatorMerge$MergeSubscriber.emitScalar(OperatorMerge.java:395)
                                                                        	at rx.intern
05-08 11:17:17.770 30139-30150/io.aeroscope.aeroscope W/art: Suspending all threads took: 8.263ms
05-08 11:18:41.585 30139-30157/io.aeroscope.aeroscope D/MessageLogger: >>> Entering MessageLogger.dump. to Dump MSG HISTORY <<<
05-08 11:18:41.603 30139-30157/io.aeroscope.aeroscope D/MessageLogger: MSG HISTORY IN MAIN THREAD:
                                                                       Current kernel time : 76453073ms
                                                                       Last record : Msg#:106 >>>>> Dispatching to Handler (android.os.Handler) {2c258b99} android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1@ddb3e5e: 0
                                                                       Last record dispatching elapsedTime:211160 ms/upTime:211160 ms
                                                                       Last record dispatching time : 2017-05-08 11:15:10.440
                                                                       Msg#:105 <<<<< Finished to Handler (android.view.Choreographer$FrameHandler) {a12d602} android.view.Choreographer$FrameDisplayEventReceiver@ec5eb13 elapsedTime:1 ms/upTime:2 ms
                                                                       Msg#:104 >>>>> Dispatching to Handler (android.view.Choreographer$FrameHandler) {a12d602} android.view.Choreographer$FrameDisplayEventReceiver@ec5eb13: 0 from 2017-05-08 11:15:10.382
                                                                       Msg#:103 <<<<< Finished to Handler (android.view.Choreographer$FrameHandler) {a12d602} android.view.Choreographer$FrameDisplayEventReceiver@ec5eb13 elapsedTime:2 ms/upTime:3 ms
                                                                       Msg#:102 >>>>> Dispatching to Handler (android.view.Choreographer$FrameHandler) {a12d602} android.view.Choreographer$FrameDisplayEventReceiver@ec5eb13: 0 from 2017-05-08 11:15:10.365
                                                                       Msg#:101 <<<<< Finished to Handler (android.view.Choreographer$FrameHandler) {a12d602} android.view.Choreographer$FrameDisplayEventReceiver@ec5eb13 elapsedTime:2 ms/upTime:3 ms
                                                                       Msg#:100 >>>>> Dispatching to Handler (android.view.Choreographer$FrameHandler) {a12d602} android.view.Choreographer$FrameDisplayEventReceiver@ec5eb13: 0 from 2017-05-08 11:15:10.349
                                                                       Msg#:99 <<<<< Finished to Handler (android.view.Choreographer$FrameHandler) {a12d602} android.view.Choreographer$FrameDisplayEventReceiver@ec5eb13 elapsedTime:3 ms/upTime:3 ms
                                                                       Msg#:98 >>>>> Dispatching to Handler (android.view.Choreographer$FrameHandler) {a12d602} android.view.Choreographer$FrameDisplayEventReceiver@ec5eb13: 0 from 2017-05-08 11:15:10.331
                                                                       Msg#:97 <<<<< Finished to Handler (android.view.Choreographer$FrameHandler) {a12d602} android.view.Choreographer$FrameDisplayEventReceiver@ec5eb13 elapsedTime:1 ms/upTime:1 ms
                                                                       Msg#:96 >>>>> Dispatching to Handler (android.view.Choreographer$FrameHandler) {a12d602} android.view.Choreographer$FrameDisplayEventReceiver@ec5eb13: 0 from 2017-05-08 11:15:10.316
                                                                       ***Flushing, Current Size Is:2135,boolfalse***TAIL
```

dariuszseweryn, 2017-05-12:
As it turned out - the cause was `jack` which was not regenerating lambdas as it should. Clean build solved the problem.

closed 2017-05-12 by dariuszseweryn

RobLewis, 2017-05-12:
Just as an additional data point, I have now run several development versions of the app multiple times. As long as I Clean the project before each run, I don't have a problem. I'm still using Jack. The Clean operation is fairly quick, so it's not too burdensome.